pulp-platform/pulpissimo #126: Memory Limitations

asyms, 2020-04-13:
Hi,

I had a question about the memory of the Pulpissimo platform.
I've been running come code that runs a convolutional network in C. The input and weights do not exceed the 512 kB memory limit, but I believe that due to other global variables defined in the code, it does exceed the limit.

This results in the following error after running the code:

```
# [STDOUT-CL31_PE0] Simple CONVNET !
# 162667311ns: Illegal instruction (core 0) at PC 0xbadacce6:
# 162667321ns, Branch decision is X in module tb_pulp.i_dut.soc_domain_i.pulp_soc_i.fc_subsystem_i.FC_CORE.lFC_CORE.id_stage_i
# ** Note: $stop    : /../ips/riscv/./rtl/riscv_id_stage.sv(1635)
#    Time: 162667320591 ps  Iteration: 16  Instance: /tb_pulp/i_dut/soc_domain_i/pulp_soc_i/fc_subsystem_i/FC_CORE/lFC_CORE/id_stage_i
# Break at /../ips/riscv/./rtl/riscv_id_stage.sv line 1635
# Stopped at/../ips/riscv/./rtl/riscv_id_stage.sv line 1635
# End time: 20:00:34 on Apr 09,2020, Elapsed time: 0:11:23
# Errors: 4, Warnings: 24
Simulator reported an error, leaving
make: *** [run] Error 1
```

If I run the code with smaller dimensions overall, the code does work without generating that error.

Is there a way to increase the L2 memory size? Or another fix for this issue?

Kind regards,
Arne

asyms, 2020-04-26:
Hi, thought I'd update this, as I've been doing some more experimenting, trying to fix the issue. 

After investigating the trace_core_1f_0.log and reading the ELF created, I don't believe the memory of the platform is the actual issue. In a new example execution, I was getting the same error, but at a different PC value: "0x3ef363ba". Nowhere in the ELF file any of the objects or functions are actually at this location. 

Below is a snippet of the log file where the line containing the illegal return address
is highlighted:

```
Time                Cycles   PC      Instr  Mnemonic
75542968ns         2087644 1c00af02 fff60613 addi             x12, x12, -1        x12=00000003 x12:00000004
75542988ns         2087645 1c00af04 00e00393 addi             x7, x0, 14           x7=0000000e
75543008ns         2087646 1c00af06 00d5c533 xor              x10, x11, x13       x10=00000000 x11:00000000 x13:00000000
75543028ns         2087647 1c00af0a 001f0e13 addi             x28, x30, 1         x28=00000001 x30:00000000
75543048ns         2087648 1c00af0e 0ec3e963 bltu             x7, x12, 242         x7:0000000e x12:00000003
75543069ns         2087649 1c00af12 1c000f37 lui              x30, 0x1c000000     x30=1c000000
75543089ns         2087650 1c00af16 00261613 slli             x12, x12, 0x2       x12=0000000c x12:00000003
75543109ns         2087651 1c00af18 624f0f13 addi             x30, x30, 1572      x30=1c000624 x30:1c000000
75543129ns         2087652 1c00af1c 20cf7603 lw               x12, x12(x30)       x12=3ef363ba x12:0000000c x30:1c000624  PA:1c000630
-->75543190ns         2087655 1c00af20 00060067 jalr             x0, x12, 0          x12:3ef363ba
75543311ns         2087661 1c008000 3970106f jal              x0, 7062           
75543371ns         2087664 1c009b96 fe112e23 sw               x1, -4(x2)           x1:1c0091fe  x2:1c000fe0  PA:1c000fdc
75543392ns         2087665 1c009b9a fea12c23 sw               x10, -8(x2)         x10:00000000  x2:1c000fe0  PA:1c000fd8
75543412ns         2087666 1c009b9e 00006517 auipc            x10, 0x6000         x10=1c00fb9e
75543432ns         2087667 1c009ba2 5ca50513 addi             x10, x10, 1482      x10=1c010168 x10:1c00fb9e
75543452ns         2087668 1c009ba6 010000ef jal              x1, 16               x1=1c009baa
75543493ns         2087670 1c009bb6 f8010113 addi             x2, x2, -128         x2=1c000f60  x2:1c000fe0
75543513ns         2087671 1c009bb8 00112023 sw               x1, 0(x2)            x1:1c009baa  x2:1c000f60  PA:1c000f60
```

Any ideas as to where this address might come from, or how I can further debug this?
Hopefully the above information helps.

Kind regards,
Arne

bluewww, 2020-04-26:
Normally I do a `objdump -D my_elf` and look at the instruction that causes the issue. If it turns out that at the PC where the exception happens there is no code (e.g. there is a data section) then therere is a linker bug/problem with laying out the `.text` section. Other cases are sometimes that the entry point is set incorrectly in the simulation, bug I think I fixed that bug. Other than that I can't really guess what your specific problem is.

By the way looking at the log you pasted initially
```
# 162667311ns: Illegal instruction (core 0) at PC 0xbadacce6:
```
the `0xbadacce6` reads to me `badacces`. Maybe grep your code for this string and check out where you can go from there.

asyms, 2020-04-27:
Thanks for your reply @bluewww!

Trying to objdump the created elf outputs the following:
```
build/pulpissimo/test_conv_causal/test_conv_causal:     file format elf32-little

objdump: can't disassemble for architecture UNKNOWN!
```
I don't know how to fix this?

About the `0xbadacce6`, I hadn't made the connection with `badacces`. However, I don't have that anywhere in my code, and changing the code by for example adding more prints to see where I am in execution, changes this PC. So I believe that was a pure coincidence.

Thanks for your time,
Arne

meggiman, 2020-04-27:
You have to use the objdump provided by the RISC-V GCC toolchain i.e. riscv32-unknown-elf-objdump. If I am not mistaken 0xbadacce6 is the value returned by the memory bus when you try to access an illlegal memory location. So this kind of error to me looks like your program tried to jump to a function using a function pointer it loaded from an illegal memory location. I would suggest you to search the wave diagrams in your simulation for the occurence of the 0xbadacce6 at the data input port of the core. This will allow you to pinpoint the location where the illegal memory access happens. The illlegal instruction error you get is most likely just a consequential error of this illegal bus access. 

Btw: If you are using the pulp-sdk to compile your application you can use 'make dis' instead of the usual 'make all' to emit the annotated disassembly of your source code.

asyms, 2020-04-27:
Thanks for the response. I'm certainly not an expert on debugging disassembly so I really appreciate the help.

Looking at the disassembly I noticed this line in one of the functions where things start going south (sometimes the STDOUT starts printing pure gibberish:

```1c00dfba:	fff90793          	addi	a5,s2,-1 # 7fffffff <pulp__FC+0x80000000>```

`0x7fffffff` is also the status code returned by the core after the "Illegal instruction ... " occurs. What exactly does `<pulp__FC+0x80000000>` mean in this case and is it related to the status code?

FrancescoConti, 2020-04-27:
Hi @asyms , just a tip: you mentioned at the very beginning that you are near the L2 memory size with your data, remember that also your code goes in the same memory. A possible explanation of your behavior is that you could be polluting your code, for example with a dynamically allocated array (or your stack).
Looking at the instruction trace you posted earlier makes this possibility look very likely to me. That's where I would dig into!

asyms, 2020-04-28:
It seems as tho @FrancescoConti was right. I had already moved a lot of locally defined variables to a global definition, but not all. After doing this for the remaining local variables, the issues are gone. Just wondering, how big is the stack usable for local variables?

bluewww, 2020-11-17:
Stack size depends on the linker script options. Normally its  `0x800` bytes.

closed 2020-11-17 by bluewww

prajwalkashyap, 2023-06-19:
@asyms How did you finally fix this issue ?

bluewww, 2023-06-19:
You have to increase the memory size in RTL or make your program use less memory.