pulp-platform/pulpissimo #127: Accelerator HWPE Example Loops on ZedBoard

acastillo18, 2020-04-18:
Right now I am trying to run the accelerator example on the zedboard, but right now I have a issue regarding of running the application. For some reason after this line of code __rt_periph_wait_event(ARCHI_SOC_EVENT_FCHWPE0, 1), the application loops. What can be a good approach to fix it?  
[image]

bluewww, 2020-04-20:
You could try to set a breakpoint in `main` and single step

acastillo18, 2020-04-20:
Hi, I did what you suggest me but still the examples loops.

bluewww, 2020-04-20:
It is hard to tell what is going wrong in your case. If you don't get enough insight running it on an FPGA you can always use a simulator, but I believe single stepping through the assembly should give some insight.

LPLATU, 2020-08-27:
Maybe this will help?
https://github.com/pulp-platform/pulpissimo/issues/104

bluewww, 2021-02-02:
Please open another issue if this is still relevant. `pulp_soc` (https://github.com/pulp-platform/pulp_soc/commit/84aa62fe43df072433ef02c80b61e6cc54c1a95f) has some hwpe updates that will soon propagate into pulpissimo master

closed 2021-02-02 by bluewww