pulp-platform/pulpissimo #128: Run Pulpissimo on Nexys DDR 4?

studentfhdo, 2020-04-22:
Does pulpissimo or pulpino run on the Nexys DDR 4 with an Atrix 7 FPGA? If so are there already projects based on it?

meggiman, 2020-04-27:
So far we do not directly support the Nexys 4 DDR board. However, porting PULPissimo to a new Xilinx based FPGA board is not that hard. In fact, a lot of the FPGA boards we currently support are community contributions. The procedure mostly boils down to creating a new folder in the fpga subdirectory using one of the existing ports as a template and adapting the board specific XDC constraints file to the specifics of your board i.e. the mapping of the PULPissimo IO ports to peripherals and pins on the FPGA board. Pull requests are welcome ;-)

heavySea, 2020-06-12:
I'm currently working on a port for the Nexus 4 modifying the Nexys Video port. 

But I have a problem regarding the JTAG I/O constraints. The Nexys 4 has a FTDI FT2232HQ for USB-UART/JTAG.
In the documentation the connection between the  FT2232 and FPGA I/O bank for JTAG is not mentioned, only for the UART. And after long research I could not find any other information about it. Digilent seems to keep it proprietary.
So I probably need an external JTAG cable if nobody else has an idea.

heavySea, 2020-07-04:
> Does pulpissimo or pulpino run on the Nexys DDR 4 with an Atrix 7 FPGA? If so are there already projects based on it?

This is now possible with #136

bluewww, 2020-07-04:
Indeed. I'll be closing this.

closed 2020-07-04 by bluewww