pulp-platform/pulpissimo #133: I2C Issues

bbo49, 2020-05-29:
Hi,

I'm trying out the I2S interface of Pulpissimo and there are some things which didn't work as expected, when using the `pulp-rt-examples/tree/master/periph/i2s/pdm_capture` example with the following settings.
```
i2s_conf.frequency = SAMPLING_FREQ;
i2s_conf.dual = 0;
i2s_conf.width = 16;
i2s_conf.id = 0;
i2s_conf.channels = 2;
```

1. The sck frequency is wrong. This is caused by the following line in the driver, as it does not include the number of channels in the calculation of the required frequency:
https://github.com/pulp-platform/pulp-rt/blob/8c4202e89d0b60146d73a84e33737c9285227d9f/drivers_deprecated/i2s/i2s-v2.c#L97
It seems that this has been fixed in v3 of the driver.
https://github.com/pulp-platform/pulp-rt/blob/8c4202e89d0b60146d73a84e33737c9285227d9f/drivers_deprecated/i2s/i2s-v3.c#L99
When I update the driver from version 2 to 3 in the following line, the clock frequency is right.
https://github.com/pulp-platform/pulp-configs/blob/cb500e9de327684168a3f284f5bce3a1d43bdc18/configs/chips/pulpissimo/udma.json#L62

2. When working with `channels=2` the frequency of `WS` is wrong. This because the channels is written to the slave_words configuration of the I2S Slave.
https://github.com/pulp-platform/pulp-rt/blob/8c4202e89d0b60146d73a84e33737c9285227d9f/drivers_deprecated/i2s/i2s-v3.c#L225-L232
This value is picked up by the `udma-i2s/i2s_ws_gen.sv causing it to generate only half of the required ws frequency required in the case of channels=2.
https://github.com/pulp-platform/udma_i2s/blob/407e4c6fb89854aae67ac7f89276f77a775b0e95/rtl/i2s_ws_gen.sv#L63-L78
If I temporary change the slave_words in the driver to 0 for my test case, the frequency for `WS` is ok.

3. The sampling of 'WS' in the `udma_i2s/rtl/i2s_rx_channel.sv` happens on the falling edge of sck. 
https://github.com/pulp-platform/udma_i2s/blob/407e4c6fb89854aae67ac7f89276f77a775b0e95/rtl/i2s_rx_channel.sv#L143-L161
According to the I2S spec 'WS' updates on the falling edge, should the sampling be done on rising edge instead? In the udma_i2s repo there are some branches regarding the VEGA chip which have a lot of updates compared to the master branch. It seems to address the sampling of WS and many others. Do these changes still need to be merged into master?

Thanks !

bluewww, 2020-05-29:
@haugoug Do you know the state of the vega i2s compared to the one we have in PULPissimo?

haugoug, 2020-05-29:
Yes they are really different. What was done in vega was never finished and should not be merged to master. The driver for Pulpissimo is i2s-v2.c, but maybe indeed there are some fixes to do.

bbo49, 2020-05-29:
Thanks for the clarification regarding the vega branch.

If I implement the code on an FPGA platform (Nexys Video in my case) I see that the shift increases by one bit every time I reload the application over JTAG. If I press the reset button between to loads the shift of 1 bit stays stable.

Maybe this has something to do with the clock gates which are implemented differently on the FPGA platform vs the simulation platform. My guess is that the bit counters of the `i2s_ws_gen` and `i2s_rx_channel` are getting out of sync because of the clocks not shutting down completely in the FPGA?

Is there a way to simulate the FPGA behaviour for the clock gates in the simulation platform to check this?

bluewww, 2020-05-29:
I'm not super familiar with the fpga port, but I don't believe there is any clock gating when using the FPGA. See this tech cell
https://github.com/pulp-platform/pulpissimo/blob/master/fpga/pulpissimo-genesys2/rtl/pulp_clock_gating_xilinx.sv

abhiram-padikkat, 2025-01-15:
Hi,
i am trying to integrate an audio codec with udma iis and as in all standard iis interface,it only have a single data line sd. but in our udma iis,we have two data output which i think is dedicated for left and right. anyway,how this two dataline can be fed into audio codec with single input?please help. 
thanks in advance