pulp-platform/pulpissimo #134: JTAG-HS2 Adapter zcu

Elia1996, 2020-06-12:
Hi,
I'm trying to upload PULPissimo hello example on zcu102 board. Why the need for an external JTAG adapter since  J83 and J2 connector support JTAG? 
Since I've only  DLC9LP XILINX adapter I'm trying to use this adapter but where to find configuration file for openOCD?  I searched in openOCD guide and in DLC9LP datasheets but nothing.

Thank you in advance for help!

Elia1996, 2020-06-15:
Having only the DLC9LP XILINX adapter I want to understand if it is possible to use it for programming the zcu102 or not, just to know if I have necessarily to buy the HS2.

bluewww, 2020-06-16:
@olivetreezhao did the port for the zcu102, maybe you have a comment on this?

olivetreezhao, 2020-06-16:
The hardware devices supported by OpenOCD are listed on this page:
http://openocd.org/doc/html/Debug-Adapter-Hardware.html
This page might be out of date. Xilinx DLC5 is mentioned. If DLC9 is a replacement product, I assume it supported as well.  Xilinx forum or tech support should be a good source to confirm this.

Elia1996, 2020-06-17:
Thank you, Xilinx DLC5 is supported but why we strictly need an adapter since connector J83 and J2 of zcu102 board support JTAG?