pulp-platform/pulpissimo #139: Nexys 4 DDR: JTAG not working

EverythingElseWasAlreadyTaken, 2020-07-02:
Hi everyone,

I have some trouble with JTAG on the Nexys 4 DDR.

Pulpissimo is synthesized and flashed with Vivado 2019.2.1.
For the JTAG connection, I'd like to use a Segger J-Link Base and the openocd from the pulp-sdk.

This is my openocd config:

```
interface jlink

reset_config none
adapter_khz 1000

set _CHIPNAME riscv

jtag newtap $_CHIPNAME unknown0 -irlen 5 -expected-id 0x10102001
jtag newtap $_CHIPNAME cpu -irlen 5 -expected-id 0x249511C3

set _TARGETNAME $_CHIPNAME.cpu
target create $_TARGETNAME riscv -chain-position $_TARGETNAME -coreid 0x3e0

gdb_report_data_abort enable
gdb_report_register_access_error enable

riscv set_reset_timeout_sec 120
riscv set_command_timeout_sec 120

# prefer to use sba for system bus access
riscv set_prefer_sba on

# dump jtag chain
scan_chain

init
halt
echo "Ready for Remote Connections"
```

The JTAG connector is connected to pmod A on the Nexys 4 DDR, as described in the README.

Every time I try to run openocd , I get the following error:

```
$ openocd -f openocd-nexys-jlink.cfg

Open On-Chip Debugger 0.10.0+dev-00852-ge6e281197 (2020-04-30-13:34)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : J-Link V11 compiled Aug 14 2019 16:21:09
Info : Hardware version: 11.00
Info : VTarget = 0.000 V
Info : clock speed 1000 kHz
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: riscv.unknown0: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Error: Unsupported DTM version: 15
Info : Listening on port 3333 for gdb connections
Error: Target not examined yet

Error: Unsupported DTM version: 15
```

Maybe someone has an idea what the problem is.

Thanks a lot and best regards.

heavySea, 2020-07-04:
Hi,
I don't have an J-Link at hand and use a HS2 Adapter. But I get a similar error, when the FPGA is not completely configured yet. 
Except for this message:

>  Info : VTarget = 0.000 V

But as soon as the DONE LED is flashing (and LED1) it works for me. (A7 Board)

Maybe this helps: https://github.com/riscv/riscv-openocd/issues/362

EverythingElseWasAlreadyTaken, 2020-07-06:
Thanks a lot for your help.

I got it working now, it seems that I just had some broken jumper wires. :(

closed 2020-07-06 by EverythingElseWasAlreadyTaken

Ahmed1Zaky, 2023-02-07:
Hi @heavySea, 

Could you please elaborate on your setup as I am trying to do a similar thing? Did you connect the HS2 Adatper directly to the PMOD or did you use an interface board similar to this one? 

https://digilent.com/reference/pmod/pmodlvlshft/reference-manual

My thinking is that I can use this interface board and then connect the HS2 adapter to the PMODLVLSHFT. I hope you can help with this!

Thanks!