pulp-platform/pulpissimo #143: Accelerator hwme test example stuck

HarshitSharm-a, 2020-07-21:
Hi,

While running an example accelerator test provided in pulp-rt-examples, simulation stuck after one iteration.

[image]

One the other hand, hello world test executes properly.
[image]

FrancescoConti, 2020-07-21:
I remember that there was some issue with an update to the SDK and to the HWME itself. Unfortunately I don't have time for helping; maybe @anga93 has some code that is working in the current PULPissimo version.
Else, you can have a look at the standalone testbench https://github.com/pulp-platform/hwpe-tb

HarshitSharm-a, 2020-07-21:
Thanks @FrancescoConti for guiding, I will check the link provided.

HarshitSharm-a, 2020-07-24:
I tried running standalone testbench (https://github.com/pulp-platform/hwpe-tb), but still problem persist. 
Simulation gets stuck in loop.
[image]

bluewww, 2020-07-24:
Can you please open an issue in the `hwpe-tb` for this?

abuck97, 2020-08-13:
Hi,
I was facing the same issue before. I fixed it by updating the version of the IP in Pulpissimo. The hwpe-stream and hwpe-ctrl modules were older and had some bugs in them.

Adding the following lines to `ips_list.yml` and then rerunning `update-ips` fixed it for me.
```
hwpe-mac-engine:
  commit: v1.3
  domain: [cluster, soc]
```
The proper fix is probably to update the `ips_list.yml` in the the `pulp_soc` ip. That one is referencing v1.2 which seems to have some bugs (due to hwpe-mac-engine v1.2 referencing the older hwpe-stream and hwpe-ctrl module)

abuck97, 2020-08-13:
Oh, another change I made was in the `hwme.c` file to call `hwme_trigger_job();` **after** `soc_eu_fcEventMask_setEvent(ARCHI_SOC_EVENT_FCHWPE0);`.

I found that the mac-engine finished before it set up the EventMask so the finish signal by the mac-engine was lost forever and the processor was waiting forever for that signal.

msnagar, 2024-04-23:
@abuck97
Can you share your code for MAC-engine?

abuck97, 2024-04-23:
Sorry, this was 4 years ago and I don't know where my code is anymore. Versions might also be different and my workaround irrelevant now?

msnagar, 2024-04-23:
> Oh, another change I made was in the `hwme.c` file to call `hwme_trigger_job();` **after** `soc_eu_fcEventMask_setEvent(ARCHI_SOC_EVENT_FCHWPE0);`.
>
> I found that the mac-engine finished before it [...]

@abuck97 
we were trying to run the HWME with ZedBoard but we were stuck at `soc_eu_fcEventMask_setEvent(ARCHI_SOC_EVENT_FCHWPE0);`. 

We also made the changes as you suggested above. 
However, it wouldn't work on PULP-SDK or FPGA.