pulp-platform/pulpissimo #145: RTL simulation for ZERORISCY failing

Risto97, 2020-07-22:
I am trying to run simulation for ZERORISCY (IBEX?) instead of RISCY CPU.
When I follow the steps from the repo and run RISCY core it works, verified with Hello and GPIO output example.
However when I change CORE_TYPE in pulpissimo/rtl/tb/tb_pulp.sv to 1, to run ZERORISCY core it is failing.
Is there anything else that I need to change?

The messages that I get when I run the pulp-rt-examples/hello 
`make clean all run`

```
# //  Questa Sim-64
# //  Version 2019.2_2 linux_x86_64 Jun 15 2019
....
# START_ADDR[0] = 0xxxxxxxxx; END_ADDR[0] = 0Xxxxxxxxx
# START_ADDR[1] = 0xxxxxxxxx; END_ADDR[1] = 0Xxxxxxxxx
# [TB]       0ns - Entry point is set to 0x1c008080
# [TB]       0ns - Asserting hard reset
# ** Warning: (vsim-3533) [FOFIW] - Failed to open file "fs/file_0_0.txt" for writing.
# No such file or directory. (errno = ENOENT)    : /.../pulp/pulpissimo/sim/../rtl/tb/tb_fs_handler.sv(269)
#    Time: 0 ps  Iteration: 0  Instance: /tb_pulp/i_fs_handler/FS_PRINTF_CLUSTER
# ** Warning: (vsim-3533) [FOFIW] - Failed to open file "fs/file_31_0.txt" for writing.
# No such file or directory. (errno = ENOENT)    : /.../pulp/pulpissimo/sim/../rtl/tb/tb_fs_handler.sv(269)
#    Time: 0 ps  Iteration: 0  Instance: /tb_pulp/i_fs_handler/FS_PRINTF_FC
# [TB]       1ns - Using FLL
# [TB]       1ns - Not using CAM SDVT
# Loading default stimuli
# [JTAG] SoftReset Done(    701ns)
# [JTAG] Bypass Test Passed (  33301ns)
# [JTAG] Tap ID: 249511c3 (  43701ns)
# [JTAG] Tap ID Test PASSED (  43701ns)
# [test_mode_if]   50301ns - Init
# [TB]   50301ns - Enabling clock out via jtag
# [TB]   50301ns - Releasing hard reset
# [TB]   51901ns - Init PULP TAP
#   53385ns: Illegal instruction (hart 3e0) at PC 0x1a0002e0: 0xf647b7b3
#   53840ns: Illegal instruction (hart 3e0) at PC 0x1a0002e0: 0xf647b7b3
...................................
#   65661ns: Illegal instruction (hart 3e0) at PC 0x1a0002e0: 0xf647b7b3
# [pulp_tap_if] WRITE32 burst @1c008080 for           4 bytes.
# [TB]   66001ns - Write32 PULP TAP
#   66116ns: Illegal instruction (hart 3e0) at PC 0x1a0002e0: 0xf647b7b3
...................................
#  129768ns: Illegal instruction (hart 3e0) at PC 0x1a0002e0: 0xf647b7b3
#  130222ns: Illegal instruction (hart 3e0) at PC 0x1a0002e0: 0xf647b7b3
# [JTAG] R/W test of L2 succeeded
#  130677ns: Illegal instruction (hart 3e0) at PC 0x1a0002e0: 0xf647b7b3
............................
#  176143ns: Illegal instruction (hart 3e0) at PC 0x1a0002e0: 0xf647b7b3
# [TB]  176201ns - Halting the Core
#  176597ns: Illegal instruction (hart 3e0) at PC 0x1a0002e0: 0xf647b7b3
.................................
#  195693ns: Illegal instruction (hart 3e0) at PC 0x1a0002e0: 0xf647b7b3
# [TB]  235001ns - Writing the boot address into dpc
# [TB]  279101ns - Loading L2
# [JTAG] Loading L2 with pulp tap jtag interface
# [pulp_tap_if] WRITE32 burst @1c000000 for        1024 bytes.
# [pulp_tap_if] WRITE32 burst @1c000400 for        1024 bytes.
# [pulp_tap_if] WRITE32 burst @1c000800 for        1024 bytes.
# [pulp_tap_if] WRITE32 burst @1c000c00 for         688 bytes.
# [pulp_tap_if] WRITE32 burst @1c008000 for        1024 bytes.
# [pulp_tap_if] WRITE32 burst @1c008400 for        1024 bytes.
# [pulp_tap_if] WRITE32 burst @1c008800 for        1024 bytes.
# Break key hit
# Simulation stop requested.
# End time: 20:37:55 on Jul 22,2020, Elapsed time: 0:00:22
# Errors: 0, Warnings: 18
```

Illegal instruction message is repeating where there are dots in the report ( I shortened it).
I would guess that there is a problem with the compiler possibly, I compiled the compiler with this command:
`./configure --prefix=/scratch/rpejasin/tmp/rv32imc --with-arch=rv32imc --with-cmodel=medlow --enable-multilib`

Output of this command
`riscv32-unknown-elf-gcc --print-multi-lib`
Is:
```
.;
rv32imfcxpulpv2/ilp32;@march=rv32imfcxpulpv2@mabi=ilp32
rv32imfcxpulpv2/ilp32f;@march=rv32imfcxpulpv2@mabi=ilp32f
rv32imcxpulpslim/ilp32;@march=rv32imcxpulpslim@mabi=ilp32
rv32ic/ilp32/mreg16;@march=rv32ic@mabi=ilp32@mreg=16
rv32imcxgap8/ilp32;@march=rv32imcxgap8@mabi=ilp32
rv32imcxgap9/ilp32;@march=rv32imcxgap9@mabi=ilp32
```

I find it a bit confusing because I guess that ZERORISCY is Ibex now? Should I go to lowrisc repo for simulation?

bluewww, 2020-07-22:
The zeroriscy configuration is not really maintained anymore. Did you use ibex or is it really zeroriscy?

Risto97, 2020-07-22:
Thank you for the answer.

Well I don't really know the difference really. I guess zeroriscy is legacy and ibex is version that is active?
Only thing that I did is to change parameter in testbench, but I guess that is not enough.
And there is no documentation on how to run it.
Is there any other SoC platform that you recommend for trying simpler MCU cores (if PULP doesn't maintain other cores)? Litex?

bluewww, 2020-07-22:
ibex is the continuation of zeroriscy and now maintained by a company named lowrisc.

Regarding the issue you reported it looks like there is a problem with the entry point. Ill look into it

closed 2020-10-19 by bluewww