pulp-platform/pulpissimo #146: Error when I build RTL simulation platform

yywang1993, 2020-07-24:
Recently I want to use pulpissimo in my CentOS6.8
At first, I download the V4.0 release about pulpissimo and v1.0.16-pulp-riscv-gcc-centos-7(could this used in CentOS6.8? )
Then I **source sourceme.sh** in v1.0.16-pulp-riscv-gcc-centos-7 dir,
According to README, I download pulp-runtime(simple runtime), and **source configs/pulpissimo.sh**.
Then I change to pulpissimo-pulpissimov4 dir, **./update-ips**, **./generate-scripts**, **source setup/vsim.sh**.
Finlay, when I execute **make build**, I got error about this:

 Makefile:30: warning: overriding commands for target `/home/pulpissimo/pulpissimo-pulpissimov4/install/tcl_files/config/vsim_ips.tcl'
Makefile:30: warning: ignoring old commands for target `/home/pulpissimo/pulpissimo-pulpissimov4/install/tcl_files/config/vsim_ips.tcl'
make -C rtl/tb/remote_bitbang all
make[1]: Entering directory `/home/pulpissimo/pulpissimo-pulpissimov4/rtl/tb/remote_bitbang'
cc -MT remote_bitbang.o -MMD -MP -MF ./.d/remote_bitbang.Td -std=gnu11 -fno-strict-aliasing -Wall -Wextra -Wno-missing-field-initializers -Wno-unused-function -Wno-missing-braces -O2 -g -march=native -DENABLE_LOGGING -DNDEBUG -fPIC -I./  \
		-c  remote_bitbang.c -o remote_bitbang.o 
cc1: error: unrecognized command line option "-std=gnu11"
make[1]: *** [remote_bitbang.o] Error 1
make[1]: Leaving directory `/home/pulpissimo/pulpissimo-pulpissimov4/rtl/tb/remote_bitbang'
make: *** [build] Error 2

Did I miss any steps?

bluewww, 2020-07-24:
You are using CentOS6 which we don't support, minimum CentOS version required is 7. Your gcc version is probably really old: It doesn't recognized the `-std=gnu11` argument which is available in gcc since roughly 2011...

One workaround is to point the `CC` environment variable to a newer GCC version, this *might* work.

Can you paste the output of `gcc --version`?

yywang1993, 2020-07-24:
The version is 4.4.7, just like below. Any way, thanks for your reply, I will change to CentOS7 and try again.
gcc (GCC) 4.4.7 20120313 (Red Hat 4.4.7-23)
Copyright (C) 2010 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

closed 2020-07-26 by bluewww