pulp-platform/pulpissimo #148: Syntax error in buliding the RTL simulation platform

First-Killer, 2020-07-27:
I've looked at problems similar to mine, but I still have no idea how to solve them. After executing “make build”, I get the following error. My vlog - version is questasim vlog 10.7c compiler 2018.08 Aug 17 2018.

error:
[image]
Please  help me!Thanks!

First-Killer, 2020-07-27:
Building pulp_soc/pulp_soc
** Error: ../ips/pulp_soc/rtl/pulp_soc/pulp_soc.sv(16): (vlog-13006) Could not find the package (dm).  Design read will continue, but expect a cascade of errors after this failure.  Furthermore if you experience a vopt-7 error immediately before this error then please check the package names or the library search paths on the command line.
** Error: ../ips/pulp_soc/rtl/pulp_soc/pulp_soc.sv(289): (vlog-2164) Class or package 'dm' not found.
** Error: ../ips/pulp_soc/rtl/pulp_soc/pulp_soc.sv(289): (vlog-2900) The localparam 'dm' is declared without a default value.
** Error: (vlog-13069) ../ips/pulp_soc/rtl/pulp_soc/pulp_soc.sv(289): near "::": syntax error, unexpected ::, expecting ';' or ','.
** Error: ../ips/pulp_soc/rtl/pulp_soc/pulp_soc.sv(294): (vlog-2164) Class or package 'dm' not found.
** Error: ../ips/pulp_soc/rtl/pulp_soc/pulp_soc.sv(295): (vlog-2164) Class or package 'dm' not found.
** Error: ../ips/pulp_soc/rtl/pulp_soc/pulp_soc.sv(298): (vlog-2164) Class or package 'dm' not found.
** Error: ../ips/pulp_soc/rtl/pulp_soc/pulp_soc.sv(298): (vlog-13205) Syntax error found in the scope following 'dm'. Is there a missing '::'?
/opt/riscv/pulp/sim/vcompile//ips/pulp_soc.mk:43: recipe for target '/opt/riscv/pulp/install/modelsim_libs/pulp_soc_lib/pulp_soc.vmake' failed
make[3]: *** [/opt/riscv/pulp/install/modelsim_libs/pulp_soc_lib/pulp_soc.vmake] Error 2
/opt/riscv/pulp/sim//vcompile/ips.mk:14: recipe for target 'build' failed
make[2]: *** [build] Error 2
Makefile:40: recipe for target 'build' failed
make[1]: *** [build] Error 2
make[1]: Leaving directory '/opt/riscv/pulp/sim'
Makefile:44: recipe for target 'build' failed
make: *** [build] Error 2

First-Killer, 2020-07-27:
error:
Building pulp_soc/pulp_soc
** Error: ../ips/pulp_soc/rtl/pulp_soc/pulp_soc.sv(16): (vlog-13006) Could not find the package (dm).  Design read will continue, but expect a cascade of errors after this failure.  Furthermore if you experience a vopt-7 error immediately before this error then please check the package names or the library search paths on the command line.
** Error: ../ips/pulp_soc/rtl/pulp_soc/pulp_soc.sv(289): (vlog-2164) Class or package 'dm' not found.
** Error: ../ips/pulp_soc/rtl/pulp_soc/pulp_soc.sv(289): (vlog-2900) The localparam 'dm' is declared without a default value.
** Error: (vlog-13069) ../ips/pulp_soc/rtl/pulp_soc/pulp_soc.sv(289): near "::": syntax error, unexpected ::, expecting ';' or ','.
** Error: ../ips/pulp_soc/rtl/pulp_soc/pulp_soc.sv(294): (vlog-2164) Class or package 'dm' not found.
** Error: ../ips/pulp_soc/rtl/pulp_soc/pulp_soc.sv(295): (vlog-2164) Class or package 'dm' not found.
** Error: ../ips/pulp_soc/rtl/pulp_soc/pulp_soc.sv(298): (vlog-2164) Class or package 'dm' not found.
** Error: ../ips/pulp_soc/rtl/pulp_soc/pulp_soc.sv(298): (vlog-13205) Syntax error found in the scope following 'dm'. Is there a missing '::'?
/opt/riscv/pulp/sim/vcompile//ips/pulp_soc.mk:43: recipe for target '/opt/riscv/pulp/install/modelsim_libs/pulp_soc_lib/pulp_soc.vmake' failed
make[3]: *** [/opt/riscv/pulp/install/modelsim_libs/pulp_soc_lib/pulp_soc.vmake] Error 2
/opt/riscv/pulp/sim//vcompile/ips.mk:14: recipe for target 'build' failed
make[2]: *** [build] Error 2
Makefile:40: recipe for target 'build' failed
make[1]: *** [build] Error 2
make[1]: Leaving directory '/opt/riscv/pulp/sim'
Makefile:44: recipe for target 'build' failed
make: *** [build] Error 2

meggiman, 2020-07-27:
Did you checkout out the IP dependencies before trying to build the RTL?
You can do this with the 'checkout' make target or by executing the update-ips script.

First-Killer, 2020-07-27:
> Did you checkout out the IP dependencies before trying to build the RTL?
> You can do this with the 'checkout' make target or by executing the update-ips script.

Thanks for your comment.Yeah ,I have execute "./update-ips" before the next work,and I got the messages as follows.

> Retrieving ips_list.yml dependency list for all IPs (may take some time)...
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/pulp_soc/v1.1.0/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/pulp_cluster/pulp_cluster_v1.0/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/tbtools/v0.1/ips_list.yml
Generated IP dependency tree.

Updating ip 'pulp_soc'...
M	CHANGELOG.md
M	LICENSE
M	README.md
M	ips_list.yml
M	rtl/.editorconfig
M	rtl/components/apb_clkdiv.sv
M	rtl/components/apb_dummy.sv
M	rtl/components/apb_soc_ctrl.sv
M	rtl/components/apb_timer_unit.sv
M	rtl/components/axi_slice_dc_master_wrap.sv
M	rtl/components/axi_slice_dc_slave_wrap.sv
M	rtl/components/freq_meter.sv
M	rtl/components/glitch_free_clk_mux.sv
M	rtl/components/glitch_free_clk_mux_xilinx.sv
M	rtl/components/memory_models.sv
M	rtl/components/pulp_interfaces.sv
M	rtl/components/scm_2048x32.sv
M	rtl/components/scm_512x32.sv
M	rtl/components/src_files.yml
M	rtl/components/tcdm_arbiter_2x1.sv
M	rtl/fc/fc_demux.sv
M	rtl/fc/fc_hwpe.sv
M	rtl/fc/fc_subsystem.sv
M	rtl/fc/src_files.yml
M	rtl/pulp_soc/boot_rom.sv
M	rtl/pulp_soc/l2_ram_multi_bank.sv
M	rtl/pulp_soc/lint_jtag_wrap.sv
M	rtl/pulp_soc/periph_bus_wrap.sv
M	rtl/pulp_soc/pulp_soc.sv
M	rtl/pulp_soc/soc_clk_rst_gen.sv
M	rtl/pulp_soc/soc_event_arbiter.sv
M	rtl/pulp_soc/soc_event_generator.sv
M	rtl/pulp_soc/soc_event_queue.sv
M	rtl/pulp_soc/soc_interconnect.sv
M	rtl/pulp_soc/soc_interconnect_wrap.sv
M	rtl/pulp_soc/soc_peripherals.sv
M	rtl/pulp_soc/src_files.yml
M	rtl/pulpemu/pulpemu.sv
M	rtl/pulpemu/pulpemu_apb_demux.sv
M	rtl/pulpemu/pulpemu_clk_gating.sv
M	rtl/pulpemu/pulpemu_gpio.sv
M	rtl/pulpemu/pulpemu_i2c.sv
M	rtl/pulpemu/pulpemu_spi_master.sv
M	rtl/pulpemu/pulpemu_spi_slave.sv
M	rtl/pulpemu/pulpemu_stdout.sv
M	rtl/pulpemu/pulpemu_trace.sv
M	rtl/pulpemu/pulpemu_uart.sv
M	rtl/pulpemu/pulpemu_zynq2pulp_gpio.sv
M	rtl/pulpemu/src_files.yml
M	rtl/udma_subsystem/src_files.yml
M	rtl/udma_subsystem/udma_subsystem.sv
M	src_files.yml
HEAD 目前位于 86e3a58... Update CHANGELOG.md for v1.1.0

Updating ip 'pulp_cluster'...
M	LICENSE
M	ips_list.yml
M	packages/apu_package.sv
M	packages/pulp_cluster_package.sv
M	rtl/axi2mem_wrap.sv
M	rtl/axi2per_wrap.sv
M	rtl/axi_slice_wrap.sv
M	rtl/cluster_bus_wrap.sv
M	rtl/cluster_clock_gate.sv
M	rtl/cluster_event_map.sv
M	rtl/cluster_interconnect_wrap.sv
M	rtl/cluster_peripherals.sv
M	rtl/cluster_timer_wrap.sv
M	rtl/core_demux.sv
M	rtl/core_region.sv
M	rtl/dmac_wrap.sv
M	rtl/hwpe_subsystem.sv
M	rtl/per2axi_wrap.sv
M	rtl/per_demux_wrap.sv
M	rtl/periph_FIFO.sv
M	rtl/periph_demux.sv
M	rtl/pulp_cluster.sv
M	rtl/tcdm_banks_wrap.sv
M	src_files.yml
HEAD 目前位于 c085039... Merge pull request #10 from pulp-platform/alignment

Updating ip 'tbtools'...
M	dpi_models/dpi_models.sv
M	src_files.yml
M	tb_driver/tb_driver.sv
HEAD 目前位于 ab82317... Added missing return value for dpi_trace_new

SUMMARY
IPs updated successfully!
Generated new scripts for IPs!

yywang1993, 2020-07-30:
please look the end of #149 ， I had a same problem with you before. But after I re-downloaded pulpissimo and run the scripts, I got the same problem. In addition, when I run make checkout, I got error... What should I do now?

First-Killer, 2020-07-30:
> please look the end of #149 ， I had a same problem with you before. But after I re-downloaded pulpissimo and run the scripts, I got the same problem. In addition, when I run make checkout, I got error... What should I do now?

Sorry, I haven't solved this problem.If you still have the same problem as me,and if conditions permit,please try the questasim's version of 10.3e.The author said in other posts that this version works normally.

bluewww, 2020-07-30:
Sorry to clarify: We are using questasim `10.7b_1` ( 2018.07) as default, but newer versions such as questasim `2020.1` also work. Also you should just run
```bash
git clone https://github.com/pulp-platform/pulpissimo/
./update-ips
make build
```
I can't reproduce the error you guys get.

First-Killer, 2020-07-30:
> Sorry to clarify: We are using questasim `10.7b_1` ( 2018.07) as default, but newer versions such as questasim `2020.1` also work. Also you should just run
> 
> ```shell
> git clone https://github.com/pulp-platform/pulpissimo/
> ./update-ips
> make build
> ```
> 
> I can't reproduce the error you guys get.

Okay，thank you.

zyong1234, 2020-08-19:
> Sorry to clarify: We are using questasim `10.7b_1` ( 2018.07) as default, but newer versions such as questasim `2020.1` also work. Also you should just run
> 
> ```shell
> git clone https://github.com/pulp-platform/pulpissimo/
> ./update-ips
> make build
> ```
> 
> I can't reproduce the error you guys get.

The variable "DEFAULT_SERVER" is set to "https://github.com" by default in ipstools_cfg.py,  and it should be changed to "git@github.com". Otherwise, this problem will arise. Because only the two ips "pulp soc" and "tbtools" are updated, the others are ignored.

FrancescoConti, 2020-08-19:
> > Sorry to clarify: We are using questasim `10.7b_1` ( 2018.07) as default, but newer versions such as questasim `2020.1` also work. Also you should just run
> > ```shell
> > git clone https://github.com/pulp-platform/pulpissimo/
> > ./update-ips
> > make build
> > ```
> > 
> > 
> > I can't reproduce the error you guys get.
> 
> The variable "DEFAULT_SERVER" is set to "https://github.com" by default in ipstools_cfg.py, and it should be changed to "[git@github.com](mailto:git@github.com)". Otherwise, this problem will arise. Because only the two ips "pulp soc" and "tbtools" are updated, the others are ignored.

Hi @zyong1234, the default is set to HTTPS because a user without a GitHub account must be able to download the IPs, and that would not be possible using SSH.

zyong1234, 2020-08-19:
> Hi @zyong1234, the default is set to HTTPS because a user without a GitHub account must be able to download the IPs, and that would not be possible using SSH.

Thanks for your answer! I am a new user of SSH. After i set the variable "PULP_GITHUB_SSH" to "1", and add my SSH key to my GitHub account, but without change the "DEFAULT_SERVER" to "git@github.com", then i couldn't download the IPs. Maybe it is also the reason why others can't download the IPs.

yywang1993, 2020-08-20:
> > > > Sorry to clarify: We are using questasim `10.7b_1` ( 2018.07) as default, but newer versions such as questasim `2020.1` also work. Also you should just run
> > > > ```shell
> > > > git clone https://github.com/pulp-platform/pulpissimo/
> > > > ./update-ips
> > > > make build
> > > > ```
> > > > 
> > > > 
> > > > I can't reproduce the error you guys get.
> > > 
> > > 
> > > The variable "DEFAULT_SERVER" is set to "https://github.com" by default in ipstools_cfg.py, and it should be changed to "[git@github.com](mailto:git@github.com)". Otherwise, this problem will arise. Because only the two ips "pulp soc" and "tbtools" are updated, the others are ignored.
> > 
> > 
> > Hi @zyong1234, the default is set to HTTPS because a user without a GitHub account must be able to download the IPs, and that would not be possible using SSH.
> 
> Thanks for your answer! I am a new user of SSH. After i set the variable "PULP_GITHUB_SSH" to "1", and add my SSH key to my GitHub account, but without change the "DEFAULT_SERVER" to "[git@github.com](mailto:git@github.com)", then i couldn't download the IPs. Maybe it is also the reason why others can't download the IPs.

Great. Fllowing your method, finally, i download all ips.

yywang1993, 2020-08-20:
It seems not all ips...
[root@localhost pulpissimo]# ./update-ips 
Retrieving ips_list.yml dependency list for all IPs (may take some time)...
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/pulp_soc/v1.2.0/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/L2_tcdm_hybrid_interco/pulpissimo-v1.0/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/adv_dbg_if/v0.0.1/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/apb2per/v0.0.1/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/apb_adv_timer/v1.0.2/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/apb_fll_if/pulpissimo-v1.0/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/apb_gpio/v0.2.0/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/apb_node/v0.1.1/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/apb_interrupt_cntrl/v0.0.1/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/axi/v0.7.1/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/axi_node/v1.1.4/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/axi_slice/v1.1.4/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/axi_slice_dc/v1.1.3/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/timer_unit/v1.0.2/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/common_cells/v1.13.1/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/fpnew/v0.6.1/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/common_cells/v1.13.1/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/fpu_div_sqrt_mvp/v1.0.3/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/jtag_pulp/v0.1/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/riscv/pulpissimo-v3.4.0/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/lowRISC/ibex/e65a27e14dda7bb3b09cadc4e9232305fd451cec/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/scm/v1.0.1/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/generic_FLL/v0.1/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/tech_cells_generic/v0.1.6/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/udma_core/v1.0.0/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/udma_uart/v1.0.0/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/udma_i2c/vega_v1.0.0/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/udma_i2s/v1.0.0/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/udma_qspi/v1.0.0/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/udma_sdio/vega_v1.0.5/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/udma_camera/v1.0.0/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/udma_filter/v1.0.0/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/udma_external_per/v1.0.0/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/hwpe-mac-engine/v1.2/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/riscv-dbg/v0.2/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/tbtools/v0.1/ips_list.yml
Generated IP dependency tree.

After downloading is done, then I run **make build**, there is still miss some files...
make[2]: Leaving directory `/root/pulpissimo/pulpissimo/rtl/tb/remote_bitbang'
  Building L2_tcdm_hybrid_interco/soc_interconnect
  Building adv_dbg_if/adv_dbg_if
  Building apb2per/apb2per
  Building apb_adv_timer/apb_adv_timer
  Building apb_fll_if/apb_fll_if
  Building apb_gpio/apb_gpio
  Building apb_node/apb_node
  Building apb_interrupt_cntrl/apb_interrupt_cntrl
  Building axi/axi
  Building axi/axi_sim
  Building common_cells/common_cells_all
  Building axi_node/axi_node
  Building axi_slice/axi_slice
  Building axi_slice_dc/axi_slice_dc
  Building timer_unit/timer_unit
  Building fpu_div_sqrt_mvp/div_sqrt_top_mvp
  Building fpnew/fpnew
  Building jtag_pulp/jtag_pulp
  Building riscv/riscv_regfile_rtl
  Building riscv/riscv
  Building riscv/riscv_vip_rtl
  Building riscv/tb_riscv
  Building ibex/ibex
  Building ibex/ibex_vip_rtl
** Warning: ../ips/ibex/rtl/ibex_tracer.sv(907): (vlog-2584) there is a matching case item hence not unique
  Building ibex/ibex_regfile_rtl
  Building scm/scm
[... 18 lines cut ...]
** Error: ../ips/hwpe-mac-engine/rtl/mac_fsm.sv(17): (vlib-13006) Could not find the package (hwpe_ctrl_package).  Design read will continue, but expect a cascade of errors after this failure.  Furthermore if you experie [... cut ...]
** Error: ../ips/hwpe-mac-engine/wrap/mac_top_wrap.sv(10): (vlib-13006) Could not find the package (hwpe_ctrl_package).  Design read will continue, but expect a cascade of errors after this failure.  Furthermore if you experience a vopt-7 error immediately before this error then please check the package names or the library search paths on the command line.
make[3]: *** [/root/pulpissimo/pulpissimo/install/modelsim_libs/hwpe_mac_engine_lib/hw-mac-engine.vmake] Error 2
make[2]: *** [build] Error 2
make[1]: *** [build] Error 2
make[1]: Leaving directory `/root/pulpissimo/pulpissimo/sim'
make: *** [build] Error 2

yywang1993, 2020-08-20:
It seems miss _pulp-platform/hwpe-stream_...

zyong1234, 2020-08-20:
> It seems not all ips...
> [root@localhost pulpissimo]# ./update-ips
> Retrieving ips_list.yml dependency list for all IPs (may take some time)...
> Fetching ips_list.yml from [...]

Please try to run "./update-ips" again

FrancescoConti, 2020-08-20:
Please post the full log of the update-ips command, after `Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/tbtools/v0.1/ips_list.yml
Generated IP dependency tree.`

zyong1234, 2020-08-20:
> It seems miss _**pulp-platform/hwpe-stream**_...
After rerun the "./update-ips" command, the rest ips should be downloaded.

yywang1993, 2020-08-20:
> Please try to run "./update-ips" again

It doesn't work, I got an error...
[root@localhost pulpissimo]# ./update-ips 
Retrieving ips_list.yml dependency list for all IPs (may take some time)...
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/pulp_soc/v1.2.0/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/L2_tcdm_hybrid_interco/pulpissimo-v1.0/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/adv_dbg_if/v0.0.1/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/apb2per/v0.0.1/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/apb_adv_timer/v1.0.2/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/apb_fll_if/pulpissimo-v1.0/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/apb_gpio/v0.2.0/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/apb_node/v0.1.1/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/apb_interrupt_cntrl/v0.0.1/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/axi/v0.7.1/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/axi_node/v1.1.4/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/common_cells/v1.13.1/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/axi/v0.7.1/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/axi_slice/v1.1.4/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/axi_slice_dc/v1.1.3/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/timer_unit/v1.0.2/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/common_cells/v1.13.1/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/fpnew/v0.6.1/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/jtag_pulp/v0.1/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/riscv/pulpissimo-v3.4.0/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/lowRISC/ibex/e65a27e14dda7bb3b09cadc4e9232305fd451cec/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/scm/v1.0.1/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/generic_FLL/v0.1/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/tech_cells_generic/v0.1.6/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/udma_core/v1.0.0/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/udma_uart/v1.0.0/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/udma_i2c/vega_v1.0.0/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/udma_i2s/v1.0.0/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/udma_qspi/v1.0.0/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/udma_sdio/vega_v1.0.5/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/udma_camera/v1.0.0/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/udma_filter/v1.0.0/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/udma_external_per/v1.0.0/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/hwpe-mac-engine/v1.2/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/riscv-dbg/v0.2/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/tbtools/v0.1/ips_list.yml
Generated IP dependency tree.

Updating ip 'L2_tcdm_hybrid_interco'...
HEAD is now at b55055b... initial commit

Updating ip 'adv_dbg_if'...
HEAD is now at 670fef7... Add parameters in IP yml description for JG EDA tool

Updating ip 'apb2per'...
HEAD is now at 825332b... Add parameters in IP yml description for JG EDA tool

Updating ip 'apb_adv_timer'...
HEAD is now at 16dd65c... Add parameters in IP yml description for JG EDA tool

Updating ip 'apb_fll_if'...
Traceback (most recent call last):
  File "./update-ips", line 47, in <module>
    ipdb.update_ips()
  File "ipstools/ipstools/IPDatabase.py", line 525, in update_ips
ValueError: invalid literal for int() with base 10: b''

yywang1993, 2020-08-20:
> Please post the full log of the update-ips command, after `Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/tbtools/v0.1/ips_list.yml Generated IP dependency tree.`

[root@localhost pulpissimo]# ./update-ips 
Cloning into 'ipstools'...
remote: Enumerating objects: 945, done.
remote: Total 945 (delta 0), reused 0 (delta 0), pack-reused 945
Receiving objects: 100% (945/945), 172.84 KiB | 17.00 KiB/s, done.
Resolving deltas: 100% (661/661), done.
Retrieving ips_list.yml dependency list for all IPs (may take some time)...
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/pulp_soc/v1.2.0/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/L2_tcdm_hybrid_interco/pulpissimo-v1.0/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/adv_dbg_if/v0.0.1/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/apb2per/v0.0.1/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/apb_adv_timer/v1.0.2/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/apb_fll_if/pulpissimo-v1.0/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/apb_gpio/v0.2.0/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/apb_node/v0.1.1/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/apb_interrupt_cntrl/v0.0.1/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/axi/v0.7.1/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/axi_node/v1.1.4/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/common_cells/v1.13.1/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/axi/v0.7.1/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/axi_slice/v1.1.4/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/axi_slice_dc/v1.1.3/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/timer_unit/v1.0.2/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/common_cells/v1.13.1/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/fpnew/v0.6.1/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/common_cells/v1.13.1/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/fpu_div_sqrt_mvp/v1.0.3/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/jtag_pulp/v0.1/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/riscv/pulpissimo-v3.4.0/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/lowRISC/ibex/e65a27e14dda7bb3b09cadc4e9232305fd451cec/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/scm/v1.0.1/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/generic_FLL/v0.1/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/tech_cells_generic/v0.1.6/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/udma_core/v1.0.0/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/udma_uart/v1.0.0/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/udma_i2c/vega_v1.0.0/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/udma_i2s/v1.0.0/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/udma_qspi/v1.0.0/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/udma_sdio/vega_v1.0.5/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/udma_camera/v1.0.0/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/udma_filter/v1.0.0/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/udma_external_per/v1.0.0/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/hwpe-mac-engine/v1.2/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/riscv-dbg/v0.2/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/tbtools/v0.1/ips_list.yml
Generated IP dependency tree.

Cloning ip 'L2_tcdm_hybrid_interco'...
Cloning into 'L2_tcdm_hybrid_interco'...
Warning: Permanently added the RSA host key for IP address '13.229.188.59' to the list of known hosts.
remote: Enumerating objects: 165, done.
remote: Total 165 (delta 0), reused 0 (delta 0), pack-reused 165
Receiving objects: 100% (165/165), 275.48 KiB | 25.00 KiB/s, done.
Resolving deltas: 100% (90/90), done.
Note: checking out 'pulpissimo-v1.0'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

[... cut ...]
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b new_branch_name

HEAD is now at ab82317... Added missing return value for dpi_trace_new

SUMMARY
IPs updated successfully!
Generated new scripts for IPs!

This is the first download of ips. Should I build sdk before download ips?

FrancescoConti, 2020-08-20:
No the SDK has nothing to do with this. It looks like something failed in the IP dep tree generation for hwpe-stream/hwpe-ctrl (which is the only IP not referenced directly from pulp_soc, if I recall). I don't have time to debug properly right now, but a quick workaround should be to add the following inside `ips_list.yml` in the pulpissimo root:
```
hwpe-ctrl:
  commit: ec068eb
hwpe-stream:
  commit: 1a9b04f
```

yywang1993, 2020-08-20:
I already found this issue before. I can only run update-ips once, if i run this script twice, then I will got an error just like below:
...(omit part)
Updating ip 'apb_fll_if'...
Traceback (most recent call last):
File "./update-ips", line 47, in
ipdb.update_ips()
File "ipstools/ipstools/IPDatabase.py", line 525, in update_ips
ValueError: invalid literal for int() with base 10: b''

So I can't rerun **update-ips** after I add this two commit at **ips_list.yml**, this issue is really borthers me...

yywang1993, 2020-08-20:
I had manually downloaded these two commit, put at ips dir, but I still got the same error when I run **make build**...
Is there some filelist used in this process? Or tool will automatic search needed files in ips dir?

zyong1234, 2020-08-20:
> I had manually downloaded these two commit, put at ips dir, but I still got the same error when I run **make build**...
> Is there some filelist used in this process? Or tool will automatic search needed files in ips dir?

I also can't download the two ips "hwpe-stream" and "hwpe-ctrl" at the beginning, and just rerun the command "./update-ips", i downloaded the two ips. As for the error "ValueError: invalid literal for int() with base 10: b'", i also encounter it sometimes, just download the pulpissimo again, and rerun these commands. Because I haven't encountered this problem all the time, I didn't go into the cause of this error. If all kinds of attempts are ineffective, you can read the relevant code to figure out the problem. Good luck!

bluewww, 2020-11-17:
Please re-open if you stil have this issue or open a new issue. Please also post your python version and operating system when you do this.

closed 2020-11-17 by bluewww

luke-avionics, 2022-05-30:
> Hi, I kind of have a similar issue. My `./update-ips` will go through without issues, however, `make build` is failing now with `Could not find the package (cf_math_pkg) error`. This started happening over the weekend when I repeatedly rebuilt the .sv files for simulation purposes.
> 
> I have verified that there is a `cf_math_pkg` present in common_cells:
> 
> ```
> $ ls ips/common_cells/src/cf_math_pkg.sv 
> ips/common_cells/src/cf_math_pkg.sv
> ```
> 
> I am running this on a Ubuntu 20.04 LTS system and a week ago everything worked fine. One thing that I changed last week was that I started using [riscv-gnu-toolchain](https://github.com/riscv-collab/riscv-gnu-toolchain) instead of the included `pulp-riscv-gnu-toolchain` and during this, I recompiled the toolchain a lot to get the desired configuration. Which included running `make clean` in `sim` folder, `./update-ips` etc. I have tried re-configuring the environment by running
> 
> ```
> cd pulp-runtime
> source configs/pulpissimo.sh
> make build-pulp-sdk
> source env/pulpissimo.sh
> make checkout
> source setup/vsim.sh
> make build
> ```
> 
> Nothing seems to work.
> 
> Below I have the `make build` output.
> 
> ```
> [user@localhost]# ./update-ips 
> Retrieving ips_list.yml dependency list for all IPs (may take some time)...
>    Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/pulp_soc/v3.0.0/ips_list.yml
>    Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/common_cells/v1.21.0/ips_list.yml
>    Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/common_verification/v0.2.0/ips_list.yml
>    Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/L2_tcdm_hybrid_interco/v1.0.0/ips_list.yml
>    Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/cluster_interconnect/v1.1.1/ips_list.yml
>    Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/adv_dbg_if/v0.0.2/ips_list.yml
>    Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/apb/v0.1.0/ips_list.yml
>    Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/apb2per/v0.1.0/ips_list.yml
>    Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/apb_adv_timer/v1.0.4/ips_list.yml
>    Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/apb_fll_if/v0.1.3/ips_list.yml
>    Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/apb_gpio/0e9f142f2f11278445c953ad011fce1c7ed85b66/ips_list.yml
>    Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/apb_node/v0.1.1/ips_list.yml
>    Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/apb_interrupt_cntrl/v0.1.1/ips_list.yml
>    Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/axi/v0.29.1/ips_list.yml
>    Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/common_cells/v1.21.0/ips_list.yml
>    Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/common_verification/v0.2.0/ips_list.yml
>    Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/common_verification/v0.2.0/ips_list.yml
>    Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/axi_slice/v1.1.4/ips_list.yml
>    Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/timer_unit/v1.0.2/ips_list.yml
>    Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/fpnew/v0.6.5/ips_list.yml
>    Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/common_cells/v1.20.0/ips_list.yml
>    Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/common_verification/v0.1.1/ips_list.yml
>    Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/fpu_div_sqrt_mvp/v1.0.4/ips_list.yml
>    Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/jtag_pulp/v0.1/ips_list.yml
>    Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/cv32e40p/pulpissimo-v3.4.0-rev3/ips_list.yml
>    Fetching ips_list.yml from https://raw.githubusercontent.com/lowRISC/ibex/pulpissimo-v6.1.1/ips_list.yml
>    Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/scm/v1.0.1/ips_list.yml
>    Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/generic_FLL/1c92dc73a940392182fd4cb7b86f35649b349595/ips_list.yml
>    Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/tech_cells_generic/v0.2.3/ips_list.yml
>    Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/udma_core/v1.1.0/ips_list.yml
>    Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/udma_uart/v1.0.1/ips_list.yml
>    Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/udma_i2c/v1.0.0/ips_list.yml
>    Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/udma_i2s/v1.1.2/ips_list.yml
>    Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/udma_qspi/v1.0.4/ips_list.yml
>    Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/udma_sdio/v1.1.2/ips_list.yml
>    Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/udma_camera/v1.1.2/ips_list.yml
[... cut ...]
> make[3]: *** [/home/user/PULPissimo/pulpissimo/sim/vcompile//ips/axi.mk:63: /home/user/PULPissimo/pulpissimo/install/modelsim_libs/axi_lib/axi.vmake] Error 2
> make[2]: *** [/home/user/PULPissimo/pulpissimo/sim//vcompile/ips.mk:24: build] Error 2
> make[1]: *** [Makefile:50: build] Error 2
> make[1]: Leaving directory '/home/user/PULPissimo/pulpissimo/sim'
> make: *** [Makefile:114: build] Error 2
> 
> [user@localhost]# ls ips/common_cells/src/cf_math_pkg.sv 
> ips/common_cells/src/cf_math_pkg.sv
> ```

I think if you check the src_files.yml of the common_cells, you may find the cf_math_pkg.sv is not actually in the src list ....