pulp-platform/pulpissimo #149: Issues between Prerequisites

yywang1993, 2020-07-27:
I download the tag(4.0) , according the README.md about Prerequisites
**git clone https://github.com/pulp-platform/pulp-builder.git
cd pulp-builder
git checkout 7bd925324fcecae2aad9875f4da45b27d8356796
source configs/pulpissimo.sh
./scripts/clean
./scripts/update-runtime
./scripts/build-runtime
source sdk-setup.sh
source configs/rtl.sh
cd ..**

when I execute **./scripts/build-runtime**, I got this:
...(Leave out a part)
scons: done building targets.
make: Leaving directory `/root/pulpissimo-pulpissimov4/pulp-builder/archi'
+ make -C pulp-debug-bridge all BUILD_DIR=/root/pulpissimo-pulpissimov4/pulp-builder/build/pulp-debug-bridge INSTALL_DIR=/root/pulpissimo-pulpissimov4/pulp-builder/install/ws TARGET_INSTALL_DIR=/root/pulpissimo-pulpissimov4/pulp-builder/install
make: libftdi1-config: Command not found
make: Entering directory `/root/pulpissimo-pulpissimov4/pulp-builder/pulp-debug-bridge'
make: libftdi1-config: Command not found
make: libftdi-config: Command not found
make: libftdi-config: Command not found
make: sdl2-config: Command not found
make: sdl2-config: Command not found
install -D include/debug_bridge/proxy.hpp /root/pulpissimo-pulpissimov4/pulp-builder/install/ws/include/debug_bridge/proxy.hpp
install -D include/debug_bridge/debug_bridge.h /root/pulpissimo-pulpissimov4/pulp-builder/install/ws/include/debug_bridge/debug_bridge.h
install -D bin/plpbridge /root/pulpissimo-pulpissimov4/pulp-builder/install/ws/bin/plpbridge
install -D python/bridge/chips/arnold.py /root/pulpissimo-pulpissimov4/pulp-builder/install/ws/python/bridge/chips/arnold.py
install -D python/bridge/chips/fulmine.py /root/pulpissimo-pulpissimov4/pulp-builder/install/ws/python/bridge/chips/fulmine.py
install -D python/bridge/chips/gap.py /root/pulpissimo-pulpissimov4/pulp-builder/install/ws/python/bridge/chips/gap.py
install -D python/bridge/chips/gap_rev1.py /root/pulpissimo-pulpissimov4/pulp-builder/install/ws/python/bridge/chips/gap_rev1.py
install -D python/bridge/chips/pulpissimo.py /root/pulpissimo-pulpissimov4/pulp-builder/install/ws/python/bridge/chips/pulpissimo.py
install -D python/bridge/chips/pulpissimo_v2.py /root/pulpissimo-pulpissimov4/pulp-builder/install/ws/python/bridge/chips/pulpissimo_v2.py
install -D python/bridge/chips/usoc_v1.py /root/pulpissimo-pulpissimov4/pulp-builder/install/ws/python/bridge/chips/usoc_v1.py
install -D python/bridge/chips/vega.py /root/pulpissimo-pulpissimov4/pulp-builder/install/ws/python/bridge/chips/vega.py
install -D python/bridge/chips/vivosoc3.py /root/pulpissimo-pulpissimov4/pulp-builder/install/ws/python/bridge/chips/vivosoc3.py
install -D python/bridge/chips/wolfe.py /root/pulpissimo-pulpissimov4/pulp-builder/install/ws/python/bridge/chips/wolfe.py
install -D python/bridge/debug_bridge.py /root/pulpissimo-pulpissimov4/pulp-builder/install/ws/python/bridge/debug_bridge.py
install -D python/bridge/default_debug_bridge.py /root/pulpissimo-pulpissimov4/pulp-builder/install/ws/python/bridge/default_debug_bridge.py
install -D python/bridge/jtag.py /root/pulpissimo-pulpissimov4/pulp-builder/install/ws/python/bridge/jtag.py
install -D include/debug_bridge/debug_bridge.h /root/pulpissimo-pulpissimov4/pulp-builder/install/include/debug_bridge/debug_bridge.h
make: libftdi-config: Command not found
make: sdl2-config: Command not found
g++ -O3 -g -fPIC -std=gnu++11 -MMD -MP -Isrc -Iinclude -I/root/pulpissimo-pulpissimov4/pulp-builder/install/ws/include   -o /root/pulpissimo-pulpissimov4/pulp-builder/build/pulp-debug-bridge/src/python_wrapper.o -c src/python_wrapper.cpp
make: libftdi-config: Command not found
make: sdl2-config: Command not found
g++ -O3 -g -fPIC -std=gnu++11 -MMD -MP -Isrc -Iinclude -I/root/pulpissimo-pulpissimov4/pulp-builder/install/ws/include   -o /root/pulpissimo-pulpissimov4/pulp-builder/build/pulp-debug-bridge/src/cables/jtag.o -c src/cables/jtag.cpp
make: libftdi-config: Command not found
make: sdl2-config: Command not found
g++ -O3 -g -fPIC -std=gnu++11 -MMD -MP -Isrc -Iinclude -I/root/pulpissimo-pulpissimov4/pulp-builder/install/ws/include   -o /root/pulpissimo-pulpissimov4/pulp-builder/build/pulp-debug-bridge/src/reqloop.o -c src/reqloop.cpp
make: libftdi-config: Command not found
make: sdl2-config: Command not found
g++ -O3 -g -fPIC -std=gnu++11 -MMD -MP -Isrc -Iinclude -I/root/pulpissimo-pulpissimov4/pulp-builder/install/ws/include   -o /root/pulpissimo-pulpissimov4/pulp-builder/build/pulp-debug-bridge/src/cables/adv_dbg_itf/adv_dbg_itf.o -c src/cables/adv_dbg_itf/adv_dbg_itf.cpp
make: libftdi-config: Command not found
make: sdl2-config: Command not found
g++ -O3 -g -fPIC -std=gnu++11 -MMD -MP -Isrc -Iinclude -I/root/pulpissimo-pulpissimov4/pulp-builder/install/ws/include   -o /root/pulpissimo-pulpissimov4/pulp-builder/build/pulp-debug-bridge/src/gdb-server/gdb-server.o -c src/gdb-server/gdb-server.cpp
make: libftdi-config: Command not found
make: sdl2-config: Command not found
g++ -O3 -g -fPIC -std=gnu++11 -MMD -MP -Isrc -Iinclude -I/root/pulpissimo-pulpissimov4/pulp-builder/install/ws/include   -o /root/pulpissimo-pulpissimov4/pulp-builder/build/pulp-debug-bridge/src/gdb-server/rsp.o -c src/gdb-server/rsp.cpp
make: libftdi-config: Command not found
make: sdl2-config: Command not found
g++ -O3 -g -fPIC -std=gnu++11 -MMD -MP -Isrc -Iinclude -I/root/pulpissimo-pulpissimov4/pulp-builder/install/ws/include   -o /root/pulpissimo-pulpissimov4/pulp-builder/build/pulp-debug-bridge/src/gdb-server/target.o -c src/gdb-server/target.cpp
make: libftdi-config: Command not found
make: sdl2-config: Command not found
g++ -O3 -g -fPIC -std=gnu++11 -MMD -MP -Isrc -Iinclude -I/root/pulpissimo-pulpissimov4/pulp-builder/install/ws/include   -o /root/pulpissimo-pulpissimov4/pulp-builder/build/pulp-debug-bridge/src/gdb-server/breakpoints.o -c src/gdb-server/breakpoints.cpp
make: libftdi-config: Command not found
make: sdl2-config: Command not found
g++ -O3 -g -fPIC -std=gnu++11 -MMD -MP -Isrc -Iinclude -I/root/pulpissimo-pulpissimov4/pulp-builder/install/ws/include   -o /root/pulpissimo-pulpissimov4/pulp-builder/build/pulp-debug-bridge/src/cables/jtag-proxy/jtag-proxy.o -c src/cables/jtag-proxy/jtag-proxy.cpp
make: libftdi-config: Command not found
make: sdl2-config: Command not found
[... 43 lines cut ...]
install -D rules/pulp_opt_vsim.mk /root/pulpissimo-pulpissimov4/pulp-builder/install/rules/pulp_opt_vsim.mk
install -D rules/pulp_properties.mk /root/pulpissimo-pulpissimov4/pulp-builder/install/rules/pulp_properties.mk
install -D rules/pulp_rt.mk /root/pulpissimo-pulpissimov4/pulp-builder/install/rules/pulp_rt.mk
install -D rules/pulp_rules.mk /root/pulpissimo-pulpissimov4/pulp-builder/install/rules/pulp_rules.mk
install -D rules/zephyr.mk /root/pulpissimo-pulpissimov4/pulp-builder/install/rules/zephyr.mk
install -D rules/zephyr/CMakeLists.txt /root/pulpissimo-pulpissimov4/pulp-builder/install/rules/zephyr/CMakeLists.txt
make: Leaving directory `/root/pulpissimo-pulpissimov4/pulp-builder/pulp-rules'
+ make -C pulp-rt header pmsis build install BUILD_DIR=/root/pulpissimo-pulpissimov4/pulp-builder/build/pulp-rt INSTALL_DIR=/root/pulpissimo-pulpissimov4/pulp-builder/install/ws TARGET_INSTALL_DIR=/root/pulpissimo-pulpissimov4/pulp-builder/install
make: Entering directory `/root/pulpissimo-pulpissimov4/pulp-builder/pulp-rt'
/root/pulpissimo-pulpissimov4/pulp-builder/install/rules/pulp_properties.mk:23: /root/pulpissimo-pulpissimov4/pulp-builder/install/ws/rules/pulp_defs.mk: No such file or directory
/root/pulpissimo-pulpissimov4/pulp-builder/install/rules/pulp_properties.mk:32: /props.mk: No such file or directory
/root/pulpissimo-pulpissimov4/pulp-builder/install/rules/pulp_rt.mk:23: /root/pulpissimo-pulpissimov4/pulp-builder/install/ws/rules/tools.mk: No such file or directory
/root/pulpissimo-pulpissimov4/pulp-builder/install/rules/pulp_rt.mk:24: /root/pulpissimo-pulpissimov4/pulp-builder/install/ws/rules/pulp_defs.mk: No such file or directory
/root/pulpissimo-pulpissimov4/pulp-builder/install/rules/pulp_rt.mk:25: /root/pulpissimo-pulpissimov4/pulp-builder/install/ws/rules/pulp_help.mk: No such file or directory
/root/pulpissimo-pulpissimov4/pulp-builder/install/rules/pulp_rt.mk:26: /root/pulpissimo-pulpissimov4/pulp-builder/install/ws/rules/pulp_opt.mk: No such file or directory
plpflags gen --input=pulpissimo@config_file=chips/pulpissimo/pulpissimo.json  --config=**/rt/type=pulp-rt --output-dir= --makefile=/config.mk  --property=fc/archi  --property=pe/archi  --property=pulp_chip  --property=pulp_chip_family   [... cut ...]
There are so many **_Command not found_**, **_No such file or directory_** and end with **_Stop_**.
Did I miss some step caused this? 
I use CentOS7, and already source sourceme.sh in v1.0.16-pulp-riscv-gcc-centos-7 dir.
Also I had install the dependency according pulp-build README.md.

After I finished the Prerequisites, I excute the ./update-ips and ./generate-scripts. I only got components, fc, pulpemu, pulp_soc, udma_subsystem in pulp_soc/rtl dir, where could  I get the rest code used in this paltform?

bluewww, 2020-07-27:
Please use the `master` branch for now, where we fixed this issue. You should then be able to build the sdk with `make build-pulp-sdk`. I'll make a release at some point

closed 2020-07-27 by bluewww

yywang1993, 2020-07-27:
Thanks, according to your advice, I download the master branch, and install the dependencies listed in pulp-sdk.
Then, I followed the README step, when I execute **make build-pulp-sdk**,  I can see there are may files downloaded, but finally I got a fatal error like this:
make[3]: Entering directory `/root/pulpissimo/pulpissimo/pulp-sdk/platform/dpi_models/models'
g++ test/spim_verif/spim_verif.cpp -o /root/pulpissimo/pulpissimo/pulp-sdk/build/sdk/dpi-models/models/spim_tb.so  -MMD -MP -O2 -g -fpic -std=c++11 -I/root/pulpissimo/pulpissimo/pulp-sdk/pkg/sdk/dev/install/ws/include -Werror -Wfatal-errors  -O2 -g -shared -L/root/pulpissimo/pulpissimo/pulp-sdk/pkg/sdk/dev/install/ws/lib -Werror -Wfatal-errors
install -D /root/pulpissimo/pulpissimo/pulp-sdk/build/sdk/dpi-models/models/spim_tb.so /root/pulpissimo/pulpissimo/pulp-sdk/pkg/sdk/dev/install/ws/lib/spim_tb.so
g++ jtag/proxy/proxy.cpp -o /root/pulpissimo/pulpissimo/pulp-sdk/build/sdk/dpi-models/models/jtag_proxy.so  -MMD -MP -O2 -g -fpic -std=c++11 -I/root/pulpissimo/pulpissimo/pulp-sdk/pkg/sdk/dev/install/ws/include -Werror -Wfatal-errors  -O2 -g -shared -L/root/pulpissimo/pulpissimo/pulp-sdk/pkg/sdk/dev/install/ws/lib -Werror -Wfatal-errors
install -D /root/pulpissimo/pulpissimo/pulp-sdk/build/sdk/dpi-models/models/jtag_proxy.so /root/pulpissimo/pulpissimo/pulp-sdk/pkg/sdk/dev/install/ws/lib/jtag_proxy.so
g++ uart/uart/uart.cpp -o /root/pulpissimo/pulpissimo/pulp-sdk/build/sdk/dpi-models/models/uart.so  -MMD -MP -O2 -g -fpic -std=c++11 -I/root/pulpissimo/pulpissimo/pulp-sdk/pkg/sdk/dev/install/ws/include -Werror -Wfatal-errors  -O2 -g -shared -L/root/pulpissimo/pulpissimo/pulp-sdk/pkg/sdk/dev/install/ws/lib -Werror -Wfatal-errors
install -D /root/pulpissimo/pulpissimo/pulp-sdk/build/sdk/dpi-models/models/uart.so /root/pulpissimo/pulpissimo/pulp-sdk/pkg/sdk/dev/install/ws/lib/uart.so
g++ camera/camera.cpp -o /root/pulpissimo/pulpissimo/pulp-sdk/build/sdk/dpi-models/models/camera.so -I/usr/include/GraphicsMagick   -D__MAGICK__ -MMD -MP -O2 -g -fpic -std=c++11 -I/root/pulpissimo/pulpissimo/pulp-sdk/pkg/sdk/dev/install/ws/include -Werror -Wfatal-errors -lGraphicsMagick++ -lGraphicsMagick -O2 -g -shared -L/root/pulpissimo/pulpissimo/pulp-sdk/pkg/sdk/dev/install/ws/lib -Werror -Wfatal-errors
camera/camera.cpp:24:22: fatal error: Magick++.h: No such file or directory
 #include <Magick++.h>
                      ^
compilation terminated.
make[3]: *** [/root/pulpissimo/pulpissimo/pulp-sdk/build/sdk/dpi-models/models/camera.so] Error 1
make[3]: Leaving directory `/root/pulpissimo/pulpissimo/pulp-sdk/platform/dpi_models/models'
make[2]: *** [build] Error 2
make[2]: Leaving directory `/root/pulpissimo/pulpissimo/pulp-sdk/platform/dpi_models'
Reached EOF with exit status 2
FATAL ERROR: the command 'build' has failed
make[1]: *** [all] Error 255
make[1]: Leaving directory `/root/pulpissimo/pulpissimo/pulp-sdk'
make: *** [pulp-sdk] Error 2
 
why could this happened?

bluewww, 2020-07-27:
If you read the error message you see that you don't have imagemagick installed on your machine

yywang1993, 2020-07-27:
I dont think so, as you can see below, the ImageMagick is already installed.
[root@localhost pulpissimo]# convert -v
Version: ImageMagick 6.9.10-68 Q16 x86_64 2020-04-01 https://imagemagick.org
Copyright: © 1999-2019 ImageMagick Studio LLC
License: https://imagemagick.org/script/license.php
Features: Cipher DPC Modules OpenMP(3.1) 
Delegates (built-in): bzlib cairo fontconfig freetype gslib jng jpeg lcms ltdl lzma openexr pangocairo png ps rsvg tiff wmf x xml zlib
Usage: convert [options ...] file [ [options ...] file ...] [options ...] file

bluewww, 2020-07-27:
You need the development package I believe. What OS are you using?

yywang1993, 2020-07-27:
I use CentOS 7, I have installed some dependencies, from pulp-sdk README, pulp-riscv-gnu-toolchain README, just like below:
sudo yum install git python36-pip python36-devel gawk texinfo gmp-devel mpfr-devel libmpc-devel swig libjpeg-turbo-devel redhat-lsb-core doxygen python-sphinx sox GraphicsMagick-devel ImageMagick-devel SDL2-devel perl-Switch libftdi-devel cmake cmake3 scons
sudo python3 -m pip install --upgrade pip
sudo python2 -m pip install --upgrade pip
sudo python3 -m pip install artifactory twisted prettytable sqlalchemy pyelftools openpyxl xlsxwriter pyyaml numpy configparser pyvcd
sudo python2 -m pip install configparser
sudo yum install autoconf automake libmpc-devel mpfr-devel gmp-devel gawk  bison flex texinfo patchutils gcc gcc-c++ zlib-devel
 
Do I still need install some other packages?

bluewww, 2020-07-27:
Try `sudo yum install GraphicsMagick-devel ImageMagick-devel`

yywang1993, 2020-07-27:
Both packages are also installed...
[root@localhost pulpissimo]# yum install GraphicsMagick-devel ImageMagick-devel
Loaded plugins: fastestmirror, langpacks
Loading mirror speeds from cached hostfile
 * base: mirrors.cn99.com
 * epel: mirrors.njupt.edu.cn
 * extras: mirrors.cn99.com
 * updates: mirrors.163.com
Package GraphicsMagick-devel-1.3.34-1.el7.x86_64 already installed and latest version
Package ImageMagick-devel-6.9.10.68-3.el7.x86_64 already installed and latest version
Nothing to do

bluewww, 2020-07-27:
Does `/usr/include/GraphicsMagick/Magick++.h` exist for you?

yywang1993, 2020-07-27:
Great! You find the problem, I install GraphicsMagick-c++-devel and ImageMagick-c++-devel.
After that, the build could go on, but I got another problem:
sdk:archi:prepare (pulpissimo@config_file=chips/pulpissimo/pulpissimo.json): make build
make[2]: Entering directory `/root/pulpissimo/pulpissimo/pulp-sdk/runtime/archi'
scons
scons: Reading SConscript files ...
ImportError: No module named configparser:
  File "/root/pulpissimo/pulpissimo/pulp-sdk/runtime/archi/SConstruct", line 4:
    import pulp_config as plpconfig
  File "/root/pulpissimo/pulpissimo/pulp-sdk/pkg/sdk/dev/install/ws/python/pulp_config.py", line 23:
    import configparser
make[2]: *** [build] Error 2
make[2]: Leaving directory `/root/pulpissimo/pulpissimo/pulp-sdk/runtime/archi'
Reached EOF with exit status 2
FATAL ERROR: the command 'build' has failed
make[1]: *** [all] Error 255
make[1]: Leaving directory `/root/pulpissimo/pulpissimo/pulp-sdk'
make: *** [pulp-sdk] Error 2
 
But I had already installed configparser, as you can see below:
[root@localhost pulpissimo]# pip3 install configparser
Requirement already satisfied: configparser in /root/anaconda3/lib/python3.6/site-packages (5.0.0)
Did I miss some environment variable?

bluewww, 2020-07-27:
you probably use system python but installed the configparser package for anaconda. Make sure you install the configparser package for your system python installation. I'm sure you put weird stuff into your `.bashrc` that makes `which pip3` point to andaconda.

yywang1993, 2020-07-28:
Yes you are right, thanks!
By now, the **make build-pulp-sdk** is done.
Then I **source env/pulpissimo.sh**, in pulpissimo dir, I execute **./update-ips**; but I only got a few files not all needed IPs.
So when I execute make build, I got error list this:
........(leave out a part)
  Building pulp_soc/pulp_soc
** Error: ../ips/pulp_soc/rtl/pulp_soc/pulp_soc.sv(15): (vlib-13006) Could not find the package (dm).  Design read will continue, but expect a cascade of errors after this failure.  Furthermore if you experience a vopt-7 error immediately before this error then please check the package names or the library search paths on the command line.
** Error: ../ips/pulp_soc/rtl/pulp_soc/pulp_soc.sv(280): (vlib-2164) Class or package 'dm' not found.
** Error: ../ips/pulp_soc/rtl/pulp_soc/pulp_soc.sv(280): (vlib-2900) The localparam 'dm' is declared without a default value.
** Error: (vlib-13069) ../ips/pulp_soc/rtl/pulp_soc/pulp_soc.sv(280): near "::": syntax error, unexpected ::, expecting ';' or ','.
** Error: ../ips/pulp_soc/rtl/pulp_soc/pulp_soc.sv(285): (vlib-2164) Class or package 'dm' not found.
** Error: ../ips/pulp_soc/rtl/pulp_soc/pulp_soc.sv(286): (vlib-2164) Class or package 'dm' not found.
** Error: ../ips/pulp_soc/rtl/pulp_soc/pulp_soc.sv(289): (vlib-2164) Class or package 'dm' not found.
** Error: ../ips/pulp_soc/rtl/pulp_soc/pulp_soc.sv(289): (vlib-13205) Syntax error found in the scope following 'dm'. Is there a missing '::'?
make[3]: *** [/root/pulpissimo/pulpissimo/install/modelsim_libs/pulp_soc_lib/pulp_soc.vmake] Error 2
make[2]: *** [build] Error 2
make[1]: *** [build] Error 2
make[1]: Leaving directory `/root/pulpissimo/pulpissimo/sim'
make: *** [build] Error 2

It seems like the script didn't download all IPs list in **ips_list.yml**, did I miss some step or something else?

bluewww, 2020-07-28:
Seems to be same error as in https://github.com/pulp-platform/pulpissimo/issues/148. What is the output of `vsim -version` for you?

yywang1993, 2020-07-29:
Same with him, I also use questasim 10.7c. I looked #148, and deleted the old dir, re-download pulpissimo, follow README.
Also, I run **make build-pulp-sdk** successfully. But when I run **./update-ips**, I still got following message:
[root@localhost pulpissimo]# ./update-ips 
Cloning into 'ipstools'...
remote: Enumerating objects: 945, done.
remote: Total 945 (delta 0), reused 0 (delta 0), pack-reused 945
Receiving objects: 100% (945/945), 172.84 KiB | 21.00 KiB/s, done.
Resolving deltas: 100% (661/661), done.
Retrieving ips_list.yml dependency list for all IPs (may take some time)...
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/pulp_soc/v1.2.0/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/tbtools/v0.1/ips_list.yml
Generated IP dependency tree.

Cloning ip 'pulp_soc'...
Cloning into 'pulp_soc'...
remote: Enumerating objects: 94, done.
remote: Counting objects: 100% (94/94), done.
remote: Compressing objects: 100% (56/56), done.
remote: Total 1746 (delta 61), reused 56 (delta 38), pack-reused 1652
Receiving objects: 100% (1746/1746), 396.01 KiB | 16.00 KiB/s, done.
Resolving deltas: 100% (1207/1207), done.
Note: checking out 'v1.2.0'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b new_branch_name

HEAD is now at 3736e78... Merge pull request #28 from pulp-platform/for-upstream

Cloning ip 'tbtools'...
Cloning into 'tbtools'...
remote: Enumerating objects: 5, done.
remote: Counting objects: 100% (5/5), done.
remote: Compressing objects: 100% (4/4), done.
remote: Total 158 (delta 0), reused 3 (delta 0), pack-reused 153
Receiving objects: 100% (158/158), 24.85 KiB | 11.00 KiB/s, done.
Resolving deltas: 100% (47/47), done.
Note: checking out 'v0.1'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b new_branch_name

HEAD is now at ab82317... Added missing return value for dpi_trace_new

SUMMARY
IPs updated successfully!
Generated new scripts for IPs!

Then I try **make checkout**, I got errors:
[root@localhost pulpissimo]# make checkout
Makefile:30: warning: overriding recipe for target `/root/pulpissimo/pulpissimo/install/tcl_files/config/vsim_ips.tcl'
Makefile:30: warning: ignoring old recipe for target `/root/pulpissimo/pulpissimo/install/tcl_files/config/vsim_ips.tcl'
git submodule update --init
./update-ips
Retrieving ips_list.yml dependency list for all IPs (may take some time)...
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/pulp_soc/v1.2.0/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/tbtools/v0.1/ips_list.yml
Generated IP dependency tree.

Updating ip 'pulp_soc'...
Traceback (most recent call last):
  File "./update-ips", line 47, in <module>
    ipdb.update_ips()
  File "ipstools/ipstools/IPDatabase.py", line 525, in update_ips
ValueError: invalid literal for int() with base 10: b''
make: *** [checkout] Error 1

I have doubts again, why is this happening?