pulp-platform/pulpissimo #151: The GUI fails

baraujo3007, 2020-07-31:
Hello, I'm trying to run the examples but I got an error with the GUI
The version of the modelsim I'm using is the 19.4

`Failed to open executable /home/Bruna/intelFPGA_pro/19.4/modelsim_ase/bin/../linux/../linux_x86_64pe/vish in execute mode needed for the option -64.
execv: No such file or directory
** Fatal: Unable to exec the GUI /home/Bruna/intelFPGA_pro/19.4/modelsim_ase/bin/../linux/../linux_x86_64pe/vish.
/home/Bruna/pulp-runtime/rules/pulpos/default_rules.mk:226: recipe for target 'run' failed
make: *** [run] Error 1`

Thank you for your help.

bluewww, 2020-08-01:
Yeah we don't have support for intel FPGA's unfortunately. Their tooling is just too old (as in the modelsim version they provide is like 10 years old)

closed 2020-08-01 by bluewww

baraujo3007, 2020-08-03:
So, which simulator can I use instead of modelsim?

bluewww, 2020-08-03:
We support questasim (successor of modelsim) and xcelium/ncsim.

There was also some work done on a verilator port, but it is not finished and not being worked on. Things like a fll model in systemverilog (our fll is written in vhdl) and workarounds to handle systemverilog interfaces are needed. Also I don't know how well verilator interacts with clock gating logic.