pulp-platform/pulpissimo #153: OpenOCD missing remote_bitbang

hpollittsmith, 2020-08-20:
Hi,

I am trying to connect to my Questasim simulation with OpenOCD following the instructions [here](https://github.com/pulp-platform/pulpissimo/tree/master/rtl/tb). However, when I try to connect with $OPENOCD/bin/openocd -f pulpissimo_debug.cfg, I get the following output:
```
Open On-Chip Debugger 0.10.0+dev-00615-ge5aee65 (2020-08-20-07:34)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
User : 13 2 command.c:544 command_print(): debug_level: 4
Debug: 14 2 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_adapter_khz 10000
Debug: 15 2 command.c:143 script_debug(): command - adapter_khz ocd_adapter_khz 10000
Debug: 17 2 core.c:1645 jtag_config_khz(): handle jtag khz
Debug: 18 2 core.c:1612 adapter_khz_to_speed(): convert khz to interface specific speed value
Debug: 19 2 core.c:1612 adapter_khz_to_speed(): convert khz to interface specific speed value
User : 20 2 command.c:544 command_print(): adapter speed: 10000 kHz
Debug: 21 2 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_interface remote_bitbang
Debug: 22 2 command.c:143 script_debug(): command - interface ocd_interface remote_bitbang
Error: 24 2 adapter.c:182 handle_interface_command(): The specified debug interface was not found (remote_bitbang)
User : 25 2 command.c:544 command_print(): The following debug interfaces are available:
User : 26 2 command.c:544 command_print(): 1: ftdi
User : 27 2 command.c:544 command_print(): 2: usb_blaster
User : 28 2 command.c:544 command_print(): 3: ft232r
User : 29 2 command.c:544 command_print(): 4: presto
User : 30 2 command.c:544 command_print(): 5: openjtag
User : 31 2 command.c:544 command_print(): 6: jlink
User : 32 2 command.c:544 command_print(): 7: vsllink
User : 33 2 command.c:544 command_print(): 8: ulink
User : 34 2 command.c:544 command_print(): 9: hla
User : 35 2 command.c:544 command_print(): 10: osbdm
User : 36 2 command.c:544 command_print(): 11: opendous
User : 37 2 command.c:544 command_print(): 12: aice
User : 38 2 command.c:544 command_print(): 13: xds110
Debug: 39 2 command.c:651 run_command(): Command 'interface' failed with error code -101
User : 40 2 command.c:715 command_run_line():
```
It seems that remote_bitbang support isn't installed by default. How can I set this up using the plpbuild command in pulp-sdk?

bluewww, 2020-08-20:
when compiling openocd you need to set the bitbang flag, it is not included by default. We can add this to the README.

hpollittsmith, 2020-08-20:
Thanks for confirming. I edited the pulp-sdk/tools/riscv-openocd/configure.ac file and changed line 371 `[build_remote_bitbang=no]` to `[build_remote_bitbang=yes]` and re-installed OpenOCD, and that seems to have worked. However, I am now hitting the same issue as [#60](https://github.com/pulp-platform/pulpissimo/issues/60). Was this ever resolved?

closed 2021-03-23 by bluewww