pulp-platform/pulpissimo #155: Disabling FLLs

abuck97, 2020-08-25:
Hi, 
We are attempting to disable the FLLs for synthesis, but when I try to simulate, it looks like it gets stuck after restarting the core. I've commented out the `gf22_FLL` modules in `soc_clk_rst_gen.sv` and hardwired the `sel_fll_clk_i` signal to 1'b1 so that the `ref_clk_i` should be selected by the `pulp_clock_mux2` modules. However, I can see in Modelsim that the `soc_fll_slave_req_i` in `soc_clk_rst_gen.sv` goes high and never comes down because the `ack` signal was generated by the FLLs. The `apb_fll_if.sv` is where that `req` signal originates. I tried just setting the `ack` signal as one clock delay from the `req` but then I get a "Branch decision is X" error. 

Is there something else I need to do to completely disable the FLLs? I can see that in the `tb_pulp.sv` testbench, there is some commented out code regarding setting the correct `jtag_conf_reg` values, but even when I uncommented this code and set those values, I'm still getting the same error. There is also a comment in that testbench "TODO: regression: we can't propgate our FLL settings like this. Needs sw changes (?)" 
What software changes would be required to disable the FLL?

Thank you!

bbo49, 2020-08-26:
In the FPGA port the FLL's are also disabled. You could take a look at fpga_clk_gen which is instantiated inside `soc_clk_rst_gen.sv` when running the running the system on FPGA: 
https://github.com/pulp-platform/pulpissimo/blob/master/fpga/pulpissimo-genesys2/rtl/fpga_clk_gen.sv

Also it could be necessary to disable the FLL when building the SDK.

abuck97, 2020-08-26:
Yes, it looks like the FLL needs to be disabled in the SDK. When I run `make dis` for my sample programs I can see that it's calling functions like, `__rt_fll_init()` in the disassembly. Any idea how to disable this?

quangdaovu, 2020-08-27:
You have to comment-out that function on sdk.

abuck97, 2020-08-28:
Hi,
Thanks for the help. I think I solved the issue. Here's what I did:
I'm still using the `pulp-builder` way to build my programs. The file I needed to edit was `pulp-rt/kernel/init.c`. Just needed to comment out line 111 `__rt_freq_init();` and then run `build-runtime`.

closed 2020-08-28 by abuck97