pulp-platform/pulpissimo #156: How to get PULPissimo running without any errors?

LPLATU, 2020-08-26:
Hello,

lately I'm having trouble getting PULPissimo to run without any errors.
I could fix some issues by reading the solutions others posted here but some other issues still remain.

For that reason I would like to explain to you my steps and what errors I get and, if possible, how to fix them.
It would be really helpful if we could also solve the remaining errors.
Of course I try to follow the guide https://github.com/pulp-platform/pulpissimo
So here is what I do (OS is Scientific Linux 7.6):

1) Open terminal and switch to where I want to install PULPissimo (e.g. /local/lpla)
2) Clone PULPissimo (type "git clone https://github.com/pulp-platform/pulpissimo.git" in terminal)
3) Switch into PULPissimo (type "cd pulpissimo" in terminal)

The guide tells us to choose between the simple runtime or the full runtime. I try the full runtime.
Also get the RISC-V GNU Compiler Toolchain if you haven't already: https://github.com/pulp-platform/pulp-riscv-gnu-toolchain and set PULP_RISCV_GCC_TOOLCHAIN to its path.

4) Install the system dependencies https://github.com/pulp-platform/pulp-builder/blob/master/README.md#scientific-linux-74
5) In terminal type "make build-pulp-sdk" (this might take a minute or two)
6) In terminal type "source env/pulpissimo.sh"
7) In terminal type "./update-ips"

This is where I get my first error. When I run it the first time everything is fine and the IPs are updated successfully!
But from the second time onward it will result in an error:
`Updating ip 'apb_fll_if'...
Traceback (most recent call last):
  File "./update-ips", line 47, in <module>
    ipdb.update_ips()
  File "ipstools/ipstools/IPDatabase.py", line 525, in update_ips
ValueError: invalid literal for int() with base 10: b''
`

8) In terminal type "source setup/vsim.sh"
9) In terminal type "make build"
10) Get the pulp-rt-examples: https://github.com/pulp-platform/pulp-rt-examples

Note: You might need to update your rtl_vopt.tcl file in /pulpissimo/sim/tcl_files and add "-64" in lines 27, 28, 30 if you're on a 64-Bit machine.

11) In terminal type "cd pulp-rt-examples/hello"
12) In terminal type "make clean all run"

If everything went well the code will run and you will get "Hello !" from the STDOUT.

If it is okay for the moderators I would like to keep this guide active and add possible errors and solutions that might come up during the installation of PULPissimo. Kind of like a step-by-step guide with an integrated FAQ and also link to the corresponding issues so people can easily find them and contribute.

Regards,
LPLA

quangdaovu, 2020-08-27:
You don't have to update-ips everytime. Just once.

bluewww, 2020-11-17:
Please re-open if you still have this issue. I can't reproduce it.
I guess its a python version issue. You need python3.4 >=

closed 2020-11-17 by bluewww