pulp-platform/pulpissimo #157: riscv_core source code

q3104008073, 2020-08-28:
i am trying to get the url of riscv_core that called by pulpissimo,but i seem hard to locate the address!!!

meggiman, 2020-08-29:
You can find the core under:
https://github.com/openhwgroup/cv32e40p
The core was contributed to the OpenHW Group a while ago who are now in charge of its maintenance. Tag pulpissimo-v3.4.0 is the one currently referenced by pulpissimo. The information is admittedly a bit hard to find. All subips of pulpissimo are normally fetched by our build tool IPApprox. The tool looks for an ips_list.yml file that specifies all subip dependencies. This is done recursively for all IPs. Most subips of pulpissimo are however included in the subproject pulp-platform/pulp_soc. Check the file:
https://github.com/pulp-platform/pulp_soc/blob/master/ips_list.yml
The name of each ip denotes the project name within the Github pulp-platform group. The RI5CY core is still referenced through the old repository (pulp-platform/riscv) that is archived and nowadays automatically forwards to the openhwgroup repository.

q3104008073, 2020-08-31:
thank you very much! I will follow the suggestions and hope everything goes well!!

closed 2020-09-08 by bluewww