pulp-platform/pulpissimo #158: Issues between running examples

yywang1993, 2020-09-01:
I try to run examples(hello), I could successfully compile all files and in the termianl I could see "Hello wyy!", but I got errors at the end. It seems when Core resuming from halt status, Env couldn't get the value of debug register(is it _**dmstatus**_?). When this happens, did my simulation succeed or fail? The log is below:

(leaving a part)...
# [TB]       1ns - Using FLL
# [TB]       1ns - Not using CAM SDVT
# Loading default stimuli
# [JTAG] SoftReset Done(    701ns)
# [JTAG] Bypass Test Passed (  33301ns)
# [JTAG] Tap ID: 249511c3 (  43701ns)
# [JTAG] Tap ID Test PASSED (  43701ns)
# [test_mode_if]   50301ns - Init
# [TB]   50301ns - Enabling clock out via jtag
# [TB]   50301ns - Releasing hard reset
# [TB]   51901ns - Init PULP TAP
# [pulp_tap_if] WRITE32 burst @1c008080 for           4 bytes.
# [TB]   66001ns - Write32 PULP TAP
# [JTAG] R/W test of L2 succeeded
# [TB]  176201ns - Halting the Core
# [TB]  235001ns - Writing the boot address into dpc
# [TB]  279101ns - Loading L2
# [JTAG] Loading L2 with pulp tap jtag interface
# [pulp_tap_if] WRITE32 burst @1c000000 for        1024 bytes.
# [pulp_tap_if] WRITE32 burst @1c000400 for        1024 bytes.
# [pulp_tap_if] WRITE32 burst @1c000800 for        1024 bytes.
# [pulp_tap_if] WRITE32 burst @1c000c00 for         688 bytes.
# [pulp_tap_if] WRITE32 burst @1c008000 for        1024 bytes.
# [pulp_tap_if] WRITE32 burst @1c008400 for        1024 bytes.
# [pulp_tap_if] WRITE32 burst @1c008800 for        1024 bytes.
# [pulp_tap_if] WRITE32 burst @1c008c00 for        1024 bytes.
# [pulp_tap_if] WRITE32 burst @1c009000 for        1024 bytes.
# [pulp_tap_if] WRITE32 burst @1c009400 for        1024 bytes.
# [pulp_tap_if] WRITE32 burst @1c009800 for        1024 bytes.
# [pulp_tap_if] WRITE32 burst @1c009c00 for        1024 bytes.
# [pulp_tap_if] WRITE32 burst @1c00a000 for         232 bytes.
# [pulp_tap_if] WRITE32 burst @1c010000 for         240 bytes.
# [TB] 10374101ns - Resuming the CORE
# [TB] 10947601ns retrying debug reg access
# [TB] 10976901ns retrying debug reg access
# [TB] 11006201ns retrying debug reg access
# [TB] 11035501ns retrying debug reg access
# [TB] 11064801ns retrying debug reg access
# [TB] 11094101ns retrying debug reg access
# [TB] 11123401ns retrying debug reg access
# [TB] 11152701ns retrying debug reg access
# [TB] 11196701ns - Waiting for end of computation
# [STDOUT-CL31_PE0] Hello wyy!
# [TB] 11434901ns - Received status core: 0x00000000
# ** Note: $stop    : /root/pulpissimo/pulpissimo/sim/../rtl/tb/tb_pulp.sv(814)
#    Time: 11434901 ns  Iteration: 0  Instance: /tb_pulp
# Break at /root/pulpissimo/pulpissimo/sim/../rtl/tb/tb_pulp.sv line 814
# Stopped at /root/pulpissimo/pulpissimo/sim/../rtl/tb/tb_pulp.sv line 814
# End time: 19:23:47 on Aug 31,2020, Elapsed time: 0:00:37
# Errors: 4, Warnings: 22

q3104008073, 2020-09-04:
@yywang1993 I think the ERROR is due to questasim software bug itself, without much impact on the RTL running .
I saw you program have print "Hello wyy" successfully,but I am failed,can you tell me or show me the process to generate a right stim.txt for hello test case.
thanks!

bluewww, 2020-09-08:
@yywang1993 @q3104008073 
indeed the errors are due to questasim struggling to parse some (unused) files in `common_cells`. It is a bit ugly, but harmless.