pulp-platform/pulpissimo #159: print "hello" with sdk

q3104008073, 2020-09-03:
days ago,i did a test with hello case:
git clone https://github.com/pulp-platform/pulp-rt-examples.git
make clean all
and after compile all files it would print "hello"
but now i  do hello case again,it doesn't print any message and other compile information as below: 
make clean all
rm -rf /home/yujq/test/pulp-rt-examples/hello/build/pulpissimo/test/test /home/yujq/test/pulp-rt-examples/hello/build/pulpissimo/test/fc/test.o /home/yujq/test/pulp-rt-examples/hello/build/pulpissimo/test/fc//home/yujq/test/pulp-rt-examples/hello/build/pulpissimo/rt_conf.o /home/yujq/test/pulp-rt-examples/hello/build/pulpissimo/test/fc//home/yujq/test/pulp-rt-examples/hello/build/pulpissimo/rt_pad_conf.o  
plpflags gen --input=pulpissimo@config_file=chips/pulpissimo/pulpissimo.json  --config=platform=rtl  --config=**/rt/type=pulp-rt --output-dir=/home/yujq/test/pulp-rt-examples/hello/build/pulpissimo --makefile=/home/yujq/test/pulp-rt-examples/hello/build/pulpissimo/config.mk    --app=test 
plpconf --input=pulpissimo@config_file=chips/pulpissimo/pulpissimo.json  --config=platform=rtl  --config=**/rt/type=pulp-rt --output=/home/yujq/test/pulp-rt-examples/hello/build/pulpissimo/config.json 
/home/yujq/toolchain/bin/riscv32-unknown-elf-gcc  -march=rv32imfcxpulpv2 -mfdiv -D__riscv__ -O3 -g  -fdata-sections -ffunction-sections -I/home/yujq/test/pulp-sdk/pkg/sdk/dev/install/include/io -I/home/yujq/test/pulp-sdk/pkg/sdk/dev/install/include -include /home/yujq/test/pulp-rt-examples/hello/build/pulpissimo/fc_config.h    -MMD -MP -c test.c -o /home/yujq/test/pulp-rt-examples/hello/build/pulpissimo/test/fc/test.o
/home/yujq/toolchain/bin/riscv32-unknown-elf-gcc  -march=rv32imfcxpulpv2 -mfdiv -D__riscv__ -O3 -g  -fdata-sections -ffunction-sections -I/home/yujq/test/pulp-sdk/pkg/sdk/dev/install/include/io -I/home/yujq/test/pulp-sdk/pkg/sdk/dev/install/include -include /home/yujq/test/pulp-rt-examples/hello/build/pulpissimo/fc_config.h    -MMD -MP -c /home/yujq/test/pulp-rt-examples/hello/build/pulpissimo/rt_conf.c -o /home/yujq/test/pulp-rt-examples/hello/build/pulpissimo/test/fc//home/yujq/test/pulp-rt-examples/hello/build/pulpissimo/rt_conf.o
/home/yujq/toolchain/bin/riscv32-unknown-elf-gcc  -march=rv32imfcxpulpv2 -mfdiv -D__riscv__ -O3 -g  -fdata-sections -ffunction-sections -I/home/yujq/test/pulp-sdk/pkg/sdk/dev/install/include/io -I/home/yujq/test/pulp-sdk/pkg/sdk/dev/install/include -include /home/yujq/test/pulp-rt-examples/hello/build/pulpissimo/fc_config.h    -MMD -MP -c /home/yujq/test/pulp-rt-examples/hello/build/pulpissimo/rt_pad_conf.c -o /home/yujq/test/pulp-rt-examples/hello/build/pulpissimo/test/fc//home/yujq/test/pulp-rt-examples/hello/build/pulpissimo/rt_pad_conf.o
mkdir -p `dirname /home/yujq/test/pulp-rt-examples/hello/build/pulpissimo/test/test`
/home/yujq/toolchain/bin/riscv32-unknown-elf-gcc -march=rv32imfcxpulpv2 -mfdiv -D__riscv__ -MMD -MP -o /home/yujq/test/pulp-rt-examples/hello/build/pulpissimo/test/test /home/yujq/test/pulp-rt-examples/hello/build/pulpissimo/test/fc/test.o /home/yujq/test/pulp-rt-examples/hello/build/pulpissimo/test/fc//home/yujq/test/pulp-rt-examples/hello/build/pulpissimo/rt_conf.o /home/yujq/test/pulp-rt-examples/hello/build/pulpissimo/test/fc//home/yujq/test/pulp-rt-examples/hello/build/pulpissimo/rt_pad_conf.o  -nostartfiles -nostdlib -Wl,--gc-sections -L/home/yujq/test/pulp-sdk/pkg/sdk/dev/install/rules -Tpulpissimo/link.ld -L/home/yujq/test/pulp-sdk/pkg/sdk/dev/install/lib/pulpissimo -L/home/yujq/test/pulp-sdk/pkg/sdk/dev/install/lib/pulpissimo/pulpissimo -lrt -lrtio -lrt -lgcc
pulp-run --config-file=pulpissimo@config_file=chips/pulpissimo/pulpissimo.json  --config-opt=platform=rtl  --config-opt=**/rt/type=pulp-rt  --config-opt=**/rt/type=pulp-rt --dir=/home/yujq/test/pulp-rt-examples/hello/build/pulpissimo --binary=test/test prepare

what should i do if i want to represent "hello" after compiling message?
thanks!!

bluewww, 2020-09-08:
Did you try to run `make conf` ?

q3104008073, 2020-09-08:
maybe i get a misunstanding on that process,and till now i think it should not be able to print "hello" if we only make sdk source.
but it does't matter,because i can print "hello"  now by rtl and sdk cowork.
thanks very much.
another bad thing is that ,it is very hard to collect all fittable rtl source due to various commit version updated!!!!so that sometimes even designer collect all rtl and compile passed,it still could not print "hello"

bluewww, 2020-11-17:
Yes we aware of the limitations of the ip managment tooling. We intend to migrate to bender.
Anyway nowadays we always have fixed tags in the master branch, so you should get clean updates.

closed 2020-11-17 by bluewww