pulp-platform/pulpissimo #163: vcs error

q3104008073, 2020-09-12:
while i try to add files and compile them with vcs,it report ERRORS as below:
what is the problem?
Error-[SV-UIP] Unconnected interface port
/home/yujq/riscv/pulpissimo/ips/hwpe-stream/rtl/basic/hwpe_stream_mux_static.sv, 41
"push_0_i"
  The port 'push_0_i' of top-level module 'hwpe_stream_mux_static' whose type 
  is interface 'hwpe_stream_intf_stream' is left unconnected. It is illegal to
  leave the interface ports unconnected.
  Please make sure that all the interface ports are connected.

bluewww, 2020-09-12:
@FrancescoConti

FrancescoConti, 2020-09-14:
I think this is a VCS error / incomplete SV support. I do not think that the `hwpe_stream_mux_static` is used anywhere in the PULPissimo example accelerator, moreover on the inside that port is connected.
Unfortunately, since we do not currently support VCS I do not have much more to add...

closed 2020-09-14 by FrancescoConti