pulp-platform/pulpissimo #165: system compatiable problem

q3104008073, 2020-09-14:
while i use questasim ,the project run very well,
but with vcs 201606 version it report as below:
how to fix it?
Error-[ICPSD] Illegal combination of drivers
/home/riscv/pulpissimo/rtl/tb/tb_pulp_vcs.sv, 190
  Illegal combination of structural and procedural drivers.
  Variable "s_tdo" is driven by an invalid combination of structural and 
  procedural drivers. Variables driven by a structural driver cannot have any 
  other drivers.
  This variable is declared at 
  "/home/riscv/pulpissimo/rtl/tb/tb_pulp_vcs.sv", 190: logic s_tdo;
  The first driver is at "/home/riscv/pulpissimo/rtl/tb/tb_pulp_vcs.sv", 
  679: jtag_pkg::jtag_bypass_test(s_tck, s_tms, s_trstn, s_tdi, s_tdo);
  The second driver is at "/home/riscv/pulpissimo/rtl/tb/tb_pulp_vcs.sv",
  353: assign s_tdo = tmp_tdo;

FrancescoConti, 2020-09-14:
Unfortunately, we currently do not support VCS https://github.com/pulp-platform/pulpissimo#requirements

closed 2020-09-14 by FrancescoConti

tauqeerali-lm, 2020-11-24:
@q3104008073  did you find any solution for illegal combination of drivers problems as mentioned above while running on vcs?

q3104008073, 2020-11-24:
yeah,
comment this line:assign s_tdo        = tmp_tdo;
and replace it as follow:
always@(*)s_tdo     = tmp_tdo;

tauqeerali-lm, 2020-11-25:
I got it, did you able to run pulpissimo using VCS. I have compiled all modules and simv file is generated but when i run ./simv it got stuck at reading default stimuli. I have already checked it in tb_pulp.sv using $display, it is reading correct data from stim.txt file.