pulp-platform/pulpissimo #168: fsdb error

q3104008073, 2020-09-27:
i want to dump waveform with fsdb,but when i used parameter:
USE_S25FS256S_MODEL =1
and
in sim/Makefile i add -pli and fsdb and dump_fsdb_questa.tcl option 
simc:
	$(VSIM) -64 -c vopt_tb -L models_lib -L vip_lib \
		-suppress vsim-3009 -suppress vsim-8683 \
		+UVM_NO_RELNOTES -stats -t ps \
		-sv_lib $(SVLIB) $(VSIM_FLAGS) \
		-do "set StdArithNoWarnings 1; set NumericStdNoWarnings 1" \
                -pli ${VERDI_HOME}/share/PLI/MODELSIM/linux64/novas_fli.so \
                +fsdb_autoflush\
                -do ./dump_fsdb_questa.tcl \
		-do "run -all" \
	
dump_fsdb_questa.tcl:
  fsdbDumpfile "qtb.fsdb"
  fsdbDumpvars 0 "tb_pulp"
it return  error as below:
** Fatal: Trouble with Simulation Kernel.
Makefile:28: recipe for target 'simc' failed
make: *** [simc] Error 218

what should i do ?

alternatively , if USE_S25FS256S_MODEL =0 
or without fsdb dump relate option make simc is  ok.

bluewww, 2020-09-27:
If this is really the full error log, then it looks like a simulator bug to me.

q3104008073, 2020-09-29:
it maybe due to the S25FS256S simulation model parameters
change delay from 1000 to 1 as below,
fsdb dump ok!
 always @(posedge read_out)
    begin
        if (PoweredUp == 1'b1)
        begin
            oe_z = 1'b1;
            //#1000 oe_z = 1'b0;
            #1 oe_z = 1'b0;

            if (CSNeg_ipd==1'b0)
            begin
                oe = 1'b1;
                //#1000 oe = 1'b0;
                #1 oe = 1'b0;
            end
        end