pulp-platform/pulpissimo #172: How to change vsim to any other simulator

rahulraveendran15-coder, 2020-10-08:
I want to change the simulator from Questa to cadence, is it possible to do it?
How can I do which all files do I need to change for that?

bluewww, 2020-10-19:
You can simulate PULPissimo with xcelium (look into the `sim/Makefile.incisive` file), but it is not well documented, and we didn't add the necessary software changes to the pulp-sdk to make it running in a seamless fashion.

rahulraveendran15-coder, 2020-10-19:
> You can simulate PULPissimo with xcelium (look into the `sim/Makefile.incisive` file), but it is not well documented, and we didn't add the necessary software changes to the pulp-sdk to make it running in a seamless fashion.

Thank you very much for the update, Could you please let me know how can I run simulation with xcelium instead of Questasim,
I have gone through the file and didnt get much information from it how to run. If I get a bit more idea on this it will be very much helpful.

we didn't add the necessary software changes to the pulp-sdk to make it running in a seamless fashion. >> So does this means that will it cause any issue if I run with xcelium?

bluewww, 2020-10-19:
> Thank you very much for the update, Could you please let me know how can I run simulation with xcelium instead of Questasim,
> I have gone through the file and didnt get much information from it how [...]
You should be able to compile the design for xcelium by doing something like
```bash
cd sim
make -f Makefile.incisive build-sc
```
and then run it with
```bash
export XRUN_FLAGS=+stimuli=your_stimuli_file_generated_by_pulp_sdk
make -f Makefile.incisive sim-sc
```
> 
> we didn't add the necessary software changes to the pulp-sdk to make it running in a seamless fashion. >> So does this means that will it cause any issue if I run with xcelium?

No there shouldn't be any issues. It is just that when you call `make all run` in your shell for a program using pulp-sdk it already sets all the environment variables correctly and directly launches questasim. If you want to use xcelium you need to not use `make run` and instead set `XRUN_FLAGS`, pointing it to the genreated `*.stim` file.

rahulraveendran15-coder, 2020-10-19:
> > > You can simulate PULPissimo with xcelium (look into the `sim/Makefile.incisive` file), but it is not well documented, and we didn't add the necessary software changes to the pulp-sdk to make it running in a seamless fashion.
> > 
> > 
> > Thank you very much for the update, Could you please let me know how can I run simulation with xcelium instead of Questasim,
> > I have gone through the file and didnt get much information from it how to run. If I get a bit more idea on this it will be very much helpful.
> > You should be able to compile the design for xcelium by doing something like
> 
> ```shell
> cd sim
> make -f Makefile.incisive build-sc
> ```
> 
> and then run it with
> 
> ```shell
> export XRUN_FLAGS=+stimuli=your_stimuli_file_generated_by_pulp_sdk
> make -f Makefile.incisive sim-sc
> ```
> 
> > we didn't add the necessary software changes to the pulp-sdk to make it running in a seamless fashion. >> So does this means that will it cause any issue if I run with xcelium?
> 
> No there shouldn't be any issues. It is just that when you call `make all run` in your shell for a program using pulp-sdk it already sets all the environment variables correctly and directly launches questasim. If you want to use xcelium you need to not use `make run` and instead set `XRUN_FLAGS`, pointing it to the genreated `*.stim` file.

Thank you very much.
One more concern I have is that if I am using " make -f Makefile.incisive build-sc" how it will decide my test file or the c file which I need to compile and simulate. For example if my testcase is hello can I do something like this below:

cd pulp-rt-examples/hello
make -f Makefile.incisive build-sc
export XRUN_FLAGS=+stimuli=your_stimuli_file_generated_by_pulp_sdk
make -f Makefile.incisive sim-sc

Will the above be fine ?

In Questasim setup I am seeing the below commands:

cd pulp-rt-examples/hello
make clean all run

bluewww, 2020-10-19:
By setting `XRUN_FLAGS=+stimuli=your_file`
When you do a `make conf all` in `pulp-rt-examples/hello`, it will produce a binary and derive a stimuli file (`stim.txt`) from it. You can find it by running `find . -name stim.txt`. Then you set `your_file` in `XRUN_FLAGS` to that `stim.txt`

rahulraveendran15-coder, 2020-10-19:
> By setting `XRUN_FLAGS=+stimuli=your_file`
> When you do a `make conf all` in `pulp-rt-examples/hello`, it will produce a binary and derive a stimuli file (`stim.txt`) from it. You can find it by running `find . -name stim.txt`. Then you set `your_file` in `XRUN_FLAGS` to that `stim.txt`

So if i am doing `make conf all` from  pulp-rt-examples/hello, do I need to do `make -f Makefile.incisive build-sc` before it or after that what is the purpose of the command `make -f Makefile.incisive build-sc`? Does this command has any dependency with the hello test case?

Also after setting my file stimuli.txt can I run the simulation by `make -f Makefile.incisive sim-sc`

what is the use of these two make commands with incisive also the `make conf all`

bluewww, 2020-10-19:
`make -f Makefile.incisive build-sc` "compiles" the hardware design using xcelium.
`make conf all` in `pulp-rt-examples/hello` configures and compiles the hello world program using pulp-gcc.

These commands are independant from each other.

`make -f Makefile.incisive sim-sc` uses the compiled hardware design and launches a simulation instance. In order to tell the simulation instance where to look for the compiled hello world program (or any other program) you have to set XRUN_FLAGS appropriately.

rahulraveendran15-coder, 2020-10-19:
> `make -f Makefile.incisive build-sc` "compiles" the hardware design using xcelium.
> `make conf all` in `pulp-rt-examples/hello` configures and compiles the hello world program using pulp-gcc.
>
> These [...]

Thank you for the detailed explanation.
Still I have one concern so do I need to compile the compile the hw-design first using  are my steps correct as written below?
 
(1) `make -f Makefile.incisive build-sc` 
(2) `make conf all` 
(3) set XRUN_FLAGS
(4)  `make -f Makefile.incisive sim-sc`

Is `make conf all` mandatory to use?
Also one more doubt I have is that when running using QUESTASIM  this `make conf all` is not mentioned anywhere so why it is needed with xcelium is it because of  this "we didn't add the necessary software changes to the pulp-sdk to make it running in a seamless fashion"

dasbacete, 2022-11-28:
Hello, did you manage to run the project with xcelium? I am having the same issue. I do not find the Makefile.incisive, is it downloaded when the pulpissimo repository is cloned? Thanks

tanbaobaolove, 2023-04-20:
> I want to change the simulator from Questa to cadence, is it possible to do it? How can I do which all files do I need to change for that?

Did you change the Questa to cadence successfully?

dasbacete, 2024-04-11:
Unfortunately, when I worked with pulpissimo I had to use questasim. There is an older versión of pulpissimo that seems to have scripts for cadence xcelium and other simulator but I wouldn't recomend to go for that since there is no software support (SDK) for it anymore. 
If you are a student or you are working on a company I would recommend you to ask your supervisor to ask for a questa trial license.
You can also ask in the PULP forum but I don't think you'll get any solution

kypsslp3845-svg, 2026-02-23:
Can we use XRUN to run simulation now?