pulp-platform/pulpissimo #179: Where can i find the definition of __builtin_pulp functions?

adfad1, 2020-10-18:
hello!
In hwme case and pulp_nn project  I find the use of many __builtin_pulp functions like **#define log2(x) __builtin_pulp_fl1(x)** or **__builtin_pulp_OffsetedWrite**
where can i find the definitions of these functions?

bluewww, 2020-10-19:
I'm not aware of any documentation we have. You can find the compiler changes here though:
https://github.com/pulp-platform/pulp-riscv-gcc/blob/master/gcc/config/riscv/riscv-builtins.c
https://github.com/pulp-platform/pulp-riscv-gcc/blob/master/gcc/config/riscv/riscv-builtins.c

adfad1, 2020-10-19:
> I'm not aware of any documentation we have. You can find the compiler changes here though:
> https://github.com/pulp-platform/pulp-riscv-gcc/blob/master/gcc/config/riscv/riscv-builtins.c
> https://github.com/pulp-platform/pulp-riscv-gcc/blob/master/gcc/config/riscv/riscv-builtins.c

Thanks! And I want to use some arithmetic functions in my c code like log(), so i include <math.h>
[image]

, but it reports error
[image]
how can i solve it?

closed 2020-11-17 by bluewww