pulp-platform/pulpissimo #180: Pulpissimo Faster Clock Operation

ranaya123, 2020-10-19:
Hi All,

I have done Synthesis and PnR of Pulpissimo for several different frequencies and it seems the core is only functional for frequencies lower than 40 MHz ! Even with larger timing margins with proper hold fixing, I see XXXs for moderately higher frequencies (100MHz). 

Does the design have multi-cycle paths ? If yes, could anyone please give a hint on which paths should be constrained in the design ?

Thanks in advance

bluewww, 2020-10-19:
@davideschiavone

ranaya123, 2020-10-25:
@bluewww @davideschiavone I saw following recommendations in one of your posts. 
https://github.com/pulp-platform/pulpissimo/issues/77

**set_multicycle_path 2 -setup -through [get_pins soc_domain_i/pulp_soc_i/fc_subsystem_i/lFC_CORE/id_stage_i/registers_i/riscv_register_file_i/mem_reg*/Q]
set_multicycle_path 1 -hold  -through [get_pins soc_domain_i/pulp_soc_i/fc_subsystem_i/lFC_CORE/id_stage_i/registers_i/riscv_register_file_i/mem_reg*/Q]
set_multicycle_path 2 -setup -through [get_pins soc_domain_i/pulp_soc_i/fc_subsystem_i/lFC_CORE/id_stage_i/registers_i/riscv_register_file_i/mem_fp_reg*/Q]
set_multicycle_path 1 -hold  -through [get_pins soc_domain_i/pulp_soc_i/fc_subsystem_i/lFC_CORE/id_stage_i/registers_i/riscv_register_file_i/mem_fp_reg*/Q]**

However this did not solve the issue after PnR.  Can anyone help me to get this issue resolved ?

meggiman, 2020-11-20:
Did you try to synthesize with enabled FPU? If yes, you can try to insert additional pipeline registers into the FPU (which most of the time ends up being the root cause of the critical path in PULPissimo). By default, there are no pipeline registers inserted except for the division and square root operation.  You can enable pipeline-ing for additional operations by altering the pipeline configuration in the file ips/riscv/rtl/include/riscv_defines.sv:396. Change the zeros to ones to add a pipeline register for the corresponding operation. This will of course increase the latency of these operations. Maybe start by changing C_LAT_FP32 to 1. This adds 1 pipeline register for the 32-bit floating point multiplication instructions which is probably the most critical one that is not already pipelined by default.

ranaya123, 2020-11-23:
> Did you try to synthesize with enabled FPU? If yes, you can try to insert additional pipeline registers into the FPU (which most of the time ends up being the root cause of the critical path in [...]

Hi, thanks for the hint. I was also referring to the performance numbers in following paper : 
https://ieeexplore-ieee-org.ezproxy2.utwente.nl/stamp/stamp.jsp?tp=&arnumber=7864441

and according to it, higher frequencies were certainly doable. And I thing the version in the paper does not have the FPU (?). Nonetheless, I will give it try without FPU and with FPU+Pipelining and let you know.

ranaya123, 2020-11-24:
@meggiman Hi, but on a second thought, I didn't have any setup time violations during the synthesis for higher clock frequencies. If the FPU was an issue, the synthesizer would have complained about it in the first place. 

That's why I am asking about proper constraints required for this platform to work. Could you please comment on that ?

Thanks

meggiman, 2020-11-24:
For the multicycle paths you can use:
```
# Core: latch SCM multicycle paths

set_multicycle_path 2 -setup -through [get_pins soc_domain_i/pulp_soc_i/fc_subsystem_i/FC_CORE.lFC_CORE/id_stage_i/registers_i/riscv_register_file_i/mem_reg*/Q]
set_multicycle_path 1 -hold  -through [get_pins soc_domain_i/pulp_soc_i/fc_subsystem_i/FC_CORE.lFC_CORE/id_stage_i/registers_i/riscv_register_file_i/mem_reg*/Q]

# FPU: latch SCM multicycle paths and retiming
set_multicycle_path 2 -setup -through [get_pins soc_domain_i/pulp_soc_i/fc_subsystem_i/FC_CORE.lFC_CORE/id_stage_i/registers_i/riscv_register_file_i/mem_fp_reg*/Q]
set_multicycle_path 1 -hold  -through [get_pins soc_domain_i/pulp_soc_i/fc_subsystem_i/FC_CORE.lFC_CORE/id_stage_i/registers_i/riscv_register_file_i/mem_fp_reg*/Q]
set_optimize_registers true -designs  [get_designs soc_fpnew*]
# set_balance_registers  true -design   [get_designs soc_fp_fma_wrapper*]
```

For the clocks and CDCs you can use the following as an inspiration:
```
# Input clocks

# REF CLOCK
create_clock -name REF_CLK   -period      $REF_C_Period      [get_ports pad_xtal_in]
set_clock_uncertainty                     $REF_C_Uncertainty [get_clocks REF_CLK]
set_clock_transition                      0.2                [get_clocks REF_CLK]
set_clock_latency -max                    $REF_C_Latency_Max [get_clocks REF_CLK]
set_clock_latency -min                    $REF_C_Latency_Min [get_clocks REF_CLK]

# External SOC Clock
create_clock -name EXT_SOC_CLK -period    $SOC_C_Period      [get_ports pad_soc_clk_ext_in]
set_clock_uncertainty                     $SOC_C_Uncertainty [get_clocks EXT_SOC_CLK]
set_clock_transition                      0.2                [get_clocks EXT_SOC_CLK]
set_clock_latency -max                    $SOC_C_Latency_Max [get_clocks EXT_SOC_CLK]
set_clock_latency -min                    $SOC_C_Latency_Min [get_clocks EXT_SOC_CLK]

# External Peripheral Clock
create_clock -name EXT_PER_CLK -period    $PER_C_Period      [get_ports pad_per_clk_ext_in]
set_clock_uncertainty                     $PER_C_Uncertainty [get_clocks EXT_PER_CLK]
set_clock_transition                      0.2                [get_clocks EXT_PER_CLK]
set_clock_latency -max                    $PER_C_Latency_Max [get_clocks EXT_PER_CLK]
set_clock_latency -min                    $PER_C_Latency_Min [get_clocks EXT_PER_CLK]

# JTAG CLOCK
create_clock -name JTAG_CLK -period       $JTAG_C_Period      [get_ports pad_jtag_tck]
set_clock_uncertainty                     $JTAG_C_Uncertainty [get_clocks JTAG_CLK]
set_clock_transition                      0.2                 [get_clocks JTAG_CLK]
set_clock_latency -max                    $JTAG_C_Latency_Max [get_clocks JTAG_CLK]
set_clock_latency -min                    $JTAG_C_Latency_Min [get_clocks JTAG_CLK]

# CAMERA IF CLOCK
create_clock -name CAM_CLK  -period       $CAM_C_Period       [get_ports pad_cam_pclk]
set_clock_uncertainty                     $CAM_C_Uncertainty  [get_clocks CAM_CLK]
set_clock_transition                      0.2                 [get_clocks CAM_CLK]
set_clock_latency -max                    $CAM_C_Latency_Max  [get_clocks CAM_CLK]
set_clock_latency -min                    $CAM_C_Latency_Min  [get_clocks CAM_CLK]

# SPI MASTER
set_dont_touch         [ get_cells soc_domain_i/pulp_soc_i/soc_peripherals_i/i_udma/i_spim_gen[0].i_spim/u_clockgen/clk_mux_i/i_tc_clk_mux2        ]
set_case_analysis 0    [ get_pins  soc_domain_i/pulp_soc_i/soc_peripherals_i/i_udma/i_spim_gen[0].i_spim/u_clockgen/clk_mux_i/i_tc_clk_mux2/clk_sel_i ]
create_generated_clock [ get_pins  soc_domain_i/pulp_soc_i/soc_peripherals_i/i_udma/i_spim_gen[0].i_spim/u_clockgen/i_clkdiv_cnt/clk_o_reg/Q ] \
    -name SPIM_CLK -source [get_pins soc_domain_i/pulp_soc_i/i_clk_rst_gen/clk_mux_fll_per_i/i_tc_clk_mux2/clk_o ] -divide_by $SPIM_C_Min_DIV

# FLL clocks

# FLL1 CLK
create_clock -name FLL1_SOC_CLK -period $SOC_C_Period [ get_pins soc_domain_i/pulp_soc_i/i_clk_rst_gen/i_fll_soc/FLLCLK ]
set_clock_uncertainty                     $SOC_C_Uncertainty [get_clocks FLL1_SOC_CLK]
set_clock_transition                      0.2                [get_clocks FLL1_SOC_CLK]
set_clock_latency -max                    $SOC_C_Latency_Max [get_clocks FLL1_SOC_CLK]
set_clock_latency -min                    $SOC_C_Latency_Min [get_clocks FLL1_SOC_CLK]

# FLL2 CLK
create_clock -name FLL2_PER_CLK -period $PER_C_Period [ get_pins soc_domain_i/pulp_soc_i/i_clk_rst_gen/i_fll_per/FLLCLK ]
set_clock_uncertainty                     $PER_C_Uncertainty [get_clocks FLL2_PER_CLK]
set_clock_transition                      0.2                [get_clocks FLL2_PER_CLK]
set_clock_latency -max                    $PER_C_Latency_Max [get_clocks FLL2_PER_CLK]
set_clock_latency -min                    $PER_C_Latency_Min [get_clocks FLL2_PER_CLK]

# Async clock groups
set_clock_groups -asynchronous -group REF_CLK
set_clock_groups -asynchronous -group FLL1_SOC_CLK
set_clock_groups -asynchronous -group FLL2_PER_CLK
set_clock_groups -asynchronous -group EXT_SOC_CLK
set_clock_groups -asynchronous -group EXT_PER_CLK
set_clock_groups -asynchronous -group JTAG_CLK
set_clock_groups -asynchronous -group {SPIM_CLK}
set_clock_groups -asynchronous -group CAM_CLK
```
There is nothing special going on and I don't really see how this should help you to "make the design work". If your netlist is not operational at higher frequencies you most likely have an issue in your backend flow or the timing at the interface between TB and timing back-annotated netlist. Since you claim the synthesizer and PnR went through without timing violations for your intended target frequency I don't really see how the above constraints could make a difference to make your postlayout simulation work at this particular frequency.