pulp-platform/pulpissimo #183: How to use riscv+fpu？

adfad1, 2020-10-21:
Hello!
I want to use fpu to accelerate computation, my c code is shown below, where inputs and is defined as float.
[image]
I use this command to disassemble: make dis disopt="-d -l" | less, but i can not see any float instructions been used. What should I do to config the compiler?
Thanks in advance!

bluewww, 2020-10-21:
You should try to compile with something like `-march=rv32imfcxvpulpv3`

adfad1, 2020-10-21:
> You should try to compile with something like `-march=rv32imfcxvpulpv3`

I changed the Makefile like this
[image]
but error is reported
[image]
And I changed it into -march=rv32imfcxpulpv2 but it still no float instructions

bluewww, 2020-11-04:
I cannot reproduce. Please post a self contained c file and the compile command

adfad1, 2020-11-08:
> I cannot reproduce. Please post a self contained c file and the compile command

[loss_function.tar.gz](https://github.com/pulp-platform/pulpissimo/files/5506060/loss_function.tar.gz)
I compile these codes under /pulpissimo/pulp-rt-examples/loss_function and didn't make any other changes on pulpissimo codes

Ankasysemre, 2021-01-05:
use PULP_CFLAGS = -O -g value in makefile.