pulp-platform/pulpissimo #186: Error building pulp-sdk

FrankieTankie, 2020-11-15:
Hello,
I'm stuck at building pulp-sdk. I get the following error for 'make build-pulp-sdk':
g++ -c pulp/itc/itc_v1_impl.cpp -o /opt/pulpissimo/pulp-sdk/build/sdk/gvsoc/models/pulp/itc/itc_v1_impl/pulp/itc/itc_v1_impl.o   -MMD -MP -O2 -g -fpic -D__GVSOC__=1 -I/opt/pulpissimo/pulp-sdk/pkg/sdk/dev/install/ws/include -I/usr/include/python3.6m -I/usr/include/python3.6m -Werror -Wfatal-errors -std=c++11
In file included from /opt/pulpissimo/pulp-sdk/pkg/sdk/dev/install/ws/include/archi/itc/itc_v1.h:28,
                 from pulp/itc/itc_v1_impl.cpp:26:
/opt/pulpissimo/pulp-sdk/pkg/sdk/dev/install/ws/include/archi/utils.h:23:82: error: extended character   is not valid in an identifier
   23 | #define _ARCHI_DEF_REG_MASK(x) __ARCHI_DEF_REG_STR(x ## _MASK ARCHI_REG_MASK(x ##_BIT, x ## _BITS))
      |                                                                                  ^
compilation terminated due to -Wfatal-errors.
/opt/pulpissimo/pulp-sdk/pkg/sdk/dev/install/ws/rules/vp_models.mk:135: recipe for target '/opt/pulpissimo/pulp-sdk/build/sdk/gvsoc/models/pulp/itc/itc_v1_impl/pulp/itc/itc_v1_impl.o' failed
make[3]: *** [/opt/pulpissimo/pulp-sdk/build/sdk/gvsoc/models/pulp/itc/itc_v1_impl/pulp/itc/itc_v1_impl.o] Error 1
make[3]: Verzeichnis „/opt/pulpissimo/pulp-sdk/platform/gvsoc/models“ wird verlassen
Makefile:24: recipe for target 'build' failed
make[2]: *** [build] Error 2
make[2]: Verzeichnis „/opt/pulpissimo/pulp-sdk/platform/gvsoc“ wird verlassen
Reached EOF with exit status 2
FATAL ERROR: the command 'build' has failed
Makefile:6: recipe for target 'all' failed
make[1]: *** [all] Error 255
make[1]: Verzeichnis „/opt/pulpissimo/pulp-sdk“ wird verlassen
Makefile:84: recipe for target 'pulp-sdk' failed
make: *** [pulp-sdk] Error 2

Any help would be appreciated.
Regards

bluewww, 2020-11-16:
Ok interesting it looks like there are two different underscores at the location your compilation fails:
```
Char:   (160, #o240, #xa0, file ...) point=943 of 3300 (29%) column=73
Char: _ (95, #o137, #x5f) point=797 of 3300 (24%) column=82
```
I wonder why other compilers don't complain about it... what operating system are you running?

FrankieTankie, 2020-11-16:
Hello,
Ubuntu 18.04.05 at the moment.
Regards,
Philipp

FrankieTankie, 2020-11-16:
Thank you for your effort, I just found out, that I used gcc10 what led to the problem.