pulp-platform/pulpissimo #187: How to Run RTL- Simulation of PULPissimo with Intel/ModelSim Starter Edition.

designer9406, 2020-11-17:
Hi, all.
I'm trying to run rtl-simulation of pulpissimo with intel/modelsim SE (Actually latest version).
I have fixed some codes like run.tcl and rtl_vopt.tcl.
Why I have fixed some codes is because intel/modelsim SE (No license version) doesn't include vopt.

/// run.tcl ////////////////////////////////////
#!/bin/bash
#set TB "vopt_tb -L models_lib -L vip_lib"
set TB "tb_pulp -L vip_lib -L L2_tcdm_hybrid_interco_lib -L apb_fll_if_lib -L axi_lib -L < ... > "
source ./tcl_files/config/vsim.tcl
///////////////////////////////////////////////

I commented the lines of running vopt.
/// rtl_vopt.tcl ////////////////////////////////////
line 26~32
///////////////////////////////////////////////

Also, I wrote simple application code.
/// test.c ////////////////////////////////////
#include <stdio.h>
#include "pulp.h"
int main(){
    v2s a,b,res;
    a = __PACK2(1,3);
    b = __PACK2(2,4);
    res = a+b;
    printf("result : %d", res);
  return 0;}
///////////////////////////////////////////////

/// Makefile (in pulp-rt-examples) /////////
...
#PULP_CFLAGS = -O3 -g
PULP_CFLAGS = -g
...
///////////////////////////////////////////////

<PULPissimo root>$make build                   <-- If (tb_pulp.sv or run.tcl) is modified, this command must be executed.
<pulp-rt-examples>$make clean all             <-- This command will compile the test.c code.
No Error is printed here.
(But If Something is wrong, Please let me know...)

Then, <make run gui=1> or <make run vsim/script=export_run.tcl> (Actually, Output Same Results.)
The result of the command is that Any instruction is not seem to be executed in RISCV-core.

And I think it's because of the tb_pulp.sv
...
$display("[TB] %t - Halting the Core", $realtime);
debug_mode_if.halt_harts(s_tck, s_tms, s_trstn, s_tdi, s_tdo);
...

Can you help me to fix this error ?
(Actually The error is not printed, but any instruction that is generated from test.c is not executed.)

These is the thing I've tried from here.
After commenting 'debug_mode_if.halt_harts(s_tck, s_tms, s_trstn, s_tdi, s_tdo);' in tb_pulp.sv and load the instructions into the L2_memory using 'pulp_tap_pkg::load_L2(num_stim, stimuli, s_tck, s_tms, s_trstn, s_tdi, s_tdo);'.
It takes a long time to load the instruction.
But After end of loading the instruction, Riscy-core is still Not working...

How can i see the instruction generated from test.c run on riscy-core?

bluewww, 2020-11-17:
I don't have intel modelsim and we don't support it. This version of modelsim is just too limited feature wise to compile and simulate PULPissimo. Maybe someone with more experience with intel modelsim can help you.

Anyway here is a random guess: Did you try to add signals to the wave window and observe if the simulation even runs at all?

designer9406, 2020-11-18:
Thanks for reply.
I have tried that. 

Here is VSIM log on modelsim

VSIM5> run -all
# [CORE] Core settings: PULP_SECURE =           1, N_PMP_ENTRIES =          16, N_PMP_CFG           4
# START_ADDR[0] = 0xxxxxxxxx; END_ADDR[0] = 0Xxxxxxxxx
# START_ADDR[1] = 0xxxxxxxxx; END_ADDR[1] = 0Xxxxxxxxx
# [TB]       0ns - Entry point is set to 0x1c008080
# [TB]       0ns - Asserting hard reset
# ** Warning: (vsim-3533) [FOFIW] - Failed to open file "fs/file_0_0.txt" for writing.
# No such file or directory. (errno = ENOENT)    : /home/eom/files/pulp/pulpissimo/sim/../rtl/tb/tb_fs_handler.sv(269)
#    Time: 0 ps  Iteration: 0  Instance: /tb_pulp/i_fs_handler/FS_PRINTF_CLUSTER
# ** Warning: (vsim-3533) [FOFIW] - Failed to open file "fs/file_31_0.txt" for writing.
# No such file or directory. (errno = ENOENT)    : /home/eom/files/pulp/pulpissimo/sim/../rtl/tb/tb_fs_handler.sv(269)
#    Time: 0 ps  Iteration: 0  Instance: /tb_pulp/i_fs_handler/FS_PRINTF_FC
# [TB]       1ns - Using FLL
# [TB]       1ns - Not using CAM SDVT
# Loading default stimuli
# [JTAG] SoftReset Done(    701ns)
# [JTAG] Bypass Test Passed (  33301ns)
# [JTAG] Tap ID: 249511c3 (  43701ns)
# [JTAG] Tap ID Test PASSED (  43701ns)
# [test_mode_if]   50301ns - Init
# [TB]   50301ns - Enabling clock out via jtag
# [TB]   50301ns - Releasing hard reset
# [TB]   51901ns - Init PULP TAP
# [pulp_tap_if] WRITE32 burst @1c008080 for           4 bytes.
# [TB]   66001ns - Write32 PULP TAP
# [JTAG] R/W test of L2 succeeded
# [TB]  176201ns - Halting the Core
#############  Autofindloop Analysis  ###############
#############  Loop found at time 199187908 ps ###############
#   Active process: /tb_pulp/i_dut/soc_domain_i/pulp_soc_i/i_dm_top/i_dm_mem/#IMPLICIT-WIRE(cmdbusy_o)#48 @ sub-iteration 0
#     Source: /home/eom/files/pulp/pulpissimo/sim/../ips/riscv-dbg/src/dm_top.sv:187
#     Assigning reg to (val=0)
#   Active process: /tb_pulp/i_dut/soc_domain_i/pulp_soc_i/i_dm_top/i_dm_mem/#ALWAYS#113 @ sub-iteration 1
#     Source: /home/eom/files/pulp/pulpissimo/sim/../ips/riscv-dbg/src/dm_top.sv:187
#   Active process: /tb_pulp/i_dut/soc_domain_i/pulp_soc_i/i_dm_top/i_dm_mem/#IMPLICIT-WIRE(cmdbusy_o)#48 @ sub-iteration 2
#     Source: /home/eom/files/pulp/pulpissimo/sim/../ips/riscv-dbg/src/dm_top.sv:187
#     Assigning reg to (val=0)
################# END OF LOOP #################
# ** Error (suppressible): (vsim-3601) Iteration limit 5000 reached at time 199187908 ps

and Signals of IF-stage
[image]
That is not the instructions generated from test.c

closed 2022-11-02 by designer9406

tanbaobaolove, 2023-04-19:
> 

Did you fix this problem? I want to use xcelium to run the rtl simulation.