pulp-platform/pulpissimo #192: OpenOCD won't esatblish JTAG connection between ZCU104 and JTAG-HS02

FrankieTankie, 2020-11-22:
Hello,
I tried to run the UART Looback example on a ZCU104, but I can't establish the JTAG connection to the ZCU104 via a Digilent JTAG-HS02 device. I've also tried the HS03 which won't work either. Compiling the bitstream works without errors,also compiling the example does not throw errors. But as soon as I try to open the JTAG connection, I'll get the following error (the cfg file Is just a copy of the HS01.cfg with the parameters for the HS02 from the ZCU102):
`$OPENOCD/bin/openocd -f /opt/pulpissimo/fpga/pulpissimo-zcu104/openocd-zcu104-digilent-jtag-hs3.cfg
Open On-Chip Debugger 0.10.0+dev-00615-g6822e2193 (2020-11-22-11:44)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 1000 kHz
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
   TapName             Enabled  IdCode     Expected   IrLen IrCap IrMask
-- ------------------- -------- ---------- ---------- ----- ----- ------
 0 riscv.unknown0         Y     0x00000000 0x10102001     5 0x01  0x03
 1 riscv.cpu              Y     0x00000000 0x249511c3     5 0x01  0x03
Info : clock speed 1000 kHz
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: riscv.unknown0: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Error: Unsupported DTM version: 15
Info : Listening on port 3333 for gdb connections
Error: Target not examined yet

Error: Unsupported DTM version: 15
`
Any suggestions are welcomed.
Frankie

FrankieTankie, 2020-11-22:
Hello,
I tried to run the UART Looback example on a ZCU104, but I can't establish the JTAG connection to the ZCU104 via a Digilent JTAG-HS02 device. I've also tried the HS03 which won't work either. Compiling the bitstream works without errors,also compiling the example does not throw errors. But as soon as I try to open the JTAG connection, I'll get the following error (the cfg file Is just a copy of the HS01.cfg with the parameters for the HS02 from the ZCU102):

> $OPENOCD/bin/openocd -f /opt/pulpissimo/fpga/pulpissimo-zcu104/openocd-zcu104-digilent-jtag-hs3.cfg
> Open On-Chip Debugger 0.10.0+dev-00615-g6822e2193 (2020-11-22-11:44)
> Licensed under GNU GPL v2
> For bug reports, read
> http://openocd.org/doc/doxygen/bugs.html
> adapter speed: 1000 kHz
> Info : auto-selecting first available session transport "jtag". To override use 'transport select '.
> TapName Enabled IdCode Expected IrLen IrCap IrMask
> 
> 0 riscv.unknown0 Y 0x00000000 0x10102001 5 0x01 0x03
> 1 riscv.cpu Y 0x00000000 0x249511c3 5 0x01 0x03
> Info : clock speed 1000 kHz
> Error: JTAG scan chain interrogation failed: all ones
> Error: Check JTAG interface, timings, target power, etc.
> Error: Trying to use configured scan chain anyway...
> Error: riscv.unknown0: IR capture error; saw 0x1f not 0x01
> Warn : Bypassing JTAG setup events due to errors
> Error: Unsupported DTM version: 15
> Info : Listening on port 3333 for gdb connections
> Error: Target not examined yet
> 
> Error: Unsupported DTM version: 15

Any suggestions are welcomed.
Frankie

closed 2020-11-23 by FrankieTankie

MiguelCosta94, 2021-04-15:
Hi @FrankieTankie,

How did you solve your problem?

Best regards,
Miguel

FrankieTankie, 2021-04-15:
Hi @MiguelCosta94 ,
I'm ashamed to answer this, but the problem was absolutely not related to pulpissimo  😄 . The problem was caused by my lack of knowledge about the ZCU104, because I connected the JTAG device to the ZCU104 JTAG connector instead of PMOD0.
I hope this helps, but I'm afraid it won't.
best regards,
Frankie

MiguelCosta94, 2021-04-15:
Thanks for your reply. Indeed, I have the JTAG (ARM Olimex Tiny) connected to PMOD0 but I am still not able to connect to OpenOCD. 
Did you perform any additional configuration on the jumpers and switches of ZCU104 or you left them in default?

Best regards,
Miguel

FrankieTankie, 2021-04-16:
No, not at all. I just read the issue you already closed. I was facing similiar problems due to missing jtag drivers. Another point is, that I had to close vivado, else it would occupy the device.