pulp-platform/pulpissimo #194: No such file or directory: 'curl'

bhyou, 2020-12-02:
hi, could you tell me how to solve this problem? When I'm preparing to get the latest version of the IPs composing by running ./update-ips， it is reported as follow: （based on ubuntu16.04.7 LTS ）

Retrieving ips_list.yml dependency list for all IPs (may take some time)...
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/pulp_soc/v1.4.1/ips_list.yml
Traceback (most recent call last):
  File "./update-ips", line 44, in <module>
    verbose=args.verbose
  File "ipstools/ipstools/IPDatabase.py", line 132, in __init__
    self.generate_deps_tree(verbose=verbose)
  File "ipstools/ipstools/IPDatabase.py", line 292, in generate_deps_tree
    children.append(IPTreeNode(ip, self.default_server, self.default_group, self.default_commit, verbose=True))
  File "ipstools/ipstools/IPTreeNode.py", line 74, in __init__
    ips = load_ips_list_from_server(server, group, node['name'], commit, verbose=verbose)
  File "ipstools/ipstools/IPApproX_common.py", line 161, in load_ips_list_from_server
    ips_list_yml = get_ips_list_yml(server, group, name, commit, verbose=verbose)
  File "ipstools/ipstools/IPApproX_common.py", line 136, in get_ips_list_yml
    curl = subprocess.Popen(cmd.split(), stdout=subprocess.PIPE, stderr=devnull)
  File "/usr/lib/python3.5/subprocess.py", line 947, in __init__
    restore_signals, start_new_session)
  File "/usr/lib/python3.5/subprocess.py", line 1551, in _execute_child
    raise child_exception_type(errno_num, err_msg)
FileNotFoundError: [Errno 2] No such file or directory: 'curl'

bluewww, 2020-12-02:
You need to install curl. sudo yum install curl.

bhyou, 2020-12-02:
all right, thank you very much !

closed 2020-12-02 by bluewww