pulp-platform/pulpissimo #195: ZCU104 UART Loopback Examples exits due to uart==NULL

FrankieTankie, 2020-12-02:
Hello again,
I managed to run pulpissimo on a ZCU104 board, establishing JTAG conenction and uploaded the UART loopback example form the repository.
For a, until now, unknown reason the app exits with -1 because of UART==NULL after calling rt_uart_t *uart = rt_uart_open(NULL, &conf, NULL);
Furthermore gdb gives me the following warning:
Warnung: Target-supplied registers are not supported by the current architecture 0x1a000080 in ?? ()
I don't know if this leads to the example failure, or it is just a coincidence.
Any suggestions are appreceated

bluewww, 2020-12-02:
did you try single stepping through rt_uart_open to identify the issue?

FrankieTankie, 2020-12-02:
This is what I get, but I'm not able to interpret it. Sorry for posting all the steps, but I don't wont to cut sth.

`Breakpoint 1, main () at test.c:35
35	  if (uart == NULL) return -1;
(gdb) s
75	}
(gdb) s
s
s_stext () at kernel/riscv/rt/crt0.S:127
127	  mv    s0, a0
(gdb) s

130	  jal  x1, __rt_deinit
(gdb) s
s__rt_deinit () at kernel/init.c:164
164	{
(gdb) s
165	  if (rt_platform() == ARCHI_PLATFORM_GVSOC)
(gdb) s
164	{
(gdb) s
165	  if (rt_platform() == ARCHI_PLATFORM_GVSOC)
(gdb) s
173	  __rt_cbsys_exec(RT_CBSYS_STOP);
(gdb) s
s__rt_cbsys_exec (cbsys_id=cbsys_id@entry=RT_CBSYS_STOP) at kernel/utils.c:73
73	  rt_cbsys_t *cbsys = cbsys_first[cbsys_id];
(gdb) s
s72	{
(gdb) s
74	  while (cbsys)
(gdb) s
76	    if (cbsys->callback(cbsys->arg)) return -1;
(gdb) s
__rt_io_stop (arg=0x0) at libs/io/io.c:655
655	  rt_uart_close(_rt_io_uart, NULL);
(gdb) s
647	{
(gdb) s
655	  rt_uart_close(_rt_io_uart, NULL);
(gdb) s
652	  __rt_io_uart_wait_pending();
(gdb) s
__rt_io_uart_wait_pending () at libs/io/io.c:285
285	  while(__rt_io_pending_flush)
(gdb) s
292	  if (__rt_io_event_current)
(gdb) s
315	}
(gdb) s
s__rt_io_stop (arg=<optimized out>) at libs/io/io.c:655
655	  rt_uart_close(_rt_io_uart, NULL);
(gdb) s
rt_uart_close (uart=0x1c000fd8 <__rt_uart>, event=event@entry=0x0)
    at drivers_deprecated/uart/uart.c:230
230	  int irq = rt_irq_disable();
(gdb) s
rt_irq_disable () at drivers_deprecated/uart/uart.c:230
230	  int irq = rt_irq_disable();
(gdb) s
hal_irq_disable ()
    at /opt/pulpissimo/pulp-sdk/pkg/sdk/dev/install/include/hal/riscv/riscv_v5.h:253
253	  return hal_spr_read_then_clr(0x300, 0x1<<3);
(gdb) s
rt_uart_close (uart=0x1c000fd8 <__rt_uart>, event=event@entry=0x0)
    at drivers_deprecated/uart/uart.c:234
234	  uart->open_count--;
(gdb) s
236	  if (uart->open_count == 0)
(gdb) s
[... 917 lines cut ...]
77	    rt_wait_for_interrupt();
(gdb) step
hal_itc_wait_for_interrupt ()
    at /opt/pulpissimo/pulp-sdk/pkg/sdk/dev/install/include/hal/itc/itc_v1.h:87
87	  asm volatile ("wfi");
(gdb) finish
Run till exit from #0  hal_itc_wait_for_interrupt ()
    at /opt/pulpissimo/pulp-sdk/pkg/sdk/dev/install/include/hal/itc/itc_v1.h:87
__rt_uart_wait_tx_done (uart=0x1c000fd8 <__rt_uart>)
    at drivers_deprecated/uart/uart.c:78
78	    rt_irq_mask_clr(1<<ARCHI_FC_EVT_CLK_REF);
(gdb) finish
Run till exit from #0  __rt_uart_wait_tx_done (uart=0x1c000fd8 <__rt_uart>)
    at drivers_deprecated/uart/uart.c:78
rt_uart_close (uart=0x1c000fd8 <__rt_uart>, event=event@entry=0x0)
    at drivers_deprecated/uart/uart.c:244
244	      plp_uart_disable(uart->channel - ARCHI_UDMA_UART_ID(0));      
(gdb) finish
Run till exit from #0  rt_uart_close (uart=0x1c000fd8 <__rt_uart>, 
    event=event@entry=0x0) at drivers_deprecated/uart/uart.c:244
__rt_io_stop (arg=<optimized out>) at libs/io/io.c:660
660	}
(gdb) finish
Run till exit from #0  __rt_io_stop (arg=<optimized out>) at libs/io/io.c:660
/opt/pulp-riscv-gnu-toolchain/riscv-binutils-gdb/gdb/target-descriptions.c:988: internal-error: const char* tdesc_register_name(gdbarch*, int): Assertion `data->pseudo_register_name != NULL' failed.`