pulp-platform/pulpissimo #207: synthesis fails

monniaux, 2021-01-14:
Vivado 2020.2

$ git rev-parse --short HEAD
ec23506

$ make genesys2
ERROR: [Synth 8-1587] an enum variable may only be assigned to same enum typed variable or one of its values [/home/monniaux/progs/Risc-V/pulpissimo/ips/riscv/rtl/riscv_ex_stage.sv:445]
[make.log](https://github.com/pulp-platform/pulpissimo/files/5817253/make.log)

bluewww, 2021-01-15:
@meggiman what vivado version are we using?

meggiman, 2021-01-18:
We are currently using Vivado 2019.1.1. Unfortunately, we do not have Vivado 2020.1 installed on our servers so I cannot reproduce your error. However, since this version of the core has been taped-out numerous times and works without any issues in older version of Vivado I must assume that you either have:
-  a problem in your setup (e.g. forgot to update all ips)
- modified some scripts or source files and introduced a bug
- found a bug in Xilinx Vivado that should be reported to them

I would suggest to rule out the first two issues and if the error persists, try to construct a minimal failing example (e.g. synthesizing the core only with the same parameters as in pulp_soc) and report it to Xilinx.

monniaux, 2021-01-18:
1,2: the files are those obtained by ./update_ips and I have not modified them.

monniaux, 2021-01-21:
Works on Vivado 2019.2.1.
So it's a problem with Vivado 2020…

gottmic, 2021-02-01:
When changing line 421 in ..\ips\riscv\rtl\riscv_ex_stage.sv to: 
`localparam fpnew_pkg::unit_type_t C_DIV = FP_DIVSQRT ? fpnew_pkg::MERGED : fpnew_pkg::DISABLED;`
synthesis works with Vivado 2020.2 (at least on my machine).