pulp-platform/pulpissimo #209: make build won't build

monniaux, 2021-01-15:
$ git rev-parse --short HEAD
ec23506

I followed the instructions, installed the sdk, updated the ips, and then ran make build.

make[1]: Entering directory '/home/monniaux/progs/Risc-V/pulpissimo/sim'
/bin/bash: vlib: command not found

monniaux, 2021-01-15:
Some forums say one needs Modelsim or QuestaSim, but this is not specified in the documentation.

chithambaramoorthii, 2021-02-12:
I managed to solve this issue by installing Modelsim free edition from Intel and specifying the PATH but this will end up in a new error. This is due to the fact that requirement is Mentors Modelsim. I couldn't find a free version of this yet. Stuck at this point.

 ** Error: ../ips/common_cells/src/id_queue.sv(268): Questa has encountered an unexpected internal error: ../../src/vlog/vgentd.c(670). Please contact Questa support at http://supportnet.mentor.com/

closed 2021-03-23 by bluewww