pulp-platform/pulpissimo #210: building openocd fails

monniaux, 2021-01-15:
pulpissimo-6.0.0/pulp-sdk$ source sourceme.sh && ./pulp-tools/bin/plpbuild checkout build --p openocd --stdout

libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I./src -I./src -I./src/helper -DPKGDATADIR=\"/home/monniaux/progs/Risc-V/pulpissimo6/pulpissimo-6.0.0/pulp-sdk/pkg/openocd/1.0/share/openocd\" -DBINDIR=\"/home/monniaux/progs/Risc-V/pulpissimo6/pulpissimo-6.0.0/pulp-sdk/pkg/openocd/1.0/bin\" -I./jimtcl -I./jimtcl -Wall -Wstrict-prototypes -Wformat-security -Wshadow -Wextra -Wno-unused-parameter -Wbad-function-cast -Wcast-align -Wredundant-decls -Werror -g -O2 -MT src/flash/mflash.lo -MD -MP -MF src/flash/.deps/mflash.Tpo -c src/flash/mflash.c -o src/flash/mflash.o
In file included from /usr/include/string.h:495,
                 from ./src/helper/system.h:27,
                 from ./config.h:341,
                 from src/flash/mflash.c:19:
In function ‘memset’,
    inlined from ‘mg_gen_ataid’ at src/flash/mflash.c:1162:2,
    inlined from ‘mg_storage_config’ at src/flash/mflash.c:1174:2:
/usr/include/x86_64-linux-gnu/bits/string_fortified.h:71:10: error: ‘__builtin_memset’ offset [509, 512] from the object at ‘buff’ is out of the bounds of referenced subobject ‘reserved7’ with type ‘mg_io_uint8[186]’ {aka ‘unsigned char[186]’} at offset 322 [-Werror=array-bounds]
   71 |   return __builtin___memset_chk (__dest, __ch, __len, __bos0 (__dest));
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors

gottmic, 2021-02-15:
I faced the same problem when trying to build openocd within the pulp-sdk.
You do not need to use the special pulp version of openocd anymore.
The patch (make more then 32 harts accessible) which is needed to communicate with the ibex or riscy core is available on the mainline repository now.  [https://github.com/riscv/riscv-openocd](url)

bluewww, 2021-02-15:
Indeed I upstreamed the patch a while ago

closed 2021-02-15 by bluewww