pulp-platform/pulpissimo #214: update-ips scripts seems NOT git clone AXI module

jackiezhangcn, 2021-01-25:
Hi, All:
When I run simple simulation, I was blocked at the build stage, it reported as below:

hongzhi2015, 2021-01-28:
I encounter the same issue.

Master version `update-ips` only clones pulp_soc and tbtools from  `pulpissimo/ips_list.yml`. 
Their dependencies are not cloned.

I prepared a hacked ips_list.yml, just use it to overwrite `pulpissimo/ips_list.yml`

[ips_list.yml.hack.tar.gz](https://github.com/pulp-platform/pulpissimo/files/5887036/ips_list.yml.hack.tar.gz)

And then
$ rm ips -R # Sometimes get errors on a second update-ips
$ ./update-ips
$ make build

With the hacked `ips_list.yml` I can run `pulp-runtime-examples/hello` succefully.

meggiman, 2021-01-28:
I just tried it on my own machine with the current master branch in a clean repo. I cannot reproduce your error. It would help a lot if you could provide the log output of running "./update-ips". Maybe you are using a very old version of pyhton? Anyhow, without the log it is impossible for me to guess what the issue might be.

bluewww, 2021-01-28:
I think this might be a python version issue:
Can you provide your `python --version` please

hongzhi2015, 2021-01-29:
The first line of `update-ips` is `#!/usr/bin/env python3`, and I think python3 version matters.

Here is my environment:

Linux distribution
```
$ lsb_release -a
LSB Version:	:core-4.1-amd64:core-4.1-ia32:core-4.1-noarch:cxx-4.1-amd64:cxx-4.1-ia32:cxx-4.1-noarch:desktop-4.1-amd64:desktop-4.1-ia32:desktop-4.1-noarch:languages-4.1-amd64:languages-4.1-noarch:printing-4.1-amd64:printing-4.1-noarch
Distributor ID:	CentOS
Description:	CentOS Linux release 7.8.2003 (Core)
Release:	7.8.2003
Codename:	Core
```

Python3 version
```
$ python3 --version
Python 3.6.8
```

`update-ips` log
```
$ ./update-ips 
Cloning into 'ipstools'...
remote: Enumerating objects: 945, done.
remote: Total 945 (delta 0), reused 0 (delta 0), pack-reused 945
Receiving objects: 100% (945/945), 172.84 KiB | 126.00 KiB/s, done.
Resolving deltas: 100% (661/661), done.
Retrieving ips_list.yml dependency list for all IPs (may take some time)...
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/pulp_soc/v2.0.1/ips_list.yml
   Fetching ips_list.yml from https://raw.githubusercontent.com/pulp-platform/tbtools/v0.1/ips_list.yml
Generated IP dependency tree.

Cloning ip 'pulp_soc'...
Cloning into 'pulp_soc'...
remote: Enumerating objects: 39, done.
remote: Counting objects: 100% (39/39), done.
remote: Compressing objects: 100% (31/31), done.
remote: Total 2172 (delta 16), reused 20 (delta 8), pack-reused 2133
Receiving objects: 100% (2172/2172), 508.52 KiB | 56.00 KiB/s, done.
Resolving deltas: 100% (1514/1514), done.
Note: checking out 'v2.0.1'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b new_branch_name

HEAD is now at bc10477... Update changelog

Cloning ip 'tbtools'...
Cloning into 'tbtools'...
remote: Enumerating objects: 13, done.
remote: Counting objects: 100% (13/13), done.
remote: Compressing objects: 100% (11/11), done.
remote: Total 166 (delta 3), reused 7 (delta 1), pack-reused 153
Receiving objects: 100% (166/166), 26.89 KiB | 0 bytes/s, done.
Resolving deltas: 100% (50/50), done.
Note: checking out 'v0.1'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b new_branch_name

HEAD is now at ab82317... Added missing return value for dpi_trace_new

SUMMARY
IPs updated successfully!
Generated new scripts for IPs!
```

hongzhi2015, 2021-01-29:
I finally figured out the root cause:  raw.githubusercontent.com dns resolution has been polluted.
I edited /etc/hosts and ips can be cloned.

`IPApproX_common.py: get_ips_list_yml` uses curl to retrive `ips_list.yml` from raw.githubusercontent.com, error occurred but not captured:
1. stderr has been redirected to /dev/null
1. curl return code not checked, though non-zero returned

It is best to report errors.
Probably, use ips_list.yml cloned to local disk is a safe approach.

jackiezhangcn, 2021-02-01:
Hongzhi,
Your solution works perfectly, agree with you that it's better to create a clean ips_list.yml file under the project root directory, even tried to add raw.githubXXX to /etc/hosts, sometimes still failed.

Regards,
Jackie

  2.  curl return code not checked, though non-zero returned