pulp-platform/pulpissimo #219: Booting pulpisimo from rom with flash model

Ankasysemre, 2021-02-04:
Hi,

I wanna try to boot pulpisimo from ROM . To do this first ı set "s_bootsel =1" and "parameter  USE_S25FS256S_MODEL = 1;" 
after some execution of Boot code suddenly an assertion stop simulation which is loacated in riscv_id_stage.sv line 1635 whit printing "Branch decision is X in module" message. What can be missed part in here ? Can some one explain me 
What is the correct procedure to boot Pulpisimo from ROM ?

and Is boot_code.c usable with s25f256s flash model(flash loaded with application image in .mem format) to load appliication image from flash to L2 memory ?

bluewww, 2021-02-04:
Branch decision is X means that the core executed an instruction that had the bit pattern `32'hxxxx_xxxx`. In effect, this means that the program image was not correctly loaded into the L2 RAM.

The bootcode theoretically supports loading from the flash model, but I it is not a well tested code path. This might very well be a bug.

Ankasysemre, 2021-02-24:
Hi again,
We solved the issue about loading data from flash to L2 data memory space by updating bootFromRom function. As following

static void bootFromRom(int hyperflash, int qpi)
{
  boot_code_t *data = &bootCode;

  data->hyperflash = hyperflash;
  if (hyperflash) data->blockSize = HYPER_FLASH_BLOCK_SIZE;
  else data->blockSize = FLASH_BLOCK_SIZE;
  data->qpi = qpi;
  init(data);
  flash_checkAndConf(data);
// This for statement and jump function added to solve issue.
  for(int ij=0; ij<8192*2 ; ij++){
 	 flash_read(data, ij*4 ,  (469762048 +ij*4), sizeof(4));
  }
  jump_to_address(0x1c008080);
}

bluewww, 2021-03-12:
Ok thanks good to know

closed 2021-05-06 by bluewww