pulp-platform/pulpissimo #222: Attaching a hardware accelerator

jonesemmanuel, 2021-02-11:
Hi,

I am trying to attach a hardware accelerator (for deep learning) to PULPissimo.

I have read the document about the HWPE available at https://hwpe-doc.readthedocs.io/en/latest/protocols.html and tried to understand the code of the example MAC unit.
I have also went through the WOSH: Understanding and working with PULP video(https://www.youtube.com/watch?v=27tndT6cBH0).

But I was not able to figure out how to implement the hwpe in the code. One of the main issue is the many parameters and logics in the code whose flow and purpose I am not able to understand.

It would be great if some one can point me to any other documents available or provide some pointers.

Thanks,
Jones

GiannaP, 2021-02-11:
Hi,

First of all, I need some more information...

Based on a datasheet that mentions AXI, I assume you mean the datasheet of the accelerator which you want to integrate? And I guess that the accelerator does not make use of the HWPE IPs and the classic HWPE structure (https://hwpe-doc.readthedocs.io/en/latest/). Is this correct? 

In general: when following the HWPE concept, we have two interfaces: a control interface (Peripheral/APB interface) connected to the APB bus, which allows the FC to configure the accelerator in a memory-mapped fashion, and TCDM ports (behaves as Master) which are connected to the interleaved logarithmic interconnect giving access to the memory banks. If I am not mistaken, since recently peripheral interfaces, e.g., the control interface of an accelerator, can be AXI and connected to an AXI crossbar, which is connected to the APB/peripheral bus via an AXI2APB converter.

For your case: does your accelerator have a TCDM data interface and an AXI control interface? If not, what interfaces does it have?

jonesemmanuel, 2021-02-11:
Hi,

By datasheet I meant the PULPissimo datasheet from https://github.com/pulp-platform/pulpissimo/tree/master/doc/datasheet. 
This datasheet mentions about the option of using an AXI interface for accelerator: "The AXI plug can be used to extend
the microcontroller with a multi-core cluster or an accelerator." 
So I was looking at the possibility of using AXI(or AXI-Lite) to connect my accelerator as a memory mapped device.

About the second point, you are correct. The accelerator does not use the HWPE IPs.

My intention is to use AXI interface for both control and data transfer.

quangdaovu, 2021-02-15:
You can use `s_data_out_bus` in `pulp_soc` as AXI master for your purpose. Make sure to remove cluster related stuffs (`axi_slice_dc_slave_wrap` and in/out signals).
Note that you will need to add xbar if you wanna connect multiple accelerators or multiple AXI slaves.

meggiman, 2021-02-15:
@quangdaovu thanks for your input. This used to be the preferred option but is not engouraged anymore. With version 2.0.0 of the pulp_soc sub-IP the interconnect fabric was completely overhauled. The new interconnect contains a fully parametrizable AXI XBAR. I would suggest attaching your AXI slave IP there. You will have to add a new Interface port to `ips/pulp_soc/rtl/pulp_soc/soc_interconnect_wrap.sv` and change the parameters and memory regions for the instantiation of `soc_interconnect.sv`. This module can be parametrized to generate as many AXI slave ports as you need. By default, there are 2 AXI slaves. One for the aforementioned s_data_out_bus that's used to attach a cluster (not in use for PULPissimo) and one that is converted to APB to communicate with the peripherals within the uDMA domain and the FLL. The memory region Macros used in soc_interconnect_wrap.sv are defined in `pulpissimo/rtl/includes/soc_mem_map.svh`.

jonesemmanuel, 2021-02-15:
@quangdaovu and @meggiman 
Thanks for the inputs. I will go through the codes you suggested and will get back in case of further questions.

Meanwhile, I have a question about the address range that can be used for the external hardware accelerator. If I create a new AXI slave bus and connect it to the the AXI XBAR, what address range can I use and where can I find these details?

pulpissimo/rtl/includes/soc_mem_map.svh mentions the address range for AXI-PLUG as 0x1000_0000 to 0x1040_0000.
But isnt this the address range assigned for the Cluster (pulpissimo/rtl/includes/pulp_soc_defines.sv has a comment that hints so).

meggiman, 2021-02-16:
Indeed, that is the address range for the cluster. You can use every address range not already declared in `soc_mem_map.svh`. The default behavior is to route all transactions to the AXI bridge/crossbar if they do not fall in any of the declared address ranges. Check the code in `soc_interconnect_wrap.sv` for the details. If you want to attach a new slave, just add any memory range not already in use to `soc_mem_map.svh` and use the macros you declared there within `soc_interconnect_wrap.sv` to pass them to the parametrizable interconnect (`soc_interconnect.sv`).

jonesemmanuel, 2021-02-16:
Thank you @meggiman.

I am still working to understand the code. Since I am a newbie, it will take some time.
I will try as you instructed and will get back in case of any issues.

closed 2021-03-16 by jonesemmanuel