pulp-platform/pulpissimo #225: HWPE stuck in second operation on Zedboard/apparently wrongly addressed tcdm accesses

marwinkirchhofs, 2021-02-20:
I'm currently implementing a first own HWPE for being able to work with the system. The functionality is dead-simple, it uses two TCDM ports, fetches one 32-bit word, performs a left shift on it and writes it to the result stream. The user-defined part of the register file only holds the data_in and result address which are directly forwarded to the streamer consisting of one source/sink and one fifo on the tcdm side per memory port. I left out the uloop processor for this first implementation. The functionality is meant to be extended later on. 
My c-code for the riscv is built mostly based on the example code from the pulp-rt-examples accelerator, including my headers.
 
I am currently simulating the system with a stripped down version of your hwpe-tb as on the one hand we currently have no suitable modelsim available, on the other hand generating the stim_instr/data.txt with the hwpe-tb's make target throws an error when compiling the crt0.S. Thus I can't use the full hwpe_tb. In my tb, I directly address the hwpe by using the hwpe_ctrl_intf_periph's read and write tasks. On the tcdm side, it simply grants every read and write access. It is a modified version of your hwpe_tb file. I'm using Vivado 2019.1 for simulation.

Considering IP versions, I use pulpissimo v6.0.0 while I tried different recent versions of the streamer/ctrl IP. Currently, I'm on hwpe-stream v1.6.3 and hwpe-ctrl v1.6.2.

By the way, I'm not referring to the rt-example's accelerator as it still gets stuck within the first operation as a lot of other people mentioned here. I tried the suggested solutions (using different versions of the streamer IP), but the behaviour didn't change.

Currently, there are two major problems:
1. When running it on the zedboard, the first time I operate the hwpe from my c-program, it terminates fine by triggering the event unit. In its second operation, it gets stuck. I encountered the same problem in simulation and solved it by clearing the HWPE after each complete operation with a write to its clear-register (otherwise, for example the stream-sink doesn't deassert its valid signal and if one solves this, the ctrl_slave triggers itself infinitely). Thus, my hwpe works fine in my simulation with multiple consecutive operations. On the zedboard, it seems like writing to the clear-register has no effect. I tried different locations for the write call (before/after acquiring, before/after cg_disable which is the last hwpe call of one operation), none of them works. I double-checked the address definition of the clear register with the specifications, your c-headers and also with the successfully running address in my testbench. Do you have an idea why this call could have no effect or what else could be the reason for the hwpe getting stuck?

2. Also during the first seemingly successful hwpe operation on the board, I encounter a strange behaviour concerning the dataflow. Although the termination flag is set correctly, there are either no or faulty tcdm accesses. The value in my allocated result data location does not change, but it should hold the shifted version of the input data. In gdb, I checked the data at the transferred memory addresses as well as the addresses written to the hwpe's register file, both are correct except of the non-happening write (and maybe also read) access by the hwpe. In simulation, the system behaves as intended, writing back the correct value with the correct adress. Is there maybe some kind of address resolvement between hwpe's tcdm ports and the phyical memory which would explain this? I tried setting my data_in and result values/addresses by directly defining the variables in the code as well as by allocating them using the rt_alloc_align function with PERIPH as memory usage type.

Additionally, I have/had two problems when writing to the user-define part of the register file (hwpe_params in ctrl_regfile_t, as far as I see it). First, a write access always caused the first 16 of those registers to take the value, not just the intended one (simulation and fpga). This seems to happen due to a non-working clock-gating in regfile_latch, causing always all these registers to be written. I simply bypassed this, afterwords the writing worked correctly in simulation and on the fpga. 
Furthermore, I understood these hwpe_params as representing the user-defined part of the register file which is not used for transferring bytecode from the riscv to the hwpe. hwpe_params[0] should then correspond to an address offset of 0x40. In my simulation, hwpe_params[0] is accessed by 0x20 as the address offset. Therefore, when using addresses starting at 0x40 for transferring my data/result addresses as your examples do, in the rtl code I have to use hwpe_params[9:8]. Is that intentional and if yes, what are the ctrl_regfile_t.generic_params then corresponding to? In the examples, it looks as those should be written to by the offsets 0x20-0x38. If not, can you think of a reason for the behaviour in my case? 
I thought that this issue could also affect my problem with the wrong tcdm accesses, but I had the same non-working dataflow before finding the hwpe_params behaviour, so to say when hwpe_params[15:0] all held my data_in address. That should at least have caused the result to be written back to the data_in's memory address.

I know that this is a lot of text, but there are simply several issues occuring and I wanted be as precise as possible. I don't know if I will be allowed to share any code here in public as it is technically property of the university... But I thought that these topic's discussion might be interesting also for other people.

marwinkirchhofs, 2021-02-21:
update:
I want to update this issue as I did some more testing today. I consider it now pretty likely that for some reason, my rtl code causes the hwpe to be "bypassed", meaning it automatically raises the software event when being called. I tested this by statically assigning a non-zero value to the 'result' stream output while simply writing the same address from software to all offsets from 0x20 to 0x60 (for sure in 4-byte steps). It still does not change the value at the given address which makes me think that the hwpe's functionality is not triggered.
I made sure that I only assert the ctrl_slave_o.done in the termination step of my control fsm and that it is tied to 0 in the always_comb block before the case statement.
I think that this needs proper (post-synthesis) simulation analysis, in the best case with the full hwpe-tb since the implementation is working fine in the smaller behavioral synthesis. I hope that we get our modelsim running this week and that I can then maybe update this issue with either a more precise question or a solution. Either way, if you nevertheless have an idea from this information or encountered something similar while developing I could check, I'll be glad to hear about that!

Concerning the error with the hwpe-tb, I was now able to obtain the stim_data/instr.txt files for the given example code with a defined environment. If there are any more problems, I think that rising a specific issue in the hwpe-tb will be more fitting.

yttuncel, 2021-10-05:
Hi @marwinkirchhofs,

I'm working on getting the default HWPE work on ZCU102, and am currently stuck as explained in #274.
I'm seeing the same or very similar behavior as you described in your initial post regarding the regfile.

Have you found time to investigate/debug the problem? Is the HWPE running correctly on FPGA on your end?