pulp-platform/pulpissimo #230: Generating bitstream for ZCU104

FrankieTankie, 2021-03-12:
Hello guys,
I just set up a fresh installation of the current pulpissimo repo to generate the bit stream for A ZCU104.
make zcu104 stops with error 2 due to missing ../pulpissimo/tcl/ips_inc_dirs.tcl
After comparing it with my old successful installation, it seems as the pulpissimo folder in /fpga folder is completely missing.
Seems to me like there is something broken in the repo. can anyone confirm this?
Regards,
Franky

FrankieTankie, 2021-03-12:
Just found the missing command:
in the readme "make checkout" is only mentioned under Building the RTL simulation platform. Therefore if you head directly to the FPGA section, you will miss it and won't be able to generate the bitstream. A hint in the FPGA section might be helpful.

FrankieTankie, 2021-03-13:
Meanwhile I ran into another problem. For some reason Vivado Synthesis seems to stuck. TCL console shows "Waiting on synth_1 to finish" for about three hours (normally it takes about 45 min to finish). Can anyone confirm that bitstream generation works with the latest PULPissimo release for ZCU104?

bluewww, 2021-03-13:
> Just found the missing command:
> in the readme "make checkout" is only mentioned under Building the RTL simulation platform. Therefore if you head directly to the FPGA section, you will miss it and won't be able to generate the bitstream. A hint in the FPGA section might be helpful.

PR with this change would be welcome

bluewww, 2021-03-13:
[fpga_synth_nexys_zcu104-master-ab1a4130.zip](https://github.com/pulp-platform/pulpissimo/files/6134779/fpga_synth_nexys_zcu104-master-ab1a4130.zip)

This is the zcu104 synthesis of our CI pipeline. It is a few commits back (https://github.com/pulp-platform/pulpissimo/commit/ab1a41309f55d9b7e0c1bc83bc6942938b1f4e2b). It should have the bitstream and some logs, maybe you can gleam something from it.

I'm running the CI with the latest commit now but I would not expect for anything to change.

FrankieTankie, 2021-03-13:
Hello blue,
Thank you for your effort. I just reinstalled vivado, for some reason it seems that it was broken. Now i was able to finish bitstream generation. So I can confirm it works with the latest release for the ZCU104. Therefore only the change of the readme is recommended from my side.
Regards,
Franky

PR #236 (2021-03-23): doc: Fix various issues
Fixes #234 
Fixes #230 
Fixes https://github.com/pulp-platform/pulpissimo/issues/209
Fixes https://github.com/pulp-platform/pulpissimo/issues/208
Fixes https://github.com/pulp-platform/pulpissimo/issues/181
Fixes https://github.com/pulp-platform/pulpissimo/issues/167
Fixes https://github.com/pulp-platform/pulpissimo/issues/153
closed 2021-03-23 by bluewww