pulp-platform/pulpissimo #231: Cadence Excellium: Assignment pattern - LHS with a type parameter datatype is not currently supported.

bbo49, 2021-03-15:
In the past I succeeded to simulate the platform using the SDK and Xcellium. However when I checkout the most recent version of Pulpissimo, building using `make build-incisive` fails with the following errors.

When initially compiling the most recent version I got the following errors:
```
file: /magics/design/tesla/codes/hdl/pulpissimo/ips/register_interface/src/reg_test.sv
    parameter int  AW     ,
                          |
xmvlog: *E,SVVMAP (/magics/design/tesla/codes/hdl/pulpissimo/ips/register_interface/src/reg_test.sv,18|26): Omitting the assignment to constant_param_expression in a param_assignment in a parameter_port_list currently is not supported.
    parameter int  DW     ,
                          |
xmvlog: *E,SVVMAP (/magics/design/tesla/codes/hdl/pulpissimo/ips/register_interface/src/reg_test.sv,19|26): Omitting the assignment to constant_param_expression in a param_assignment in a parameter_port_list currently is not supported.
```
These got resolved by setting the environment variable `CADENCE_ENABLE_AVSREQ_44905_PHASE_1` before invoking the `make build-incisive`. According to Cadence Support this works starting from Xcellium 19.09.

After applying this fix, the following errors appear:
```
  `AXI_LITE_ASSIGN_TO_REQ(axi_req, slv)
                                      |
xmelab: *E,APNOTP (../ips/axi/axi/src/axi_lite_to_apb.sv,393|38): Assignment pattern - LHS with a type parameter datatype is not currently supported.
  `AXI_LITE_ASSIGN_TO_REQ(axi_req, slv)
                                      |
xmelab: *E,APNOTP (../ips/axi/axi/src/axi_lite_to_apb.sv,393|38): Assignment pattern - LHS with a type parameter datatype is not currently supported.
  `AXI_LITE_ASSIGN_TO_REQ(axi_req, slv)
                                      |
xmelab: *E,APNOTP (../ips/axi/axi/src/axi_lite_to_apb.sv,393|38): Assignment pattern - LHS with a type parameter datatype is not currently supported.
  assign axi_lite_resp_o = '{
                            |
xmelab: *E,APNOTP (../ips/axi/axi/src/axi_lite_to_apb.sv,125|28): Assignment pattern - LHS with a type parameter datatype is not currently supported.
            apb_req_o[apb_sel_idx] = '{
                                      |
xmelab: *E,APNOTP (../ips/axi/axi/src/axi_lite_to_apb.sv,250|38): Assignment pattern - LHS with a type parameter datatype is not currently supported.
        apb_req_o[apb_sel_idx] = '{
                                  |
xmelab: *E,APNOTP (../ips/axi/axi/src/axi_lite_to_apb.sv,276|34): Assignment pattern - LHS with a type parameter datatype is not currently supported.
```

Is there an easy workaround for this problem? I'm using Xcellium 20.03-s007.

bluewww, 2021-03-15:
@accuminium Looks like an AXI issue. Do you know anything about this?

bbo49, 2021-03-25:
When Identifying the problematic lines, it looks like Xcelium doesn't support assigning a `assignment pattern` (`` `{field1: value1, field2: value2}``) in places where the Left Hand Side (LHS) of the assignment is a variable with a data type inherited from a type parameter. 
So in short, the following is not supported in Xcelium 20.03-s007:
``assign <variable_with_data_type_inherited_from_a_type_parameter> = `{assignment_pattern};``

A clear example are the last 3 erroneous lines from the log in my first post. The first of them is the following line https://github.com/pulp-platform/axi/blob/2c5f98813dc6ac5eac6d39e869bdc2caab486a72/src/axi_lite_to_apb.sv#L125. The data type of `axi_lite_resp_o` is `axi_lite_resp_t`, which is defined as a parameter type in the parameters section of that module

A workaround for this issue is replacing these erroneous assignment patterns, with equivalent struct assignments.
For example for the line mentioned in the link above, the following fix can be done:
``` diff
-  assign axi_lite_resp_o = '{
-    aw_ready: axi_req_valid[WR] & axi_req_ready[WR],        // if AXI AW & W valid & tree gnt_o[WR]
-    w_ready:  axi_req_valid[WR] & axi_req_ready[WR],        // if AXI AW & W valid & tree gnt_o[WR]
-    b:       '{resp: axi_bresp},                            // from spill reg
-    b_valid:  axi_bresp_valid,                              // from spill reg
-    ar_ready: axi_req_valid[RD] & axi_req_ready[RD],        // if AXI AR valid and tree gnt[RD]
-    r:       '{data: axi_rresp.data, resp: axi_rresp.resp}, // from spill reg
-    r_valid:  axi_rresp_valid                               // from spill reg
-  };
+  assign axi_lite_resp_o.aw_ready = axi_req_valid[WR] & axi_req_ready[WR];         // if AXI AW & W valid & tree gnt_o[WR]
+  assign axi_lite_resp_o.w_ready  = axi_req_valid[WR] & axi_req_ready[WR];         // if AXI AW & W valid & tree gnt_o[WR]
+  assign axi_lite_resp_o.b.resp   = axi_bresp;                            // from spill reg
+  assign axi_lite_resp_o.b_valid  = axi_bresp_valid;                               // from spill reg
+  assign axi_lite_resp_o.ar_ready = axi_req_valid[RD] & axi_req_ready[RD];         // if AXI AR valid and tree gnt[RD]
+  assign axi_lite_resp_o.r.data   = axi_rresp.data; // from spill reg
+  assign axi_lite_resp_o.r.resp   = axi_rresp.resp;
+  assign axi_lite_resp_o.r_valid  = axi_rresp_valid;                               // from spill reg
+  
```
Repeating this approach on the last 3 lines fixes these issues.

For fixing the first 3 errors, I needed to do similar changes in the assignment macros( https://github.com/pulp-platform/axi/blob/v0.24.1/include)/axi/assign.svh)

After doing so, also some similar issues appeared in https://github.com/pulp-platform/common_cells/blob/master/src/id_queue.sv. These can be resolved by following the same strategy.

After this the platform compiles fine and I can simulate hello world application without any issues in Xcelium. 
Should I open a Pull Request in the affected repositories?

meggiman, 2021-06-07:
Hi @bbo49
Sorry for the late reaction. If this is still relevant for you, we would appreciate a PR if it improves tool compatibility. @accuminium might comment on this as he is the person responsible for https://github.com/pulp-platform/axi.

accuminium, 2021-06-07:
> Should I open a Pull Request in the affected repositories?

Yes, that would be highly appreciated!