pulp-platform/pulpissimo #232: Illegal instruction in simulation with USE_PULP_BUS_ACCESS = 0

shivampotdar, 2021-03-18:
I was trying to run the testbench with `USE_PULP_BUS_ACCESS=0` (as we want to replicate FPGA behavior in our current application).
But in simulation an illegal instruction is received at `0x1c0083b8`
The log says - 
```
# [TB] 22933401ns - Resuming the CORE
# 22965775ns: Illegal instruction (core 0) at PC 0x1c0083b8:
# Instruction is valid, but has at least one X
```
Also this PC does not exist in `trace_core_1f_0.log` (jal after 83b4)
```
22965661ns          403186 1c0083b0 00e6a023 sw               x14, 0(x13)         x14:ffffffff x13:1a10601c  PA:1a10601c
22965718ns          403187 1c0083b4 02078793 addi             x15, x15, 32        x15=1a106020 x15:1a106000
22966172ns          403195 1c008000 33c0006f jal              x0, 828            
```

This is the "hello" test from https://github.com/pulp-platform/pulp-runtime-examples (running with PULP Runtime)

Although I have made some additions in the RTL, I don't think so anything should be breaking this part. 
Can somenone please help me figure out what might be wrong here.

shivampotdar, 2021-03-18:
From the objdump of the binary, this is the return of pos_soc_init.
Also in the trace, it jumps to irq_vector_base before this return, so I believe it is supposed to be squashed but something is broken there.

prajwalkashyap, 2023-06-24:
@shivampotdar  How did you solve this issue ?