pulp-platform/pulpissimo #235: Basic doubt in working of Debug Module

shivampotdar, 2021-03-23:
I have a conceptual question on how the system works when in debug mode on an FPGA. 
I see that in simulation there is a debug ROM #221  which is looped upon while the testbench loads L2 with the binary. And then a resume request is sent which directly executes the program binary.

In case of an FPGA, I was wondering that since we never resume (debugger is always connected), is it that between every step (or communication between the core and debugger), the core loops on some set of instructions internally? If yes, is this behavior affected by the way breakpoints are set? 
I tried to go through the RISC-V Debug specs but could not find an answer to this. 

This concerns me as we are working on a monitoring (trace + graph traversal) unit which works correctly on the simulation platform but is possibly giving unexpected results because of the debug loop.

Please let me know if the question lacks clarity on any aspect.
Thanks!

bluewww, 2021-03-23:
> I have a conceptual question on how the system works when in debug mode on an FPGA.
> I see that in simulation there is a debug ROM #221 which is looped upon while the testbench loads L2 with the binary. And then a resume request is sent which directly executes the program binary.
Indeed
> In case of an FPGA, I was wondering that since we never resume (debugger is always connected), is it that between every step (or communication between the core and debugger), the core loops on some [...]
The way the debug module is implemented in RI5CY (and ariane and rocket) is that the core is held in a busy loop in the debug rom, where the debug module dynamically injects instructions to make the core read/write addresses/registers. This implementation is outlined in risc-v debug spec under `A.2 Execution Based [Hardware Implementation]`
> 
> This concerns me as we are working on a monitoring (trace + graph traversal) unit which works correctly on the simulation platform but is possibly giving unexpected results because of the debug loop.
> 
> Please let me know if the question lacks clarity on any aspect.
> Thanks!

shivampotdar, 2021-03-23:
Hello @bluewww thanks a lot for the detailed and quick reply. 
So if I understand correctly [this](https://github.com/pulp-platform/riscv-dbg/blob/master/debug_rom/debug_rom.S) is our defined memory region as per the spec. And [these](https://github.com/pulp-platform/riscv-dbg/blob/master/doc/debug-system.md#debug-memory-map) (halted, going, resuming) are the memory locations being polled by the hart.

In simulation since we resume before executing the binary compiled by runtime/SDK, this loop never occurs thereafter.

My small followup on this is when I perform load (from GDB) while connected to FPGA and continue without breakpoints, would the flow still be same or is it that the binary is loaded and executed in parts and you would still have the "park loop" in middle of the binary's execution. 
Our application just expects the binary to be executed in order (like it does in simulation) and I am not sure if this is the case on FPGA as well. 

Thanks again for your time.

bluewww, 2021-03-23:
> Hello @bluewww thanks a lot for the detailed and quick reply.
> So if I understand correctly [this](https://github.com/pulp-platform/riscv-dbg/blob/master/debug_rom/debug_rom.S) is our defined memory [...]

Yes

> 
> In simulation since we resume before executing the binary compiled by runtime/SDK, this loop never occurs thereafter.
> 

We are only really engaging the debug module at the very start when loading the binary into the RAM

> My small followup on this is when I perform load (from GDB) while connected to FPGA and continue without breakpoints, would the flow still be same or is it that the binary is loaded and executed in [...]

GDB uses software breakpoints with RI5CY so it will replace the next instruction with an `ebreak` when single stepping, sending the core into the park loop.

> 
> Thanks again for your time.

shivampotdar, 2021-03-23:
I see, so without single step, the execution should go uninterrupted.
Thanks for the details!

closed 2021-03-23 by shivampotdar