pulp-platform/pulpissimo #237: Synthesized Xilinx IPs not found with Vivado 2020.2

accuminium, 2021-03-25:
With Vivado 2020.2.1, `make zedboard` (in the `fpga/` directory) fails with
```
ERROR: [Synth 8-439] module 'xilinx_slow_clk_mngr' not found [/path/to/pulpissimo/fpga/pulpissimo-zedboard/rtl/fpga_slow_clk_gen.sv:48]
ERROR: [Synth 8-6156] failed synthesizing module 'fpga_slow_clk_gen' [/path/to/pulpissimo/fpga/pulpissimo-zedboard/rtl/fpga_slow_clk_gen.sv:24]
ERROR: [Synth 8-6156] failed synthesizing module 'safe_domain' [/path/to/pulpissimo/rtl/pulpissimo/safe_domain.sv:12]
ERROR: [Synth 8-6156] failed synthesizing module 'pulpissimo' [/path/to/pulpissimo/rtl/pulpissimo/pulpissimo.sv:13]
ERROR: [Synth 8-6156] failed synthesizing module 'xilinx_pulpissimo' [/path/to/pulpissimo/fpga/pulpissimo-zedboard/rtl/xilinx_pulpissimo.v:23]
```
even though the `xilinx_slow_clk_mngr` IP has previously been synthesized, see [`xilinx_slow_clk_mngr.log`](https://github.com/pulp-platform/pulpissimo/files/6203566/xilinx_slow_clk_mngr.log).

One of the last messages in that log mentions
```
WARNING: [Common 17-576] 'use_project_ipc' is deprecated. This option is deprecated and no longer used.
```

Could it be that the IP pre-synthesis and caching flow of Vivado that PULPissimo seems to use has changed with Vivado 2020.2? It might help to compare [UG896 for 2020.2](https://www.xilinx.com/support/documentation/sw_manuals/xilinx2020_2/ug896-vivado-ip.pdf), Chapter 2, with the instructions for previous Vivado versions and review the [Changelog for 2020.2, especially the *IP Integrator* section](https://www.xilinx.com/products/design-tools/vivado/vivado-whats-new.html#integrator).

The problem does not occur with Vivado 2020.1.

MiguelCosta94, 2021-04-13:
Hi @accuminium 

Have you already solved the problem? I am facing the same issue with ZCU104 and Vivado 2020.1.

accuminium, 2021-04-13:
Hi @MiguelCosta94, thanks for your question.

> Have you already solved the problem?

I reported this issue, but I am not working on the solution.  The workaround for me is to use Vivado 2020.1.  @meggiman might be able to give us an update.

> I am facing the same issue with ZCU104 and Vivado 2020.1.

Are you sure about Vivado 2020.**1**?  2020.1 works for me, whereas 2020.2 does not.

MiguelCosta94, 2021-04-14:
Yes. I was not able to generate the bitstream for ZCU104 with Vivado 2020.1. However, Vivado 2019.2 solved my problem.

meggiman, 2021-04-19:
I will have a look into it today. I will keep you updated.

PR #246 (2021-04-26): Fix fpga ports to be compatible with vivado 2020.2
Fixes #237
closed 2021-04-26 by meggiman