pulp-platform/pulpissimo #238: Running hello application for ZedBoard FPGA ( no thing displayed with screen /dev/ttyUSB0 115200)

mohamedElbouazzati, 2021-03-29:
## i'm running hello application for ZedBoard FPGA and no thing displayed with  screen /dev/ttyUSB0 115200

### Configuration 

- PULP-RISCV-TOOLCHAIN ( Pre-built) 
- PULP-SDK 
- Processor : CV32E40P
- Soc : Pulpissimo 
- Board: ZedBoard FPGA

### JTAG and UART Connections: 

| JTAG Signal | PMOD Pin |
|-------------|----------|
| TMS         | JA1      |
| TDI         | JA2      |
| TDO         | JA3      |
| TCK         | JA4      |
| GND         | JA5      |
| VCC (trgt)  | JA6      |

| UART Signal | PMOD Pin |
|-------------|----------|
| RXD         | JA7      |
| TXD         | JA8      |
| RTS         | JA9      |
| CTS         | JA10     |
| GND         | JA11     |

### C Program 

[image]

### Openocd

[image]
### GDB 

[image]

### screen /dev/ttyUSB0 115200

[image]

I don't know why nothing is displayed on the screen. 
Any suggestions are appreciated 
Thanks!

meggiman, 2021-04-22:
Hi @mohamedElbouazzati 
Did you try stepping through the printf call using GDB to ensure that the program is actually trying to print to uart? Did you check that there is actually any activity on the UART pins (e.g. with an oscilloscope)?

mohamedElbouazzati, 2021-04-27:
Hi @meggiman  I tried to do this on a fpga zcu104 and didn't get enough clarity on the uart like this on putty 
[image]
I have to set the baud rate somewhere in the program, I guess.

Ahmed1Zaky, 2023-02-07:
Hi @mohamedElbouazzati,

Could you please tell me how did you connect the PMOD to the PC? 

Did you use this lvl shifter board along with the HS2 adapter? https://digilent.com/reference/pmod/pmodlvlshft/reference-manual

or is there any other way to do it? Please advise.

Ahmed1Zaky, 2023-03-02:
Hi @mohamedElbouazzati , 

Did you end up solving this issue? Did you manage to get the output printed on the Terminal screen?

zealshah29, 2024-03-26:
Hi @mohamedElbouazzati , @meggiman ,
I am facing the same issue. I also added the breakpoint before continue, yet I can't see any output. If the following issue has been solved, can you help me out with the same?
Also is there any other way to verify if softcore dump was successful on zedboard or not?
I have also created an issue specifying my problem in detail : #409

closed 2024-04-05 by mohamedElbouazzati