pulp-platform/pulpissimo #240: Vivado version to generate bitstream for ZCU104

MiguelCosta94, 2021-04-13:
Hi, 

I'm facing some errors while generating the PULPissimo bitstream for ZCU104. I have already tested Vivado 2020.2 and 2020.1 but both return the same errors:

`ERROR: [Synth 8-439] module 'xilinx_slow_clk_mngr' not found [/home/miguel/PULPissimo/pulpissimo/fpga/pulpissimo-zcu104/rtl/fpga_slow_clk_gen.sv:48]
ERROR: [Synth 8-6156] failed synthesizing module 'fpga_slow_clk_gen' [/home/miguel/PULPissimo/pulpissimo/fpga/pulpissimo-zcu104/rtl/fpga_slow_clk_gen.sv:24]
ERROR: [Synth 8-6156] failed synthesizing module 'safe_domain' [/home/miguel/PULPissimo/pulpissimo/rtl/pulpissimo/safe_domain.sv:12]
ERROR: [Synth 8-6156] failed synthesizing module 'pulpissimo' [/home/miguel/PULPissimo/pulpissimo/rtl/pulpissimo/pulpissimo.sv:13]
ERROR: [Synth 8-6156] failed synthesizing module 'xilinx_pulpissimo' [/home/miguel/PULPissimo/pulpissimo/fpga/pulpissimo-zcu104/rtl/xilinx_pulpissimo.v:22]`

Here you can see the Vivado log file: 
[vivado.log](https://github.com/pulp-platform/pulpissimo/files/6305403/vivado.log)

Which Vivado version should I use to be able to generate the bitstream for ZCU104?

MiguelCosta94, 2021-04-14:
I was able to generate the bitstream for ZCU104 with Vivado 2019.2.

closed 2021-04-14 by MiguelCosta94

MohamedElbouazzati-ensem, 2021-04-16:
Hello @MiguelCosta94  i found the same error with 2020.2 and 2019.2 versions of vivado 

`ERROR: [Synth 8-439] module 'xilinx_slow_clk_mngr' not found [/home/med/Desktop/openHWG/core-v-mcu/fpga/pulpissimo-zcu104/rtl/fpga_slow_clk_gen.sv:48]` 
`ERROR: [Synth 8-6156] failed synthesizing module 'fpga_slow_clk_gen' [/home/med/Desktop/openHWG/core-v-mcu/fpga/pulpissimo-zcu104/rtl/fpga_slow_clk_gen.sv:24]`
`ERROR: [Synth 8-6156] failed synthesizing module 'safe_domain' [/home/med/Desktop/openHWG/core-v-mcu/rtl/pulpissimo/safe_domain.sv:12]`
`ERROR: [Synth 8-6156] failed synthesizing module 'pulpissimo' [/home/med/Desktop/openHWG/core-v-mcu/rtl/pulpissimo/pulpissimo.sv:13]`
`ERROR: [Synth 8-6156] failed synthesizing module 'xilinx_pulpissimo' [/home/med/Desktop/openHWG/core-v-mcu/fpga/pulpissimo-zcu104/rtl/xilinx_pulpissimo.v:22]`

how did you fix this with the 2019.2 version with on ZCU104 board 
---------------------------------------------------------------------------------

MiguelCosta94, 2021-04-16:
Hi @MohamedElbouazzati-ensem 

Can you check if you have installed Vivado 2019.2.1 or Vivado 2019.2.0? I have not tested Vivado 2019.2.1.

MohamedElbouazzati-ensem, 2021-04-16:
Thank you @MiguelCosta94 for your answer , yes i tried also with 2019.2.1 i got the same error, have you done other changes ?