pulp-platform/pulpissimo #241: OpenOCD issue with ZCU104

MiguelCosta94, 2021-04-15:
Hi,

As explained in the README file, I executed the command `source sourceme.sh && ./pulp-tools/bin/plpbuild checkout build --p openocd --stdout` to install a patched version of OpenOCD. However, I got the following error:
`In function ‘memset’,
    inlined from ‘mg_gen_ataid’ at src/flash/mflash.c:1162:2,
    inlined from ‘mg_storage_config’ at src/flash/mflash.c:1174:2:
/usr/include/x86_64-linux-gnu/bits/string_fortified.h:71:10: error: ‘__builtin_memset’ offset [509, 512] from the object at ‘buff’ is out of the bounds of referenced subobject ‘reserved7’ with type ‘mg_io_uint8[186]’ {aka ‘unsigned char[186]’} at offset 322 [-Werror=array-bounds]
`
When I tried to solve the problem I come across this issue which mentions that a patched version of OpenOCD is not required now: [issue](https://github.com/pulp-platform/pulpissimo/issues/210)

However, after installing the riscv-openocd from the riscv repository [riscv-openocd](https://github.com/riscv/riscv-openocd), I am still not able to connect to ZCU104 with the Olimex arm-usb-tiny-h. When I use openocd-zcu104-olimex-arm-usb-tiny-h.cfg from $PULPISSIMO_DIR/fpga/pulpissimo-zcu104, I get this error:
`Error: no device found
Error: unable to open ftdi device with vid 15ba, pid 002b, description 'Olimex OpenOCD JTAG ARM-USB-OCD-H', serial '*' at bus location '*'
`
When I run OpenOCD as: `openocd -f ($riscv-openocd)/tcl/interface/ftdi/olimex-arm-usb-tiny-h.cfg -f ($riscv-openocd)/tcl/target/xilinx_zynqmp.cfg`, OpenOCD issues this warning over and over:
`Warn : Haven't made progress in mpsse_flush() for...`

The JTAG interface is listed in my devices as: `Bus 001 Device 005: ID 15ba:002a Olimex Ltd. ARM-USB-TINY-H JTAG interface`

How can I solve this issue and be able to program and debug application in PULPissimo for ZCU104?

closed 2021-04-15 by MiguelCosta94

marrrk, 2021-06-15:
Hi, can you please share how you solved this issue? I'm facing the same thing with the Nexys 7 A7 100T.
Thanks

MiguelCosta94, 2021-06-16:
Hi @marrrk,
I am not sure already how I solved the problem, but try to use this file as the second cfg file when calling the OpenOCD command "pulpissimo/fpga/pulpissimo-genesys2/openocd-genesys2.cfg"

marrrk, 2021-06-18:
Thanks for this @MiguelCosta94, I was able get around it.
As usual I have faced another problem. Just out of curiosity how long did it take for anything to happen once entering the command `continue` in the gdb server? I'm trying to see the hello world example but it appears to be hanging.

MiguelCosta94, 2021-06-21:
Hi @marrrk . It was almost instantaneous.

hellstorm666666, 2025-03-28:
Hello @marrrk, I know it's been so long but have you solved this problem?