pulp-platform/pulpissimo #245: Run time error on FPGA and Xilinx Vivado After Spi Flash Boot Process

LatifAkcayGithub, 2021-04-22:
Hello,

We are trying to boot a C app from SPI flash. Interestingly, when we make a behavioral simulation with modelsim, we see that all processes run well (boot code, spi flash boot process etc). But, when we try this with Xilinx Vivado (for post-synthesis sim), we see that the processor jump a different point from the one in modelsim and so the code does not run correctly. The scenario is the same when we try it on Genesys-2 board. I mean, bootcode runs well and application is taken from flash succcessfully. But then, it does not run correctly. We could not find from where the error comes out. 

I have added two screenshots to the attachment that shows the different behavior on simulators.

Any idea makes us happy. 

Thanks. 

[image]
[image]

meggiman, 2021-04-22:
Hi @LatifAkcayGithub 
Did you actually generate the bootrom for the FPGA? At the moment, the bootrom is not mapped to the FPGA and just traps the core in a busy loop (have a look at fpga/<your board>/rtl/fpga_bootrom.sv. This is a missing feature of the FPGA port and should probably have been mentioned in the FPGA specific readme section.
If you want to emulate SPI booting on the FPGA, you have to generate the bootrom yourself. The toplevel boot_code directory contains the sourcecode and the scripts to generate the SystemVerilog model of the ROM. You might want to extend this with some scripts to generate e.g. a mem file for a Xilinx parametrized ROM macro.

meggiman, 2021-04-26:
Not really. Maybe the core receives an interrupt? However, I think by default the runtime does not enable any interrupts on init. You should observe the instructions fetched by the core in the FPGA and, together with the disassembly of your program (compile it with 'make dis' to generate it) you can then observe which branches the core takes and why. In order to figure out the execution flow in the FPGA you could use GDB and step through execution. However be careful since you cannot step over hardware loop instructions due to some bug in the RISC-V debug unit, there you should use breakpoints instead. Alternatively, if that does not work (i.e. the core does not respond anymore) Xilinx ILA cores (in-logic analyzer IP you can insert in Vivado) are an invaluable tool to observe internal FPGA signals in real-time. 
Best,
Manuel

LatifAkcayGithub, 2021-05-01:
Hi again,

I think, we have solved the problem. 
Actually, it was our mistake. The error comes out because of the wrong FLL frequency set in soc.c
It must be determined for the target platform. As the work done by different guys, the parameter was set wrong.
@meggiman Thanks for your interest again.

closed 2021-05-01 by LatifAkcayGithub