pulp-platform/pulpissimo #249: Debug takes too much time to start

MiguelCosta94, 2021-04-27:
Hi,

I am trying to debug an application which uses only 210KB of L2 RAM. However, the debugger takes too much time (more than 10 minutes) to reach the start address of the application. I have two large arrays defined in the .data section (one with 79KB and another with 61KB). If I remove one these arrays, the debugging session starts immediately. However, the same is not verified when both are in memory. Is there any debug limitation that I am missing?

bluewww, 2021-04-27:
Can you clarify your setup?
Are you debugging with openocd on an FPGA?
Are you debugging by running the RTL simulation?

MiguelCosta94, 2021-04-27:
Thanks for your reply @bluewww. I am debugging on a ZCU104 with OpenOCD.