pulp-platform/pulpissimo #255: Post Synthesis Simulation Fails in RI5CY Core

vignajeth, 2021-05-11:
Hi,
   I am trying to make the pulpissimo with RI5CY core to work with the timing. I notice that the pulpissimo breaks completely with unknown values in most of the signals, especially signals related to memories. When I debugged the issue, I find the **riscv_register_fie** is causing me Timing troubles. To make my issue clear I have attached images of my simulation in various configurations.

**Behavioural Simulation**
[image]
[image]

**Synthesized Netlist Simulation - NO SDF Added**
[image]
[image]

**Synthesizes Timing Simulation - SDF Added**
 I have removed path delays by adding the +nospecify flag, without this the clock gating cell is not giving output clock, have you faced such an issue?
[image]

I have added multicycle path as per this thread [https://github.com/pulp-platform/pulpissimo/issues/77](https://github.com/pulp-platform/pulpissimo/issues/77)
```
set_multicycle_path 2 -setup -through [get_pins soc_domain_i/pulp_soc_i/fc_subsystem_i/lFC_CORE/id_stage_i/registers_i/riscv_register_file_i/mem_reg*/Q]
set_multicycle_path 1 -hold  -through [get_pins soc_domain_i/pulp_soc_i/fc_subsystem_i/lFC_CORE/id_stage_i/registers_i/riscv_register_file_i/mem_reg*/Q]
set_multicycle_path 2 -setup -through [get_pins soc_domain_i/pulp_soc_i/fc_subsystem_i/lFC_CORE/id_stage_i/registers_i/riscv_register_file_i/mem_fp_reg*/Q]
set_multicycle_path 1 -hold  -through [get_pins soc_domain_i/pulp_soc_i/fc_subsystem_i/lFC_CORE/id_stage_i/registers_i/riscv_register_file_i/mem_fp_reg*/Q]
```

If you feel this information is vague, let me know I can give a few more details

vignajeth, 2021-05-12:
@bluewww @FrancescoConti @meggiman any thoughts where the post-synthesis timing simulation might have gone wrong?

bluewww, 2021-05-12:
I'm not aware where it could possibly break. It might be your multicycle constraints.

Looks like you need to dig properly into the problem (backtracking wher ethe X comes from, understanding what the constraints do to your design etc.)