pulp-platform/pulpissimo #260: Running Pulpissimo Example in Nexys Video FPGA

ruschigo, 2021-06-14:
Hi! I'm Igor, From Caninos-Loucos.

We actually running tests on Pulpissimo!

Board: Digilent Nexys Video
SOC: Pulpissimo

We was able to synthetize Pulpissimo, install toolchain(Pulp Option), and install sdk.

in the bashrc the following Paths was add

> export PULP_RISCV_GCC_TOOLCHAIN=/opt/riscv
> export PATH=$PULP_RISCV_GCC_TOOLCHAIN/bin:$PATH
> 
> export OPENOCD=/home/ruschigo/Documentos/Pulpissimo/repositorio/pulpissimo/pulp-sdk/pkg/openocd/1.0/

the only thing i must to add was the environment variable OPENOCD, the SDK do not set it automatically in my case.

Using Vivado we could Synthetize Pulpissimo in NEXYS VIDEO using hardware manager tool.

to compile the application we use the following steps outlined on git README

set up the necessary environment variables with
`
source env/pulpissimo.sh`

in the SDK's root directory, set up SDK

```
source configs/pulpissimo.sh
source configs/fpgas/pulpissimo/genesys2.sh 
```
Using Hello Example to start with we get stucked

I just add big counter to do a easy track of serial signals

```
#include <stdio.h>
#include <rt/rt_api.h>

int __rt_fpga_fc_frequency = 20000000; // e.g. 20000000 for 20MHz;
int __rt_fpga_periph_frequency = 10000000; // e.g. 10000000 for 10MHz;

int main()
{
	int i = 0;
	while(i<100000){
  		printf("Hello !\n");
		i++;
	}
	printf("Finish!\n");

  return 0;
}
```

To do a Easy debug i'm Using the Bitstream of Nexys Video from github Releases.

I could succesfull compile using the 

`make clean all`

So here i found the first thing i could not understand, when a do the 

`$make dis`

I get a lot of alerts in the fom ELF file, check ahead the start o disasembly;

```
/opt/riscv/bin/riscv32-unknown-elf-objdump -Mmarch=rv32imfcxpulpv2 /home/ruschigo/Documentos/Pulpissimo/repositorio/pulp-rt-examples/hello/build/pulpissimo/test/test -d
[... 1191 lines cut ...]
> (gdb) continue
> Continuing.
> 

of course there are no signals in serial.

When i broke the conection of gdb with Ctrl+C, i got the following messages;

> unable to halt hart 992
>   dmcontrol=0x83e00001
>   dmstatus =0x00030c82
> Remote failure reply: E0E
> Remote failure reply: E0E

The only hart it could not halt, is the only one that is enabled...

In OpenOCD i got this;

> Error: unable to halt hart 992
> Error:   dmcontrol=0x83e00001
> Error:   dmstatus =0x00030c82

What step we are missing? or doing wrong?

Best Regards, Igor Ruschi

marrrk, 2021-06-21:
Hi @ruschigo, have you by any chance been able to solve your issue?

ruschigo, 2021-06-22:
Hi @marrrk , not yet, i still don't know how to solve it...

marrrk, 2021-06-23:
I too had a similar problem, and it seems like #238 was facing the same thing.I had a look at the pulp platform forum and this post https://pulp-platform.org/community/showthread.php?tid=256 seemed to solve it. Perhaps it could help to try it out.

meggiman, 2021-06-23:
Hi @ruschigo , @marrrk,

So let me go through your questions step-by-step:

First, what do you mean with "alerts" in the assembly? The disassembly looks perfectly fine to me. The thing you see in the snippet starting at address 1c008000 is the interrupt vector table which contains jumps to a function called __rt_illegal_instr which tells the core what to do when it encounters an illegal instruction (thus raising the corresponding Interrupt). The other interrupts are not handled by default (you can however install your own IRQ handler if you want) which is why these jump to a function called __rt_no_irq_handler which basically does nothing at all and clears the interrupt. If there is anything else that looks concerning to you that I might have missed please let me know.

Next, yes, the only hart that should be enabled in pulpissimo is hart 992 which is the hart id assigned to the risc-v core within pulpissimo (the reason we use 992 has to do with legacy reasons when the core is instantiated in a setup with multiple  clusters containing multiple cores). 

About the warning in gdb on "target-supplied registers are not supported": Just ignore it, it does not matter.

Now on to the real issue: I can't say why execution does not seem to work. My guess is that the program actually starts executing but at some point crashes during runtime initialization before it enters the main() function. As soon as that happens, the core will no longer be responsive to debug commands which is why you don't manage to break execution with Ctrl+C. Try to debug your application from the very beginning (**which is not main()**). Set a breakbpoint on the label `_entry` (`b _entry`). If you use the pulp-runtime instead of pulp-sdk I believe the entrypoint is called  `_stext` but your disassembly tells me that you are using pulp-sdk so `_entry` is the right one for you.
Afterwards, step through your program instruction by instruction using `si`. However, be careful about hardware loop instructions (the ones called `lp.setup ...`  in the disassembly). These are a bit nasty because the debug unit cannot single step through them. There you need to set a breakpoint after the loop **body** and use `continue` to jump over it. Also you might want to enable GDB's TUI mode (https://www.youtube.com/watch?v=RM3xwFxr9-Q) to see disassembly, and source code in split screen during debugging.
With this approach you should be able to pinpoint, where your application crashes. If you are not able to break on `_entry` (e.g. the sequence `load`, `b _entry`, `continue` does not break `) your application is really not executing at all and you have to investigate other potential issues. Actually, since you are able to communicate with the core using GDB and load the programm I don't think that is the issue. Most likely you somehow compiled your application with vitual stdout instead of UART output. By default it should use UART for printf statements when you source the FPGA configuration **before** compilation with the pulp-sdk/pulp-runtime. However, for RTL simulation it uses the virtual stdout bus. This bus does not exist in actual hardware which will cause the core to hangup when it tries to communicate with it on the FPGA.

ruschigo, 2021-06-24:
Hi @marrrk  @meggiman 

@meggiman thank you very much for your explanation, it was very important to me, using your guess i was able to run the hello example!

using the breakpoint on the label and running the application instruction by instruction is possible to see that application go far away from label _entry, so using the the forum question pointed for @marrrk i force the output to be uart using the flag "io=uart" during compiling, than i had some trash in serial, but the processor still works fine, then with Ctrl+C  GDB was able to Exit properly. The second stuff was that processor clock in 20MHZ and peripheral clock in 10MHZ could not be able to provide properly serial signals, i saw the default clock from pulpissimo/fpga/pulpissimo-nexys_video/README.md it should be 10MHZ for clock and 5MHZ to peripheral clock.

Resuming the solution is using the flag "io" compile the application;  
`$make clean all io=uart`
And use Default processor clock of 10MHZ and 5Mhz for peripherals

now i can proceed i with other tests with uart and all other peripherals of pulpissimo.

*I will update this comment later to insert where the application break, i could be interesting know why uart is not the default output, probably the soc is using the virtual stdout bus as you said before.

Thanks you!

Best Regards, Igor Ruschi

CWerberger, 2021-06-28:
Hello @meggiman 

you mentioned tui support. As far as i checked -tui is not supported by the pulp-toolchain gdb. Can you give me and others a hint how to enable tui?

meggiman, 2021-06-28:
Hi @CWerberger 

I cannot confirm this. I just checked the precompiled releases of version 1.0.16 on both CentOS and Ubuntu. For both of them, tui mode (risc32-unknown-elf-gdb -tui) works just fine.

CWerberger, 2021-06-28:
Hi @meggiman 

Thank you for this hint with the precompiled chain. It's works for me now!
I compiled the toolchain myself as described here https://github.com/pulp-platform/pulp-riscv-gnu-toolchain. Seems so that you get no support for tui this way...

meggiman, 2021-06-29:
It's an optional feature of GDB. You need to enable it during compilation. (`./configure --enable-tui=yes`). Glad it works now.