pulp-platform/pulpissimo #266: Cadence Xcelium - Assignment pattern - LHS with a type parameter datatype is not currently supported.

enqnwhgdk317, 2021-08-03:
Hello. I'm trying RTL simulation of files, but axi_lite_to_apb.sv has an error during compilation.
I'm using xcelium 19.03-s009.

in axi_lite_to_apb.sv line 329~336,

      Access: begin
        // `Access` step
        apb_req_o[apb_sel_idx] = '{
          paddr:   apb_req.addr,
          pprot:   apb_req.prot,
          psel:    1'b1,
          penable: 1'b1,
          pwrite:  apb_req.write,
          pwdata:  apb_req.data,
          pstrb:   apb_req.strb
        };

Assignment pattern - LHS with a type parameter datatype is not currently supported.
this error is occured.

In other issue like this, the solution was replacing them with equivalent struct assignments.
(https://github.com/pulp-platform/pulpissimo/issues/231)

-  assign axi_lite_resp_o = '{
-    aw_ready: axi_req_valid[WR] & axi_req_ready[WR],        // if AXI AW & W valid & tree gnt_o[WR]
-    w_ready:  axi_req_valid[WR] & axi_req_ready[WR],        // if AXI AW & W valid & tree gnt_o[WR]
-    b:       '{resp: axi_bresp},                            // from spill reg
-    b_valid:  axi_bresp_valid,                              // from spill reg
-    ar_ready: axi_req_valid[RD] & axi_req_ready[RD],        // if AXI AR valid and tree gnt[RD]
-    r:       '{data: axi_rresp.data, resp: axi_rresp.resp}, // from spill reg
-    r_valid:  axi_rresp_valid                               // from spill reg
-  };
+  assign axi_lite_resp_o.aw_ready = axi_req_valid[WR] & axi_req_ready[WR];         // if AXI AW & W valid & tree gnt_o[WR]
+  assign axi_lite_resp_o.w_ready  = axi_req_valid[WR] & axi_req_ready[WR];         // if AXI AW & W valid & tree gnt_o[WR]
+  assign axi_lite_resp_o.b.resp   = axi_bresp;                            // from spill reg
+  assign axi_lite_resp_o.b_valid  = axi_bresp_valid;                               // from spill reg
+  assign axi_lite_resp_o.ar_ready = axi_req_valid[RD] & axi_req_ready[RD];         // if AXI AR valid and tree gnt[RD]
+  assign axi_lite_resp_o.r.data   = axi_rresp.data; // from spill reg
+  assign axi_lite_resp_o.r.resp   = axi_rresp.resp;
+  assign axi_lite_resp_o.r_valid  = axi_rresp_valid;                               // from spill reg
+  

However, the solution makes another error : Illegal use of a bit-select, part-select, member-select or mda element

I'vd replaced this part
             apb_req_o[apb_sel_idx] = '{
              paddr:   apb_req.addr,
              pprot:   apb_req.prot,
              psel:    1'b1,
              penable: 1'b0,
              pwrite:  apb_req.write,
              pwdata:  apb_req.data,
              pstrb:   apb_req.strb
             };

with this part.
 assign apb_req_o[apb_sel_idx].paddr = apb_req.addr;
 assign apb_req_o[apb_sel_idx].pprot = apb_req.prot;
 assign apb_req_o[apb_sel_idx].psel = 1'b1;
 assign apb_req_o[apb_sel_idx].penable = 1'b0;
 assign apb_req_o[apb_sel_idx].pwrite = apb_req.write;
 assign apb_req_o[apb_sel_idx].pwdata = apb_req.data;
 assign apb_req_o[apb_sel_idx].pstrb = apb_req.strb;

How can I resolve this error?
Thank you

bluewww, 2021-08-03:
Maybe instead of doing assignment statements you could try to use a `always_comb` block?

enqnwhgdk317, 2021-08-03:
I resolved this error using always_comb block!
Thank you!

bluewww, 2021-08-04:
If you think your changes are useful in general, then please make pull request. Closing this issue.

closed 2021-08-04 by bluewww